MagicMirrorOrg/MagicMirror #3005: Enhancement: DarkSkyAPI replacement PirateWeather

Tom-Hirschberger, 2023-01-14:
Found an [article](https://www.heise.de/news/Pirate-Weather-Kostenlose-Wetterdaten-API-fuer-Entwickler-7457155.html) that there is a developer who is providing a replacement api for darksky which is called [Pirate Weather](https://pirateweather.net/).
If i read correctly it should be a plugin-in replacement for Darksky and only the URLs in the modules need to be changed as the same data structures are used.
As i do not use any Darksky modules i did not test it but i found it interesting and it might be worth looking at it.

khassel, 2023-01-14:
Thanks!

Darksky is removed on `develop` but I think we should add id again under the new name.

rejas, 2023-01-14:
I'd happily merge a PR that adds this provider. Who wants this ticket assigned? ;-)

closed 2023-04-04 by khassel