MagicMirrorOrg/MagicMirror #3010: Weather provider: weathergov missing function

dWoolridge, 2023-01-16:
The weather provider "weathergov.js" throws the following error:

weathergov.js:115 Could not load data ...  TypeError: this.convertWindDirection is not a function
    at Class.generateWeatherObjectsFromHourly (weathergov.js:182:33)
    at weathergov.js:111:25

This functionality is needed for another module:  MMM-WeatherHourlyByDay

The following function needs to be added back into the weathergov.js file:

	/*
	Convert the direction into Degrees
	*/
	convertWindDirection(windDirection) {
		const windCardinals = {
			N: 0,
			NNE: 22,
			NE: 45,
			ENE: 67,
			E: 90,
			ESE: 112,
			SE: 135,
			SSE: 157,
			S: 180,
			SSW: 202,
			SW: 225,
			WSW: 247,
			W: 270,
			WNW: 292,
			NW: 315,
			NNW: 337
		};

		return windCardinals.hasOwnProperty(windDirection) ? windCardinals[windDirection] : null;
	}

rejas, 2023-01-16:
Thanks for the report @dWoolridge Can you test my PR and confirm the fixes?

dWoolridge, 2023-01-16:
Tested your PR:  The error is gone and MMM-WeatherHourlyByDay works.

PR #3011 (2023-01-16): Fix weathergov provider hourly weather forecast
Hourly forecast wasnt converted properly during the last release cycle, one fix and two cleanups were necessary.

Fixes #3010
commit c714399b (2023-01-16): Fix weathergov provider hourly weather forecast (#3011)

Hourly forecast wasnt converted properly during the last release cycle,
one fix and two cleanups were necessary.

Fixes #3010
closed 2023-04-04 by khassel