MagicMirrorOrg/MagicMirror #3014: Microsoft change the custom timezone calendar string AGAIN.., calendar fails

sdetweil, 2023-01-17:
the old string was 
```
"tzone://Microsoft/Custom"
```
the new string is 
```
"Customized Time Zone"
```
temp fix is to edit node_modules/node_ical/ical.js and change 

```
 if (tz === 'tzone://Microsoft/Custom' ){ 
```
 to 
 ```
 if (tz === 'tzone://Microsoft/Custom' || tz==='Customized Time Zone') {
 ```
 
 in two places 

I opened issue against node_ical, and will provide pr over there.. no way to fix over here..
https://github.com/jens-maus/node-ical/issues/246

note there will be messages from node_ical cause of the bad timezone into moment()
```
[ERROR] Moment Timezone has no data for Customized Time Zone. See http://momentjs.com/timezone/docs/#/data-loading/.
```

sdetweil, 2023-11-25:
node-ical 0.17.0  contains the fix for this..

note, it also includes a new rrule, which removes the problematic luxon libs for recurring events.. 

I will try to test..

khassel, 2023-11-26:
> so we could get rid of our luxon dependency too?

I think so, if we update node-ical we can remove luxon

sdetweil, 2023-11-26:
@rejas yes, as it won't be used

rejas, 2023-11-27:
cool, will wait for your PR / confirmation :-)

khassel, 2024-06-27:
> node-ical 0.17.0 contains the fix for this..

closed 2024-06-27 by khassel