MagicMirrorOrg/MagicMirror #3019: Weather module: Wind arrow is inverted

MagMar94, 2023-01-21:
If using the `showWindDirectionAsArrow`-setting, the arrow points at where the wind is blowing from. I feel it is more appropriate to point it in the direction of the wind. This is how wind arrows usually work in weather forecasts.

MagMar94, 2023-01-21:
I am going to fix this, but should I hide it behind a setting and leave the current arrow as the default? That way no one will get the change forced on them. However, I think most people want the inverted arrow.

rejas, 2023-01-21:
I also think that we should have the arrow pointed in the direction the wind is going. That makes most sense. Thanks for your report (and hopefully your fix ;-)

rejas, 2023-01-22:
Thx for your PR!

PR #3022 (2023-01-22): Point wind arrow in the direction the wind is flowing
Fixes #3019

The previous implementation had the arrow pointing in to the wind. When the wind blows from the north (0 degrees), the arrow should point straight down. In other words, no rotation of the arrow-down symbol. When the wind blows from the south (180 degrees), the arrow should point straight up (I.e. the arrow down symbol rotated 180 degrees).
commit 67e4dbaa (2023-01-22): Point wind arrow in the direction the wind is flowing (#3022)

Fixes #3019

The previous implementation had the arrow pointing in to the wind. When
the wind blows from the north (0 degrees), the arrow should point
straight down. In other words, no rotation of the arrow-down symbol.
When the wind blows from the south (180 degrees), the arrow should point
straight up (I.e. the arrow down symbol rotated 180 degrees).
closed 2023-04-04 by khassel