MagicMirrorOrg/MagicMirror #3026: Calendar module not updating date

Commiccannon, 2023-02-03:
Hello and thank you for opening an issue.

## I found a bug in MagicMirror

**Platform**: Raspberry pi 3b, chromium browser Raspbian OS

**Node Version**: 16.19.0

**MagicMirror² Version**: 2.22.0

**Description**: Calendar module is showing events from multiple days ago as in the future, it seems that it for some reason stopped updating its internal date

**Steps to Reproduce**: Install magic mirror with the default calendar, wait a few days

**Expected Results**: Events increments in days, as the event becomes past it disappears

**Actual Results**: The calendar appears to be stuck on a date in the past

**Configuration**: 
/* MagicMirror² Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * see https://docs.magicmirror.builders/configuration/introduction.html
 * and https://docs.magicmirror.builders/modules/configuration.html
 */
let config = {
        address: "0.0.0.0",     // Address to listen on, can be:
                                                        // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
                                                        // - another specific IPv4/6 to listen on a specific interface
                                                        // - "0.0.0.0", "::" to listen on any interface
                                                        // Default, when address config is left out or empty, is "localhost"
        port: 8080,
        basePath: "/",  // The URL path where MagicMirror² is hosted. If you are using a Reverse proxy
                                        // you must set the sub path here. basePath must end with a /
        ipWhitelist: [],        // Set [] to allow all IP addresses
                                                                                                                        // or add a>
                                                                                                                        // ["127.0.>
                                                                                                                        // or IPv4 >
                                                                                                                        // ["127.0.>

        useHttps: false,                // Support HTTPS or not, default "false" will use HTTP
        httpsPrivateKey: "",    // HTTPS private key path, only require when useHttps is true
        httpsCertificate: "",   // HTTPS Certificate path, only require when useHttps is true

        language: "en",
        locale: "en-US",
        logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
        timeFormat: 12,
        units: "metric",
        // serverOnly:  true/false/"local" ,
        // local for armv6l processors, default
        //   starts serveronly and then starts chrome browser
        // false, default for all NON-armv6l devices
        // true, force serveronly mode, because you want to.. no UI on this device

        modules: [
                {
                        module: "alert",
                },
                {
                        module: "updatenotification",
                        position: "top_bar"
                },
                {
                        module: "clock",
                        position: "top_left"
                },
                {
                        module: "calendar",
                        header: "Calendar",
                        position: "top_left",
                        config: {
                                calendars: [
                                        {
                                                symbol: "calendar-check",
                                                url: "#"
                                        },
                                        {
                                                symbol: "calendar-week",
                                                url: "#"
                                        }
                                ]
                        }
                },
                {
                        module: "compliments",
                        position: "middle_center",
                        config: {
                                remoteFile: "COM.json"
                        }
                },
                {
                        module: "weather",
                        position: "top_right",
                        config: {
                                weatherProvider: "openweathermap",
                                type: "current",
                                windUnits: "Knots",
                                locationID: "#", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the>
                                apiKey: "#"
                        }
                },
                {
                        module: "weather",
                        position: "top_right",
                        header: "Weather Forecast",
                        config: {
                                weatherProvider: "openweathermap",
                                type: "forecast",
                                windUnits: "Knots",
                                locationID: "#", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the>
                                apiKey: "#"
                        }
                },
        ]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

sdetweil, 2023-02-03:
we need to see the output where you start MagicMirror
 

open a terminal window, or use ssh to connect to the mm system

if using pm2 do this
pm2 stop all

then

cd ~/MagicMirror
npm start >sonefile.txt 2>&1

wait however long, 30 minutes,then press ctrl-c in the terminal window to stop MagicMirror

then examine the somefile.txt
and attach it here

Commiccannon, 2023-02-03:
Here is the log file as requested, i cannot see anything that would impact the calendar, and the graphics driver issues i googled and the consensus is that its not great but if things still work dont bother trying to fix as its unlikely to work with anything other then a full reinstall
[StartupLog.txt](https://github.com/MichMich/MagicMirror/files/10582955/StartupLog.txt)

sdetweil, 2023-02-03:
shows it fetched. processed and sent events for both calendars.   I would expect the holidays one to not change except daily.

download the personal url ics

curl  -sL the_url > somefile.ics

wait 30 minutes and do it again and compare the files, and wait overnight and do it again and compare the files. we just read what the server sends, have no control

Commiccannon, 2023-02-03:
So where it's getting an event for say the 3rd and saying it's tomorrow it's because the server is still sending all the events since the link was made or something?

sdetweil, 2023-02-03:
download the file like described above and edit that file w some text editor. the ics file is just text

each event is
```
BEGIN:VEVENT
.
.
.
END:VEVENT
```
the server writes that data out. we use a couple libraries to parse that into useful objects, and then process all that to get the collection of events that could be shown in the future. as we don't show events from the past (yesterday or before. or last hour...)

we have no idea how the server decided to provide the info, or how often it updates the data provided. we 'assume' that new events will be added and old events will be dropped, and repeated downloads of the data will have a new collection of vevents eventually.

sdetweil, 2023-03-27:
@Commiccannon did you ever download the ics data and examine it?

khassel, 2023-05-26:
no feedback, so closing ...

closed 2023-05-26 by khassel