MagicMirrorOrg/MagicMirror #3047: Default calendar module: Yearly event is displayed monthly

Tom-Hirschberger, 2023-02-21:
Hi,

as i already wrote in my pull request the default calendar module displays my yearly birthday events every month.
I attached a ics file (needed to rename it to .txt to be able to upload) with one of the events that causes the problem.

It happened with the current development version (without my changes of the pull request).

[jahrestage.txt](https://github.com/MichMich/MagicMirror/files/10795025/jahrestage.txt)

My configuration looks like the following:
```
		{
			module: "calendar",
			position: "top_center",
			config: {
				timeFormat: "absolute",
				fade: false,
				nextDaysRelative: true,
				displaySymbol: false,
				dateFormat: "dddd MMM Do",
				fullDayEventDateFormat: "dddd MMM Do",
				maxTitleLength: 50,
				//maximumNumberOfDays: 14,
				calendars: [
					{
						url: "http://localhost:8080/modules/jahrestage.ics",
						name: "Jahrestage", // <-- specify calendar name
						broadcastPastEvents: true, // <-- need to broadcast past events
						//maximumEntries: 7, // <-- assign enough number to prevent truncating new events by old events.
						symbol: [], // <-- when you want to display simbol. If you don't want, just set as `symbol:[],`
					}
				]
			},
		},
```

Tom-Hirschberger, 2023-02-22:
I removed some personal data of the ics file (did not want to share google id and birthday with name of family and friends to the public).
I tried running the ics file with `debug.js` and the effect is the same with the modified and unmodfied version.

khassel, 2023-03-01:
I compared the provided ics with one of my birthday events and found that

`RRULE:FREQ=YEARLY;WKST=MO;BYMONTHDAY=22`

seems to be problematic. My event contains only

`RRULE:FREQ=YEARLY`

The param `BYMONTHDAY` seems to be the problem, I found some documentation for RRULE [here](https://dateutil.readthedocs.io/en/stable/rrule.html).

Looking in the examples section I found

```bash
Every Friday the 13th, 4 occurrences.

 >>> list(rrule(YEARLY, count=4, byweekday=FR, bymonthday=13,
 ...            dtstart=parse("19970902T090000")))
 [datetime.datetime(1998, 2, 13, 9, 0),
  datetime.datetime(1998, 3, 13, 9, 0),
  datetime.datetime(1998, 11, 13, 9, 0),
  datetime.datetime(1999, 8, 13, 9, 0)]
```
 so for me your RRULE looks working as designed. Question is why BYMONTHDAY and WKST are set in your google calendar?

Tom-Hirschberger, 2023-03-01:
Hi,

i created the entry with the pre installed calendar app on my Samsung phone.
I deleted the entry and re-created it. The result looks like the following:

```
BEGIN:VEVENT
DTSTART;VALUE=DATE:20240122
DTEND;VALUE=DATE:20240123
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=1;BYMONTHDAY=22
DTSTAMP:20230301T204310Z
UID:6oqj2d9o75j3ib9p69j3gb9kc9gmcb9o64q3eb9hcphj8pj665ij4dpg64@google.com
CREATED:20230301T204257Z
DESCRIPTION:
LAST-MODIFIED:20230301T204257Z
LOCATION:
SEQUENCE:0
STATUS:CONFIRMED
SUMMARY:Carsten Glockann (1980)
TRANSP:OPAQUE
END:VEVENT
```

Now everything looks normal if i run `debug.js`.

I looked at the RRULE documentation, too and do have the same opinion as you. `FREQ=YEARLY` should be enough. I do not know if it is the app or if it is google that adds the bloat.

Tom-Hirschberger, 2023-03-01:
I edited the entry of "Martin Huber" with the phone and only saved (nothing changed still yearly) now it looks like:
```
BEGIN:VEVENT
DTSTART;VALUE=DATE:20220814
DTEND;VALUE=DATE:20220815
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=8;BYMONTHDAY=14
EXDATE;VALUE=DATE:20220814
DTSTAMP:20230301T205405Z
UID:vi01bt4enjr87l7fh3vmga9lhq@google.com
CREATED:20220107T203447Z
DESCRIPTION:
LAST-MODIFIED:20230301T205356Z
LOCATION:
SEQUENCE:1
STATUS:CONFIRMED
SUMMARY:Martin Huber (1983)
TRANSP:OPAQUE
END:VEVENT
```

khassel, 2023-03-01:
> 2nd Edit: The last entry of "Martin Huber" is not displayed in the debug.js

his next birthday would be 14.08.2023 but until is set to 13.08.2023 so this seems correct.

Tom-Hirschberger, 2023-03-01:
But the Samsung app showed "repeat until forever" even before i saved it again.
I think there was a bug in the Samsung Calendar app as i created some of the entries.

Tom-Hirschberger, 2023-03-01:
Checked all my entries in the calendar. Only two entries had the problematic format. I changed them and saved the new and now everything looks fine. 
```
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=8;BYMONTHDAY=14
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=12;BYMONTHDAY=9
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=8;BYMONTHDAY=18
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=8;BYMONTHDAY=16
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=1;BYMONTHDAY=22
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=3;BYMONTHDAY=27
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=10;BYMONTHDAY=7
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=9;BYMONTHDAY=19
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=9;BYMONTHDAY=1
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=7;BYMONTHDAY=24
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=7;BYMONTHDAY=17
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=7;BYMONTHDAY=9
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=6;BYMONTHDAY=4
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=6;BYMONTHDAY=4
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=5;BYMONTHDAY=3
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=3;BYMONTHDAY=19
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=3;BYMONTHDAY=18
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=3;BYMONTHDAY=17
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=3;BYMONTHDAY=12
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=1;BYMONTHDAY=28
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=1;BYMONTHDAY=20
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=12;BYMONTHDAY=26
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=12;BYMONTHDAY=10
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=12;BYMONTHDAY=8
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTH=11;BYMONTHDAY=10
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
RRULE:FREQ=YEARLY
```

khassel, 2023-04-07:
> But in short i do not think this is a problem of MagicMirror. Maybe it would be better if i create a issue in the rrule library project.

so can we close this issue @Tom-Hirschberger ?

Tom-Hirschberger, 2023-04-07:
Oh, sorry. Yes for sure.
I can close it, too.

nkl-kst, 2025-01-30:
Found this issue in the rrule lib: https://github.com/jkbrzt/rrule/issues/275 Just in case someone wants to dig even deeper 🙂

sdetweil, 2025-01-30:
@nkl-kst thx. i already work on node-ical, and not going to work on rrule problems exposed by bad cal app

has anyone retried this w v2.30 as i made significant change to the input into rrule.between()