MagicMirrorOrg/MagicMirror #3052: Fade doesn't seem to respected in calendar module

i-am-shodan, 2023-02-24:
**Platform**: 
* Docker on a raspberry Pi 4 - karsten13/magicmirror:latest
* Display is a Surface Pro 3 running Windows 11, running Edge is Kiosk mode

**Node Version**: Make sure it's version 14 or later (recommended is 16).
v18.14.1

**MagicMirror² Version**: Please let us know which version of MagicMirror² you are running. It can be found in the `package.json` file.
v2.22.0

**Description**:
I'm using the calendar component and have set it up to use Google Calendar with these settings

```
{
    module: 'calendar',
    header: 'Calendar',
    position: 'middle_center',
    fade: false,
    config: {
        calendars: [{
            symbol: 'calendar-check-o ',
            url: 'https://calendar.google.com/calendar/ical/MYTOKEN/basic.ics'
        }]
    }
},
```

I had fade set but my entries are still fading away

[image]

**Steps to Reproduce**: 
* Set up config as above
* View site with Edge

**Expected Results**:
* Items should not fade to black

**Actual Results**:
* Items are fading away

i-am-shodan, 2023-02-24:
I just checked to ensure the version of calendar.js used by the docker container is the same as the latest and can confirm they have the same MD5SUM

khassel, 2023-02-24:
try

```js
{
    module: 'calendar',
    header: 'Calendar',
    position: 'middle_center',
    config: {
        fade: false,
        calendars: [{
            symbol: 'calendar-check-o ',
            url: 'https://calendar.google.com/calendar/ical/MYTOKEN/basic.ics'
        }]
    }
},
```

closed 2023-02-24 by i-am-shodan