MagicMirrorOrg/MagicMirror #3056: disabling CLOCK_SECOND notification

slametps, 2023-03-02:
is there a way to NOT send NOTIFICATION CLOCK_SECOND every single second (flooding) with `displaySeconds = true` in CLOCK module?

sdetweil, 2023-03-02:
what does the documentation say?

sdetweil, 2023-03-02:
no..  but you could add another option and submit a pr

sendNotification:true,

if (this.config.sendNotification)

in config.js for clock
sendNotification:false

sdetweil, 2023-03-02:
u can also filter in the dev console
using a part of the module name u care about in the filter field

TimeTraveller255, 2023-03-03:
comment this line in clock.js
	self.sendNotification("CLOCK_SECOND", self.second);
like this one
//	self.sendNotification("CLOCK_SECOND", self.second);

slametps, 2023-03-06:
> comment this line in clock.js self.sendNotification("CLOCK_SECOND", self.second); like this one // self.sendNotification("CLOCK_SECOND", self.second);

Previously, I did it the hard way the same as you suggest. 
I just not OK with every second sending notification causing flooding (resource consume). Still OK with CLOCK_MINUTE.
And I did not want to disable any other notifications, CLOCK_SECOND only. But I can't find the polite way to to this.

PR #3059 (2023-04-01): Add sendNotifications option to clock module
Fixes #3056 

One question here would be if the default for this new option should be true or false.

True: keeps the current behaviour, nobody needs to change his config if they rely on this option

False: keeps the clock notifications quiet, doesnt waste time/resources, keeps the noise low

Maybe the original author @cybex-dev can weigh in on this, and why he added this notification.
commit 62235843 (2023-04-01): Add sendNotifications option to clock module (#3059)

Fixes #3056 

One question here would be if the default for this new option should be
true or false.

True: keeps the current behaviour, nobody needs to change his config if
they rely on this option

False: keeps the clock notifications quiet, doesnt waste time/resources,
keeps the noise low

Maybe the original author @cybex-dev can weigh in on this, and why he
added this notification.

---------
closed 2023-04-04 by rejas