MagicMirrorOrg/MagicMirror #3057: Where does the `.weather` class get implemented in the forecast.njk

m000lie, 2023-03-03:
## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Raspberry Pi 400 (Electron JS)

**Node Version**: v18.14.2
**MagicMirror² Version**: v2.22.0

**Description**: `weather` class is used to modify subclassed elements in `forecast.njk` and is defined in the `weather.css` file, however, there is **NO** element in `forecast.njk` file that contains the `weather` class. Despite this, I am still able to add custom styles to subclassed elements. (E.g. .weather .day { font-size:1000%;} <-- still works) I noticed that the main table element in `forecast.njk` has the class `config.tableClass` variable interpolated. This is worked out to be the literal value "small", as defined in the defaults dictionary of the `weather.js` file. Thus, modifying subclassed elements using this parent class also works. (E.g. .small .day { font-size:1000%;} <-- also works) Please explain why this is happening. My best hypothesis is that there is a parent wrapper element that encompasses all elements in the .njk files and holds the `.weather` class. 

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: /* MagicMirror² Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * see https://docs.magicmirror.builders/configuration/introduction.html
 * and https://docs.magicmirror.builders/modules/configuration.html
 */
let config = {
	address: "localhost", 	// Address to listen on, can be:
							// - "localhost", "127.0.0.1", "::1" to listen on loopback interface
							// - another specific IPv4/6 to listen on a specific interface
							// - "0.0.0.0", "::" to listen on any interface
							// Default, when address config is left out or empty, is "localhost"
	port: 8080,
	basePath: "/", 	// The URL path where MagicMirror² is hosted. If you are using a Reverse proxy
					// you must set the sub path here. basePath must end with a /
	ipWhitelist: [], 	// Set [] to allow all IP addresses
															// or add a specific IPv4 of 192.168.1.5 :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
															// or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "en",
	locale: "en-US",
	logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
	timeFormat: 24,
	units: "metric",
	// serverOnly:  true/false/"local" ,
	// local for armv6l processors, default
	//   starts serveronly and then starts chrome browser
	// false, default for all NON-armv6l devices
	// true, force serveronly mode, because you want to.. no UI on this device

	modules: [
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "helloworld",
			position: "lower_third",
			
		},
		{
			module: "weather",
			position: "top_right",
			config: {
				weatherProvider: "openweathermap",
				type: "current",
				location: "Singapore",
				locationID: "1880252", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				apiKey: "f969f4365925f10475df48a85a8533ea"
			}
		},
		{
			module: "weather",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				weatherProvider: "openweathermap",
				type: "forecast",
				location: "Singapore",
				locationID: "1880252", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				apiKey: "f969f4365925f10475df48a85a8533ea"
			}
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "New York Times",
						url: "https://rss.nytimes.com/services/xml/rss/nyt/HomePage.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: true,
				broadcastNewsFeeds: true,
				broadcastNewsUpdates: true
			}
		},
	]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

**Additional Notes**: This error obfuscates the experience of new developers who are more fallible. It doubles down on their innocence and leads them down a rabbit hole filled with irrelevance and confusion. This should have been documented in the official docs. 
[weather.zip](https://github.com/MichMich/MagicMirror/files/10886195/weather.zip)

rejas, 2023-03-03:
First: Please dont post your api keys here, privacy risk.

About the weather class: You cant find that selector in the forecast.njk because this is a selector that is in the general div for the "weather" module, see screenshot:

[image]

I hope this clears things up?

m000lie, 2023-03-03:
Right, totally missed the api key in there oops 

Where could I find the div in your screenshot?

rejas, 2023-03-03:
Topmost element which is highlighted

m000lie, 2023-03-03:
I meant in which file can i find said div?

rejas, 2023-03-03:
it gets created here: https://github.com/MichMich/MagicMirror/blob/f14e95616699353f59738b6b64c553ecbfab4a28/js/main.js#L27

sdetweil, 2023-03-03:
the module name is used as a class in the module _content element..

as such it is a wrapper for all the other content below that div

m000lie, 2023-03-04:
Thank you guys, I've managed to find it. Closing this issue now 👍

closed 2023-03-04 by m000lie