MagicMirrorOrg/MagicMirror #3075: GPU process exited unexpectedly: exit_code=134

Kiwiers, 2023-03-27:
## I'm not sure if this is a bug, no matter what I do I face the same issue:

**Platform**:  Raspberry Pi 3b+

**Node Version**: v16.19.1

**MagicMirror² Version**: 2.22.0 

**Description**: This error shows up, tried all possible  troubleshooting methods from github.

**Steps to Reproduce**: pm2 start <app>

**Actual Results**: 
```js
0|MagicMir | }
0|MagicMir | [27.03.2023 14:01.30.125] [ERROR] MagicMirror² will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
0|MagicMir | [27.03.2023 14:01.30.128] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | ../../buildtools/third_party/libc++/trunk/include/vector:1425: assertion __n < size() failed: vector[] index out of bounds[3279:0327/140132.777514:ERROR:gpu_process_host.cc(991)] GPU process exited unexpectedly: exit_code=134
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | ../../buildtools/third_party/libc++/trunk/include/vector:1425: assertion __n < size() failed: vector[] index out of bounds[3279:0327/140134.757741:ERROR:gpu_process_host.cc(991)] GPU process exited unexpectedly: exit_code=134
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | ../../buildtools/third_party/libc++/trunk/include/vector:1425: assertion __n < size() failed: vector[] index out of bounds[3279:0327/140136.825893:ERROR:gpu_process_host.cc(991)] GPU process exited unexpectedly: exit_code=134
```

**Configuration**: 
```js
/* MagicMirror² Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * see https://docs.magicmirror.builders/configuration/introduction.html
 * and https://docs.magicmirror.builders/modules/configuration.html
 */
let config = {
	address: "0.0.0.0", 	// Address to listen on, can be:
							// - "localhost", "127.0.0.1", "::1" to listen on loopback interface
							// - another specific IPv4/6 to listen on a specific interface
							// - "0.0.0.0", "::" to listen on any interface
							// Default, when address config is left out or empty, is "localhost"
	port: 8080,
	basePath: "/", 	// The URL path where MagicMirror² is hosted. If you are using a Reverse proxy
					// you must set the sub path here. basePath must end with a /
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1","192.168.1.71"], 	// Set [] to allow all IP addresses
															// or add a specific IPv4 of 192.168.1.5 :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
														// or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "en",
	locale: "en-US",
	logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
	timeFormat: 24,
	units: "metric",
	// serverOnly:  true/false/"local" ,
	// local for armv6l processors, default
	//   starts serveronly and then starts chrome browser
	// false, default for all NON-armv6l devices
	// true, force serveronly mode, because you want to.. no UI on this device

	modules: [

{
    module: 'mmm-dropbox',
    position: 'fullscreen_below', // pick whichever position you want
    config: {
        token: 'xxxxx',
        folder: 'hey-pics',

    }
},
//		  {
//   module: "MMM-Wallpaper",
//    position: "fullscreen_below",
//    config: { // See "Configuration options" for more information.
//      source: "icloud:#xxxxxxxxxxxx",
//      slideInterval: 60 * 1000,
//	  updateInterval: 60 * 60 * 1000,
//	  size: "auto", // Change slides every minute
//	  crossfade: true,
//	  shuffle: true,
//	  orientation: "auto"	  
//  }
//  },
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
			header: "US Holidays",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check",
						url: "webcal://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics"
					}
				]
			}
		},
		{
			module: "compliments",
			position: "lower_third"
		},
		{
			module: "weather",
			position: "top_right",
			config: {
				weatherProvider: "openweathermap",
				type: "current",
				location: "Austin",
				locationID: "4671654", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				apiKey: "xxxxxxxxxxxxxxxx"
			}
		},
		{
			module: "weather",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				weatherProvider: "openweathermap",
				type: "forecast",
				location: "New York",
				locationID: "5128581", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				apiKey: "xxxxxxxxxxxx"
			}
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "New York Times",
						url: "https://rss.nytimes.com/services/xml/rss/nyt/HomePage.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: true,
				broadcastNewsFeeds: true,
				broadcastNewsUpdates: true
			}
		},
	]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}
```

**Additional Notes**: Any idea on why my MMM-Wallpaper is also not receiving the feed from iCloud shared  #fodler?

Kiwiers, 2023-03-27:
[image]

I was able to fix the issue with "MMM-WallPaper" thank you.

> what version of the OS are you running?
> 
> show the output of
> 
> ```
> lsb_release -a 
> ```
> 
> and
> 
> ```
> uname -a
> ```
> 
> MMM-WallPaper is a 3rd party module.. not part of the base. there might be some help on the forums. this modules isn't used much

closed 2023-03-27 by sdetweil

bweave, 2023-03-28:
@Kiwiers I'm having the same issue. How did you fix it with MMM-Wallpaper?

Kiwiers, 2023-03-28:
> 
Honestly I just switched to Dropbox with [MMM-DropboxWallpaper](https://github.com/eouia/MMM-DropboxWallpaper)..
I believe the iCloud function is broken 😞 

Why was the issue closed??? the application works, but still shows the errors, I'm not sure if that's ok 😆

sdetweil, 2023-03-28:
I closed it cause you said the module was working, you can reopen it.  gpu errors are typically some os setup problem, but unless you are doing heavy graphics, it doesn't matter much.

bweave, 2023-03-29:
@Kiwiers FWIW, I'm experiencing it with local images.

bweave, 2023-03-29:
Thanks @sdetweil, trying it now!