MagicMirrorOrg/MagicMirror #3080: Environment Canada weather displays wrong times for hourly weather

briodan, 2023-04-05:
When configuring envcanada as the weather source the hourly time is displayed wrong, the temperatures match what is being reported by the weather station in the correct order

[image]
[image]

here is the config i'm using
```
		{
			"module": "weather",
			"header": "Current Weather",
			"position": "top_right",
			"config": {
				"location": "Calgary, AB",
				"weatherProvider": "envcanada",
				"siteCode": "s0000047",
				"provCode": "AB",
				"roundTemp": false,
				"type": "hourly",
				"units": "metric",
				"useKmh": false,
				"tempUnits": "metric",
				"windUnits": "metric",
				"updateInterval": 600000,
				"animationSpeed": 1000,
				"timeFormat": 12,
				"showPeriod": true,
				"showPeriodUpper": false,
				"showWindDirection": true,
				"showWindDirectionAsArrow": false,
				"useBeaufort": true,
				"lang": "en",
				"showHumidity": false,
				"showSun": true,
				"degreeLabel": true,
				"decimalSymbol": ".",
				"showIndoorTemperature": false,
				"showIndoorHumidity": false,
				"maxNumberOfDays": 5,
				"maxEntries": 12,
				"ignoreToday": false,
				"fade": false,
				"fadePoint": 0.25,
				"initialLoadDelay": 0,
				"appendLocationNameToHeader": true,
				"calendarClass": "calendar",
				"tableClass": "small",
				"onlyTemp": false,
				"showPrecipitationAmount": false,
				"colored": false,
				"showFeelsLike": true,
				"absoluteDates": false
			}
```

seeing this issue on both a Pi3 and a debian 11.4.0 linux vm, both using the april release, note this issue was also present in the previous release.

rejas, 2023-04-09:
Thx for the report @briodan can you maybe test my PR if it fixes your problem for real?

briodan, 2023-04-09:
yeap that did the trick, thanks

[image]
[image]

khassel, 2023-07-01:
fixed with new release `v2.24.0`

PR #3090 (2023-04-09): Fix envcanada hourly forecast time 
Fixes #3080
commit 4e336906 (2023-04-09): Fix envcanada hourly forecast time  (#3090)

Fixes #3080

---------
closed 2023-07-01 by khassel