MagicMirrorOrg/MagicMirror #3082: Fetch timeout with Electron >= 23

rejas, 2023-04-05:
After the update to Electron 24 (https://github.com/MichMich/MagicMirror/pull/3079) my calendar stopped working and just reports timeouts.

Running in server mode and accessing via a browser works fine, also downgrading to Electron 22 fixes it.

Anyone else has this problem? @khassel

khassel, 2023-04-05:
calender from config sample works with electron v24 in my docker setup.

So have you any errors in the logs or dev console? Can you try with the calender from config sample?

rejas, 2023-04-06:
Only this shows up after a while when logging the response in NodeHelper.checkFetchStatus:

[06.04.2023 15:44.02.390] [ERROR] Calendar Error. Could not fetch calendar:  https://xxxxxx.de:63175/caldav/kalender/gsiuahuk/ FetchError: request to https://xxxxxx.de:63175/caldav/kalender/gsiuahuk/ failed, reason: connect ETIMEDOUT 2a01:c22:a970:1b00:b871:9b2d:ca9a:47e8:63175
    at ClientRequest.<anonymous> (/Users/michael/Projects/MagicMirror/node_modules/node-fetch/lib/index.js:1505:11)

The config sample calendar works.

The difference is
- configsample uses webdav: protocol, mine https: 
- configsample needs no authentication, mine has baisc authentication (user/pw)

Do you know of any "public" calendars that use https and/or basic authentication?

khassel, 2023-04-06:
You can use the calendar from config sample with `http` or `https` instead of `webdav` without problems.

So I think it's a problem with your basic auth, can you provide a full example how are you doing this? Or create a public calendar with test entries and dummy user/password so others could test this? Are there any errors in the dev console (may concerning cors)?

khassel, 2023-04-06:
your example works in my setup ...

```bash
pi in 🌐 raspberrypi in magicmirror/run on  develop [?]
❯ docker logs mm
copy default modules to host ...
copy css files to host ...
chown modules and config folder ...
timezone is Europe/Berlin
start magicmirror

> magicmirror@2.24.0-develop start
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[06.04.2023 21:18.41.729] [LOG]   Starting MagicMirror: v2.24.0-develop
[06.04.2023 21:18.41.736] [LOG]   Loading config ...
[06.04.2023 21:18.41.739] [DEBUG] config template file not exists, no envsubst
[06.04.2023 21:18.41.745] [LOG]   Loading module helpers ...
[06.04.2023 21:18.41.752] [LOG]   Initializing new module helper ...
[06.04.2023 21:18.41.753] [LOG]   Module helper loaded: mmpm
[06.04.2023 21:18.41.754] [LOG]   No helper found for module: alert.
[06.04.2023 21:18.41.767] [LOG]   Initializing new module helper ...
[06.04.2023 21:18.41.767] [LOG]   Module helper loaded: updatenotification
[06.04.2023 21:18.41.768] [LOG]   No helper found for module: clock.
[06.04.2023 21:18.42.052] [LOG]   Initializing new module helper ...
[06.04.2023 21:18.42.053] [LOG]   Module helper loaded: calendar
[06.04.2023 21:18.42.054] [LOG]   No helper found for module: compliments.
[06.04.2023 21:18.42.055] [LOG]   No helper found for module: weather.
[06.04.2023 21:18.42.086] [LOG]   Initializing new module helper ...
[06.04.2023 21:18.42.086] [LOG]   Module helper loaded: newsfeed
[06.04.2023 21:18.42.087] [LOG]   All module helpers loaded.
[06.04.2023 21:18.42.103] [LOG]   Starting server on port 8080 ...
[06.04.2023 21:18.42.108] [WARN]  You're using a full whitelist configuration to allow for all IPs
[06.04.2023 21:18.42.119] [LOG]   Server started ...
[06.04.2023 21:18.42.121] [LOG]   Connecting socket for: mmpm
[06.04.2023 21:18.42.121] [LOG]   Starting module helper: mmpm
[06.04.2023 21:18.42.122] [LOG]   Connecting socket for: updatenotification
[06.04.2023 21:18.42.122] [LOG]   Starting module helper: updatenotification
[06.04.2023 21:18.42.122] [LOG]   Connecting socket for: calendar
[06.04.2023 21:18.42.123] [LOG]   Starting node helper for: calendar
[06.04.2023 21:18.42.123] [LOG]   Connecting socket for: newsfeed
[06.04.2023 21:18.42.124] [LOG]   Starting node helper for: newsfeed
[06.04.2023 21:18.42.124] [LOG]   Sockets connected & modules started ...
[06.04.2023 21:18.42.347] [LOG]   Launching application.
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
[06.04.2023 21:18.44.818] [LOG]   Create new calendarfetcher for url: https://w7bar.de/caldav/magicm/kipmqvm/ - Interval: 300000
[06.04.2023 21:18.44.878] [LOG]   Create new newsfetcher for url: https://rss.nytimes.com/services/xml/rss/nyt/HomePage.xml - Interval: 300000
[06.04.2023 21:18.44.892] [INFO]  Checking git for module: mmpm
[06.04.2023 21:18.44.901] [INFO]  Checking git for module: MagicMirror

pi in 🌐 raspberrypi in magicmirror/run on  develop [?]
❯ docker exec -it mm bash
node@raspberrypi:/opt/magic_mirror$ node -v
v19.8.1
node@raspberrypi:/opt/magic_mirror$ npm -v
9.5.1
node@raspberrypi:/opt/magic_mirror$ npx electron -v
v24.0.0
```

rejas, 2023-04-07:
Can you try again? Still doesnt work on my side and now I am not sure if that is because I am on the same network as the calendar server ....

khassel, 2023-04-07:
> Can you try again?

still working ...

rejas, 2023-04-07:
Very strange... Still no luck for me.

Well, I can use the ip adress here internally for my MM, so thats my workaround for now. Still, we should keep this in mind in case some other users are reporting network errors with the latest develop branch

closed 2023-04-07 by rejas

khassel, 2023-04-07:
> Still, we should keep this in mind in case some other users are reporting network errors with the latest develop branch

Sure!

My guess is new security features of the browser included in electron, v24 contains Chromium 112.0.5615.49.