MagicMirrorOrg/MagicMirror #3083: Not running properly in W10

MAMuhajir, 2023-04-06:
I tried to run on the MagicMirror2 on W10 but it hangs with these messages being shown:
[image]
I followed the steps from the installation guide and already tried the set ELECTRON_DISABLE_GPU=1 method
[image]

sdetweil, 2023-04-06:
originally reported thru discord

sdetweil, 2023-04-06:
the electron app start doesn't return the config 

```
	core.start().then((c) => (config = c));
```

start is an async function that does a return, this is expecting a promise resolve. 

changing to 
```
   let config = await core.start()
```
fails as electron is not an async module.. 

the no display failure is the port config is lost 
added debug in electron.js
```
  Launching application.
[06.04.2023 15:54.55.219] [LOG]   new window created
[06.04.2023 15:54.55.228] [LOG]   loading url at  http://localhost:undefined
[06.04.2023 15:54.55.277] [LOG]   Server started ...
```

sdetweil, 2023-04-07:
the problem is that app.on('ready' is set and will fire as soon as app is 'ready', but before start() returns

```
// This method will be called when Electron has finished
// initialization and is ready to create browser windows.
app.on("ready", function () {
	Log.log("Launching application.");
	createWindow();
});
```

but app.start is now async, so call to start() starts work and returns immediately, and ends electron.js inline code, config is lost (as no await result)
app is configured in app.start() and ready fires, needing config, which is not set. 

my fix

comment out app.on('ready'....

then change end of electron.js 
```
if (["localhost", "127.0.0.1", "::1", "::ffff:127.0.0.1", undefined].includes(config.address)) {
	let foo = async()=>{   // need async for await on core.start(), but async lets electron.js complete before start
		console.log("waiting for app.start to complete")
		config = await core.start()
		console.log("in foo config="+JSON.stringify(config))  // config is good here
		Log.log("Launching application.");
		createWindow();  // all the rest of app setup is completed here 
	//core.start().then((c) => (config = c));  // old code
	}
	foo() // starts app.start , but doesn't wait,
	console.log("electron finished");
}
```
old flow 
```
C:\Users\sdetw\MagicMirror>npm start

> magicmirror@2.23.0 start
> node_modules\.bin\electron js\electron.js

[07.04.2023 07:25.15.952] [LOG]   Starting MagicMirror: v2.23.0
[07.04.2023 07:25.15.956] [LOG]   Loading config ...
[07.04.2023 07:25.15.965] [DEBUG] config template file not exists, no envsubst
[07.04.2023 07:25.15.965] [LOG]    config filename=C:\Users\sdetw\MagicMirror\config\config.js
[07.04.2023 07:25.15.991] [LOG]   electron finished  <----------   core.start()   done..oops..
[07.04.2023 07:25.15.992] [LOG]   Loading module helpers ...
[07.04.2023 07:25.15.993] [LOG]   No helper found for module: alert.
[07.04.2023 07:25.16.043] [LOG]   Initializing new module helper ...
[07.04.2023 07:25.16.044] [LOG]   Module helper loaded: updatenotification
[07.04.2023 07:25.16.045] [LOG]   No helper found for module: clock.
[07.04.2023 07:25.16.935] [LOG]   Initializing new module helper ...
[07.04.2023 07:25.16.937] [LOG]   Module helper loaded: calendar
[07.04.2023 07:25.16.937] [WARN]  No C:\Users\sdetw\MagicMirror\js/../modules/currentweather/currentweather.js found for module: currentweather.
[07.04.2023 07:25.16.938] [LOG]   No helper found for module: currentweather.
[07.04.2023 07:25.17.076] [LOG]   Initializing new module helper ...
[07.04.2023 07:25.17.077] [LOG]   Module helper loaded: newsfeed
[07.04.2023 07:25.20.268] [LOG]   Initializing new module helper ...
[07.04.2023 07:25.20.268] [LOG]   Module helper loaded: MyCovid19
[07.04.2023 07:25.20.269] [WARN]  No C:\Users\sdetw\MagicMirror\js/../modules/MMM-ImagesPhotos/MMM-ImagesPhotos.js found for module: MMM-ImagesPhotos.
[07.04.2023 07:25.20.270] [LOG]   No helper found for module: MMM-ImagesPhotos.
[07.04.2023 07:25.20.270] [WARN]  No C:\Users\sdetw\MagicMirror\js/../modules/MMM-WiFiPassword1/MMM-WiFiPassword1.js found for module: MMM-WiFiPassword1.
[07.04.2023 07:25.20.270] [LOG]   No helper found for module: MMM-WiFiPassword1.
[07.04.2023 07:25.20.270] [LOG]   All module helpers loaded.
[07.04.2023 07:25.20.310] [LOG]   Starting server on port 8090 ...
[07.04.2023 07:25.20.319] [WARN]  You're using a full whitelist configuration to allow for all IPs
[07.04.2023 07:25.20.410] [LOG]   Launching application.
[07.04.2023 07:25.20.457] [LOG]   new window created  <---  createwindow called from app.ready
[07.04.2023 07:25.20.476] [LOG]   loading url at  http://localhost:undefined   <----- config needed in electron, not returned
[07.04.2023 07:25.20.494] [LOG]   Server started ...
[07.04.2023 07:25.20.494] [LOG]   Connecting socket for: updatenotification
[... 34 lines cut ...]
[07.04.2023 07:04.40.221] [LOG]   Module helper loaded: newsfeed
[07.04.2023 07:04.40.447] [LOG]   Initializing new module helper ...
[07.04.2023 07:04.40.448] [LOG]   Module helper loaded: MyCovid19
[07.04.2023 07:04.40.448] [WARN]  No C:\Users\sdetw\MagicMirror\js/../modules/MMM-ImagesPhotos/MMM-ImagesPhotos.js found for module: MMM-ImagesPhotos.
[07.04.2023 07:04.40.449] [LOG]   No helper found for module: MMM-ImagesPhotos.
[07.04.2023 07:04.40.449] [WARN]  No C:\Users\sdetw\MagicMirror\js/../modules/MMM-WiFiPassword1/MMM-WiFiPassword1.js found for module: MMM-WiFiPassword1.
[07.04.2023 07:04.40.449] [LOG]   No helper found for module: MMM-WiFiPassword1.
[07.04.2023 07:04.40.449] [LOG]   All module helpers loaded.
[07.04.2023 07:04.40.456] [LOG]   Starting server on port 8090 ...
[07.04.2023 07:04.40.466] [WARN]  You're using a full whitelist configuration to allow for all IPs
[07.04.2023 07:04.40.544] [LOG]   Server started ...
[07.04.2023 07:04.40.545] [LOG]   Connecting socket for: updatenotification
[07.04.2023 07:04.40.546] [LOG]   Starting module helper: updatenotification
[07.04.2023 07:04.40.546] [LOG]   Connecting socket for: calendar
[07.04.2023 07:04.40.568] [LOG]   Starting node helper for: calendar
[07.04.2023 07:04.40.584] [LOG]   Connecting socket for: newsfeed
[07.04.2023 07:04.40.591] [LOG]   Starting node helper for: newsfeed
[07.04.2023 07:04.40.597] [LOG]   Connecting socket for: MyCovid19
[07.04.2023 07:04.40.612] [LOG]   Starting node_helper for module: MyCovid19
[07.04.2023 07:04.40.634] [LOG]   Sockets connected & modules started ...
[07.04.2023 07:04.40.642] [LOG]   Launching application.
                                                                                           // core.start await ended, config returned
[07.04.2023 07:04.40.698] [LOG]   new window created  // createWindow called
[07.04.2023 07:04.40.716] [LOG]   loading url at  http://localhost:8090  ... 
```

sdetweil, 2023-04-07:
#2487 is broken here

rejas, 2023-04-07:
So this happens only on windows 10 or can we recreate the problem on mac/linux too?

sdetweil, 2023-04-07:
i will review that.. cause its windows its a faster machine than pi..

khassel, 2023-04-07:
Only ment as help for finding the issue: I can't reproduce this when running a docker image with electron under wsl2, it works and the create order is correct:

```bash
magic_mirror on  develop [✘!?] is 📦 v2.24.0-develop via  v19.8.1
⬢ [Docker] ❯ npm run start

> magicmirror@2.24.0-develop start
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[98:0407/224807.101418:ERROR:bus.cc(399)] Failed to connect to the bus: Failed to connect to socket /run/dbus/system_bus_socket: Permission denied
[07.04.2023 22:48.07.523] [LOG]   Starting MagicMirror: v2.24.0-develop
[07.04.2023 22:48.07.526] [LOG]   Loading config ...
[07.04.2023 22:48.07.527] [DEBUG] config template file not exists, no envsubst
[07.04.2023 22:48.07.528] [LOG]   Loading module helpers ...
[07.04.2023 22:48.07.529] [LOG]   No helper found for module: alert.
[07.04.2023 22:48.07.535] [LOG]   Initializing new module helper ...
[07.04.2023 22:48.07.536] [LOG]   Module helper loaded: updatenotification
[07.04.2023 22:48.07.536] [LOG]   No helper found for module: clock.
[07.04.2023 22:48.07.612] [LOG]   Initializing new module helper ...
[07.04.2023 22:48.07.612] [LOG]   Module helper loaded: calendar
[07.04.2023 22:48.07.612] [LOG]   No helper found for module: compliments.
[07.04.2023 22:48.07.614] [LOG]   No helper found for module: weather.
[07.04.2023 22:48.07.629] [LOG]   Initializing new module helper ...
[07.04.2023 22:48.07.629] [LOG]   Module helper loaded: newsfeed
[07.04.2023 22:48.07.630] [LOG]   All module helpers loaded.
[07.04.2023 22:48.07.637] [LOG]   Starting server on port 8080 ...
[07.04.2023 22:48.07.639] [WARN]  You're using a full whitelist configuration to allow for all IPs
[07.04.2023 22:48.07.642] [LOG]   Server started ...
[07.04.2023 22:48.07.643] [LOG]   Connecting socket for: updatenotification
[07.04.2023 22:48.07.643] [LOG]   Starting module helper: updatenotification
[07.04.2023 22:48.07.643] [LOG]   Connecting socket for: calendar
[07.04.2023 22:48.07.643] [LOG]   Starting node helper for: calendar
[07.04.2023 22:48.07.643] [LOG]   Connecting socket for: newsfeed
[07.04.2023 22:48.07.643] [LOG]   Starting node helper for: newsfeed
[07.04.2023 22:48.07.644] [LOG]   Sockets connected & modules started ...
[98:0407/224807.695924:ERROR:bus.cc(399)] Failed to connect to the bus: Failed to connect to socket /run/dbus/system_bus_socket: Permission denied
[98:0407/224807.696008:ERROR:bus.cc(399)] Failed to connect to the bus: Failed to connect to socket /run/dbus/system_bus_socket: Permission denied
[98:0407/224807.702155:ERROR:bus.cc(399)] Failed to connect to the bus: Failed to connect to socket /var/run/dbus/system_bus_socket: Permission denied
[98:0407/224807.702261:ERROR:bus.cc(399)] Failed to connect to the bus: Failed to connect to socket /var/run/dbus/system_bus_socket: Permission denied
[07.04.2023 22:48.07.760] [LOG]   Launching application.
[98:0407/224807.775032:ERROR:bus.cc(399)] Failed to connect to the bus: Failed to connect to socket /var/run/dbus/system_bus_socket: Permission denied
[122:0407/224807.792259:ERROR:gpu_memory_buffer_support_x11.cc(49)] dri3 extension not supported.
[07.04.2023 22:48.08.782] [LOG]   Create new calendarfetcher for url: http://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics - Interval: 300000
[07.04.2023 22:48.08.804] [LOG]   Create new newsfetcher for url: https://rss.nytimes.com/services/xml/rss/nyt/HomePage.xml - Interval: 300000
[07.04.2023 22:48.08.837] [INFO]  Checking git for module: MagicMirror
[07.04.2023 22:48.09.245] [INFO]  Newsfeed-Fetcher: Broadcasting 27 items.
[07.04.2023 22:48.09.712] [INFO]  Calendar-Fetcher: Broadcasting 11 events.
```

khassel, 2023-07-01:
fixed with new release `v2.24.0`

PR #3091 (2023-04-12): fix electron not running under windows (after async changes)
fixes #3083 

tested under windows 11 and linux.
commit 979f4ec6 (2023-04-12): fix electron not running under windows (after async changes) (#3091)

fixes #3083 

tested under windows 11 and linux.
closed 2023-07-01 by khassel