MagicMirrorOrg/MagicMirror #3095: Calendar: today's full day events filtered out with default config

ismarslomic, 2023-04-27:
### Use case describing the bug
WHEN fetching events from external iCalendar with default config
AND the fetched event's `start` and `end` timestamp is todays date without the time (i.e `20230427`)
THEN this event is filtered out without being displayed

### Root cause
The bug is in the [line 470](https://github.com/MichMich/MagicMirror/blob/master/modules/default/calendar/calendarutils.js#L470) in the `calendarutils.js` since `endDate` will have time `00.00:00` which is <= `today` (which also has time 00:00:00).

### Proposed solution
Move the code block at [line 485](https://github.com/MichMich/MagicMirror/blob/master/modules/default/calendar/calendarutils.js#L485) before the filtering described above, to set the time for `endDate` to end of the day.

I will create PR with this suggestion.

### Example
#### config.js
```javascript
{
  module: "calendar",
  header: "My calendar",
  position: "top_left",
  config: {
    calendars: [
      {
        url: "http://mycompany.com/calendar.ics"
      }
    ]
  }
}
```

#### iCalendar event
```
BEGIN:VEVENT
UID:a950cb3c-b067-422a-94c6-5399ffb02b7b
SEQUENCE:0
DTSTAMP:20230427T092247
DTSTART;VALUE=DATE:20230427
DTEND;VALUE=DATE:20230427
X-MICROSOFT-CDO-ALLDAYEVENT:TRUE
X-MICROSOFT-MSNCALENDAR-ALLDAYEVENT:TRUE
SUMMARY:My event
LOCATION:My location
DESCRIPTION:My description
END:VEVENT
```

sdetweil, 2023-04-27:
no.  without time, the event is full day, and it has started already for today. we show start times, but not in the past.

ismarslomic, 2023-04-27:
Hmm, interesting. So default config for the Calendar module is to only display upcoming events, which means filtering out _ongoing_ and _past_ events:
a) for non full days: endDate < now
b) for full days: endDate <= start of the day today

Is that correct?

ismarslomic, 2023-04-27:
Actually the current behaviour with default config is:

a) for non full day events: display _upcoming_ and _ongoing_ events
b) for full day events: display upcoming events

This is quite confusing and inconsistent, in my opinion at least. The module should handle consistently events no matter whether they are full day or not.

sdetweil, 2023-04-27:
we never did ongoing, then someone added it.
maybe it's a bug

we never display  past events, but do broadcast them in case other modules want to do something (the cx? modules for example)

you are welcome to submit a PR with a potential fix, ( and considers showOngoing: false)

ismarslomic, 2023-04-27:
Could you please describe what is the definition of "past events"? 

Does definition depend on comparison of 
a) `startDate` with `now`, or 
b)`endDate` compared with `now`

It seems for me that it is b) that is used for both, non full day and full day events. The only issue is that for full day dates, the endDate has time 00:00:00 which I find strange. Basically you will only display upcoming full day events, while for non full events you upcoming and ongoing are displayed.

sdetweil, 2023-04-27:
@ismarslomic 'past' is started before 'now'

as I said, up until 2021 MagicMirror didn't display ongoing events, only starting after 'now'.  someone submitted a PR for that 

this is all volunteer effort... I work mostly on the parser and getting the dates and times right. (rrule is the bane now) 
someone else works on the displaying of the data

ismarslomic, 2023-04-27:
Thanks for the clarification. I totally understand how the codebase is maintained, and I work on submitting an PR and contribute back to the community!

But I got little bit confused about the definition of "past events" based on feedbacks in this Issue, compared with the current logic I see in the source. So I wanted to understand which of these two are right so I can either make a bug fix for _non full events_, or for _full day events_. 

This is an functional clarification of the default behaviour of the Calendar.

My personal wish is to display upcoming and ongoing events in the calendar, no matter full day or not. So my PR will be based on my own needs. 

Does that sound reasonable?

sdetweil, 2023-04-27:
well, shouldn't be bases on YOUR needs, but solving the generic problem
making the behavior the same
and matching that to the config parm that disables the default behavior..

ismarslomic, 2023-04-27:
> making the behavior the same
Exactly, and that is my goal when submitting the PR. But we need to take a decision if ongoing events (full day & non full day) shall be displayed or not, by default. Who among the contributors would be most suitable to answer this question?

sdetweil, 2023-04-27:
@ismarslomic   

>if ongoing events (full day & non full day) shall be displayed or not, by default.

already done

the config parm 
hideOngoing , defaults top false (do not hide)

ismarslomic, 2023-04-27:
Alright, so then I will submit a bug fix to display the ongoing full day events that currently are hidden.

sdetweil, 2023-05-05:
will do as soon as I can

khassel, 2023-07-01:
fixed with new release `v2.24.0`

commit 83315f1f (2023-05-15): fix: don't filter out ongoing full day events in Calendar module (#3095) (#3096)

Fixes #3095.

Param `hideOngoing` is by default set to `false`, but the event
filtering handles `full day` & `non-full day` events inconsistently. For
`non-full day` _ongoing_ and _upcoming_ events are returned, while for
`full day` only _upcoming_ events where returned.
closed 2023-07-01 by khassel