MagicMirrorOrg/MagicMirror #3097: iCal results in wrong date

sielnet, 2023-05-03:
I am trying to find out where the problem lies.
The following iCal file gets loaded but the events are one day off.
Is the problem in the calendar module or the file?

BEGIN:VCALENDAR
VERSION:2.0
PRODID:-//www.admidio.org//Admidio4.2.5//DE
CALSCALE:GREGORIAN
METHOD:PUBLISH
X-WR-TIMEZONE:Europe/Berlin
BEGIN:VTIMEZONE
TZID:Europe/Berlin
X-LIC-LOCATION:Europe/Berlin
BEGIN:STANDARD
DTSTART:19701025T030000
TZOFFSETFROM:+0200
TZOFFSETTO:+0100
TZNAME:CET
RRULE:FREQ=YEARLY;BYDAY=-1SU;BYMONTH=10
END:STANDARD
BEGIN:DAYLIGHT
DTSTART:19700329T020000
TZOFFSETFROM:+0100
TZOFFSETTO:+0200
TZNAME:CEST
RRULE:FREQ=YEARLY;BYDAY=-1SU;BYMONTH=3
END:DAYLIGHT
END:VTIMEZONE
BEGIN:VEVENT
CREATED:20230502T154941
LAST-MODIFIED:20230503T202824
UID:20230502T154941+2@mein.sielnet.de
SUMMARY:SielNet Supportzeit
DESCRIPTION:Internetsupport\n
DTSTAMP:20230503T203933
LOCATION:
DTSTART;TZID=Europe/Berlin:20230502T180000
DTEND;TZID=Europe/Berlin:20230502T200000
END:VEVENT
BEGIN:VEVENT
CREATED:20230503T203922
LAST-MODIFIED:
UID:20230503T203922+2@mein.sielnet.de
SUMMARY:Testtermin2
DESCRIPTION:
DTSTAMP:20230503T203933
LOCATION:
DTSTART;TZID=Europe/Berlin:20230503T110000
DTEND;TZID=Europe/Berlin:20230503T120000
END:VEVENT
END:VCALENDAR

sdetweil, 2023-05-03:
I don't see that 
attached is the log of the processing (see this is config.js)
```
	logLevel: ["INFO", "LOG", "WARN", "ERROR","DEBUG"], // Add "DEBUG" for even more logging
```
both with my system set to my timezone and the calendar event timezone
and a screenshot with belin timezone set
I had to change the start/end date on both as they were in the past 

so, moved the day +2 in both start/end in both events

[image]
[somefile.txt](https://github.com/MichMich/MagicMirror/files/11391474/somefile.txt)

sielnet, 2023-05-04:
I will debug it later.

here is a screenshot from "post timestamp"

[image]

khassel, 2023-07-02:
@sdetweil is this something to be investigated or can we close this?

sdetweil, 2023-11-01:
@sielnet 
>I disabled the displaying of past events to work around the error. 

we don't display past events, ever.. w might display a repeating event scheduled to start in the past, but only its current and future instances..

scelfo, 2023-11-17:
I am seeing this problem as well. Full day calendar events appear on the wrong day. I have only tried using this public holiday calendar which only has full day events so I don't know if I have the same problem with shorter events. Here is a relevant snippet from https://www.google.com/calendar/ical/en.usa%23holiday@group.v.calendar.google.com/public/basic.ics
```
BEGIN:VEVENT
DTSTART;VALUE=DATE:20231123
DTEND;VALUE=DATE:20231124
DTSTAMP:20231117T230821Z
UID:20231123_6b64ndbva1c6ptufj345j4opno@google.com
CLASS:PUBLIC
CREATED:20231003T144050Z
DESCRIPTION:Public holiday
LAST-MODIFIED:20231003T144050Z
SEQUENCE:0
STATUS:CONFIRMED
SUMMARY:Thanksgiving Day
TRANSP:TRANSPARENT
END:VEVENT
```

But it is being rendered as "Wednesday". Also notice that Christmas is being shows as Dec 24th.
[image]

Here is my calendar config:
```
{
  module: "calendar",
  header: "US Holidays",
  position: "top_left",
  config: {
    maximumEntries: 5,
    calendars: [{
      fetchInterval: 7 * 24 * 60 * 60 * 1000,
      symbol: "calendar-check",
      url: "https://www.google.com/calendar/ical/en.usa%23holiday@group.v.calendar.google.com/public/basic.ics",
    }]
  }
}
```

sdetweil, 2023-11-18:
@scelfo what time zone is your system set to?

scelfo, 2023-11-18:
America/Denver

sdetweil, 2023-11-18:
I used both your individual event and the full ics like and both produced the correct info .  both in my timezone (chicago) and yours.  (denver) 

[image]

sdetweil, 2023-11-18:
@scelfo can you do this 
edit config.js and change thelogLevel like this
```
logLevel: ["INFO", "LOG", "WARN", "ERROR","DEBUG"], 
```
then stop all other MM.s 
pm2 stop all 
or whatever

then , with the one event ics URL
```
cd ~/MagicMirror
npm start >somefile.txt 2>&1
```
wait til the UI is up
then ctrl-q 

then examine the somefile.text 
search for "parsed data={"   

and post 20-50 lines. 
the data would look like this
```
[18.11.2023 08:49.57.738] [DEBUG] Processing entry...
[18.11.2023 08:49.57.740] [DEBUG] Event:
{"type":"VEVENT","params":[],"start":"2023-11-23T06:00:00.000Z","datetype":"date","end":"2023-11-24T06:00:00.000Z","dtstamp":"2023-11-17T23:08:21.000Z","uid":"20231123_6b64ndbva1c6ptufj345j4opno@google.com","class":"PUBLIC","created":"2023-10-03T14:40:50.000Z","description":"Public holiday","lastmodified":"2023-10-03T14:40:50.000Z","sequence":"0","status":"CONFIRMED","summary":"Thanksgiving Day","transparency":"TRANSPARENT"}
[18.11.2023 08:49.57.740] [DEBUG] start: Thu Nov 23 2023 00:00:00 GMT-0600 (Central Standard Time)
[18.11.2023 08:49.57.741] [DEBUG] end:: Fri Nov 24 2023 00:00:00 GMT-0600 (Central Standard Time)
[18.11.2023 08:49.57.741] [DEBUG] duration: 86400000
[18.11.2023 08:49.57.741] [DEBUG] title: Thanksgiving Day
[18.11.2023 08:49.57.742] [DEBUG]  if no tz, guess based on now
[18.11.2023 08:49.57.757] [DEBUG] initial tz=America/Chicago
[18.11.2023 08:49.57.757] [DEBUG] corrected tz=America/Chicago
[18.11.2023 08:49.57.758] [DEBUG] start date/time=Thu Nov 23 2023 00:00:00 GMT-0600 (Central Standard Time)
[18.11.2023 08:49.57.760] [DEBUG] start offset=-360
[18.11.2023 08:49.57.760] [DEBUG] start date/time w tz =Thu Nov 23 2023 00:00:00 GMT-0600 (Central Standard Time)
[18.11.2023 08:49.57.760] [DEBUG] event date=Thu Nov 23 2023 00:00:00 GMT-0600 (Central Standard Time)
[18.11.2023 08:49.57.760] [DEBUG] event offset=-360 hour=0 event date=Thu Nov 23 2023 00:00:00 GMT-0600 (Central Standard Time)
[18.11.2023 08:49.57.760] [DEBUG] adjustHours=0
[18.11.2023 08:49.57.761] [DEBUG] Processing entry...
```

scelfo, 2023-11-18:
@sdetweil thanks for looking and the detailed instructions. I can do that.

I'm running MM via docker inside kubernetes (k3s) but I'll adapt your instructions and get what you're looking for. I'll reply back soon.

sdetweil, 2023-11-18:
I bet the docker container system date is wrong. all the date calcs are done on the mm server side

scelfo, 2023-11-18:
```
8.11.2023 16:12.07.224] [DEBUG] Processing entry...
[18.11.2023 16:12.07.224] [DEBUG] Event:
{"type":"VEVENT","params":[],"start":"2023-11-23T00:00:00.000Z","datetype":"date","end":"2023-11-24T00:00:00.000Z","dtstamp":"2023-11-18T16:12:07.000Z","uid":"20231123_6b64ndbva1c6ptufj345j4opno@google.com","class":"PUBLIC","created":"2023-10-03T14:40:50.000Z","description":"Public holiday","lastmodified":"2023-10-03T14:40:50.000Z","sequence":"0","status":"CONFIRMED","summary":"Thanksgiving Day","transparency":"TRANSPARENT","method":"PUBLISH"}
[18.11.2023 16:12.07.224] [DEBUG] start: Thu Nov 23 2023 00:00:00 GMT+0000 (Coordinated Universal Time)
[18.11.2023 16:12.07.224] [DEBUG] end:: Fri Nov 24 2023 00:00:00 GMT+0000 (Coordinated Universal Time)
[18.11.2023 16:12.07.224] [DEBUG] duration: 86400000
[18.11.2023 16:12.07.224] [DEBUG] title: Thanksgiving Day
[18.11.2023 16:12.07.225] [DEBUG]  if no tz, guess based on now
[18.11.2023 16:12.07.272] [DEBUG] initial tz=Africa/Abidjan
[18.11.2023 16:12.07.272] [DEBUG] corrected tz=Africa/Abidjan
[18.11.2023 16:12.07.272] [DEBUG] start date/time=Thu Nov 23 2023 00:00:00 GMT+0000 (Coordinated Universal Time)
[18.11.2023 16:12.07.273] [DEBUG] start offset=0
[18.11.2023 16:12.07.274] [DEBUG] start date/time w tz =Thu Nov 23 2023 00:00:00 GMT+0000 (Coordinated Universal Time)
[18.11.2023 16:12.07.274] [DEBUG] event date=Thu Nov 23 2023 00:00:00 GMT+0000 (Coordinated Universal Time)
[18.11.2023 16:12.07.274] [DEBUG] event offset=0 hour=0 event date=Thu Nov 23 2023 00:00:00 GMT+0000 (Coordinated Universal Time)
[18.11.2023 16:12.07.274] [DEBUG] adjustHours=0
```

Definitely a timezone problem in docker. Do you know where npm gets the timezone configuration?

I am setting the correct /etc/localtime file for the docker container. Here's some examples of commands executed in the pod:
```
# k3s kubectl exec pod/magic-mirror-6f89d6cc7d-tfpw6 -- date
Sat Nov 18 09:16:18 MST 2023
# k3s kubectl exec pod/magic-mirror-6f89d6cc7d-tfpw6 -- tail -n 1 /etc/localtime
MST7MDT,M3.2.0,M11.1.0
# k3s kubectl exec pod/magic-mirror-6f89d6cc7d-tfpw6 -- cat /etc/timezone
Etc/UTC
```

So it looks like /etc/timezone isn't set up correctly, does npm use that? Or maybe it uses a TZ environment variable?

sdetweil, 2023-11-18:
I do not know timezone in docker container

scelfo, 2023-11-18:
My question isn't about docker, it's about how npm gets the timezone. Either way, I know what the problem is and I'll track down a fix and if necessary propose a patch to magic mirror.

sdetweil, 2023-11-18:
each container is its own little system which can have a separate time zone unrelated to what the system time zone is that docker is running on

scelfo, 2023-11-18:
I updated my kubernetes config to set the TZ environment variable inside docker to be "America/Denver" and now everything works.

sdetweil, 2023-11-19:
awesome. can we close this now?

scelfo, 2023-11-19:
Reading back through this bug, my root cause seems to be different than the original discussion in the bug. I have no more concerns.

closed 2023-11-19 by sdetweil