MagicMirrorOrg/MagicMirror #3098: Recurring events do not respect current DST if created before changing to or from current DST

Ryanclbryant, 2023-05-04:
I have several recurring meetings which are being displayed with an incorrect start/end time using the default calendar module. I'm using vdirsyncer to pull my calendar down from Google Calendar into a local file on my rpi 4 running:
`Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`

One sample of a problematic event:
```
BEGIN:VEVENT
DTSTART;TZID=America/Edmonton:20230504T140000
DTEND;TZID=America/Edmonton:20230504T150000
DTSTAMP:20230502T002709Z
UID:*******@google.com
X-GOOGLE-CONFERENCE:***********
RECURRENCE-ID;TZID=America/Toronto:20230504T160000
CLASS:PRIVATE
CREATED:20230309T052934Z
DESCRIPTION:**************
LAST-MODIFIED:20230502T002709Z
LOCATION:
SEQUENCE:1
STATUS:CONFIRMED
SUMMARY:**************
TRANSP:OPAQUE
END:VEVENT
```
This has the correct start and end time of my event in my local timezone; 2023-05-04 14:00 - 2023-05-04 15:00. The calendar module is displaying the start/end of this event as 2023-05-04 13:00 - 2023-05-04 14:00 incorrectly. 

When checking the debug for Magic mirror, appears to be adjusting the start time back 1 hour due to daylight savings, even though the start/end time has already accommodated for this.
```
[04.05.2023 09:40.52.755] [DEBUG] Processing entry...
[04.05.2023 09:40.52.757] [DEBUG] Event:
{"type":"VEVENT","params":[],"start":"2023-01-12T21:00:00.000Z","datetype":"date-time","end":"2023-01-12T22:00:00.000Z","rrule":{"_cache":{"all":false,"before":[],"after":[],"between":[]},"origOptions":{"tzid":"America/Toronto","dtstart":"2023-01-12T21:00:00.000Z","freq":2,"wkst":{"weekday":6},"interval":2,"byweekday":[{"weekday":3}]},"options":{"freq":2,"dtstart":"2023-01-12T21:00:00.000Z","interval":2,"wkst":6,"count":null,"until":null,"tzid":"America/Toronto","bysetpos":null,"bymonth":null,"bymonthday":[],"bynmonthday":[],"byyearday":null,"byweekno":null,"byweekday":[3],"bynweekday":null,"byhour":[21],"byminute":[0],"bysecond":[0],"byeaster":null}},"dtstamp":"2023-05-02T00:26:59.000Z","uid":"*********@google.com","GOOGLE-CONFERENCE":"*******","class":"PRIVATE","created":"2023-03-09T05:29:34.000Z","description":"********","lastmodified":"2023-05-02T00:26:59.000Z","location":"","sequence":"1","status":"CONFIRMED","summary":"**********","transparency":"OPAQUE","recurrences":{"2023-01-26":{"type":"VEVENT","params":[],"start":"2023-01-26T21:00:00.000Z","datetype":"date-time","end":"2023-01-26T22:00:00.000Z","dtstamp":"2023-05-02T00:26:59.000Z","uid":"*********@google.com","GOOGLE-CONFERENCE":"*********","recurrenceid":"2023-01-26T21:00:00.000Z","class":"PRIVATE","created":"2023-03-09T05:29:34.000Z","description":"*********","lastmodified":"2023-05-02T00:26:59.000Z","location":"","sequence":"1","status":"CONFIRMED","summary":"*********","transparency":"OPAQUE"},"2023-02-23":{"type":"VEVENT","params":[],"start":"2023-02-23T21:00:00.000Z","datetype":"date-time","end":"2023-02-23T22:00:00.000Z","dtstamp":"2023-05-02T00:26:59.000Z","uid":"*********@google.com","GOOGLE-CONFERENCE":"*********","recurrenceid":"2023-02-23T21:00:00.000Z","class":"PRIVATE","created":"2023-03-09T05:29:34.000Z","description":"*********","lastmodified":"2023-05-02T00:26:59.000Z","location":"","sequence":"1","status":"CONFIRMED","summary":"*********","transparency":"OPAQUE"},"2023-03-09":{"type":"VEVENT","params":[],"start":"2023-03-09T21:00:00.000Z","datetype":"date-time","end":"2023-03-09T22:00:00.000Z","dtstamp":"2023-05-02T00:26:59.000Z","uid":"*********@google.com","GOOGLE-CONFERENCE":"*********","recurrenceid":"2023-03-09T21:00:00.000Z","class":"PRIVATE","created":"2023-03-09T05:29:34.000Z","description":"*********","lastmodified":"2023-05-02T00:26:59.000Z","location":"","sequence":"1","status":"CONFIRMED","summary":"*********","transparency":"OPAQUE"},"2023-04-06":{"type":"VEVENT","params":[],"start":"2023-04-06T20:00:00.000Z","datetype":"date-time","end":"2023-04-06T21:00:00.000Z","dtstamp":"2023-05-02T00:26:59.000Z","uid":"*********@google.com","GOOGLE-CONFERENCE":"*********","recurrenceid":"2023-04-06T20:00:00.000Z","class":"PRIVATE","created":"2023-03-09T05:29:34.000Z","description":"*********","lastmodified":"2023-05-02T00:26:59.000Z","location":"","sequence":"1","status":"CONFIRMED","summary":"*********","transparency":"OPAQUE"},"2023-04-20":{"type":"VEVENT","params":[],"start":"2023-04-20T20:00:00.000Z","datetype":"date-time","end":"2023-04-20T21:00:00.000Z","dtstamp":"2023-05-02T00:26:59.000Z","uid":"*********@google.com","GOOGLE-CONFERENCE":"*********","recurrenceid":"2023-04-20T20:00:00.000Z","class":"PRIVATE","created":"2023-03-09T05:29:34.000Z","description":"*********","lastmodified":"2023-05-02T00:26:59.000Z","location":"","sequence":"1","status":"CONFIRMED","summary":"*********","transparency":"OPAQUE"},"2023-05-04":{"type":"VEVENT","params":[],"start":"2023-05-04T20:00:00.000Z","datetype":"date-time","end":"2023-05-04T21:00:00.000Z","dtstamp":"2023-05-02T00:27:09.000Z","uid":"*********@google.com","GOOGLE-CONFERENCE":"*********","recurrenceid":"2023-05-04T20:00:00.000Z","class":"PRIVATE","created":"2023-03-09T05:29:34.000Z","description":"*********","lastmodified":"2023-05-02T00:27:09.000Z","location":"","sequence":"1","status":"CONFIRMED","summary":"*********","transparency":"OPAQUE"},"2023-06-29":{"type":"VEVENT","params":[],"start":"2023-06-29T20:00:00.000Z","datetype":"date-time","end":"2023-06-29T21:00:00.000Z","dtstamp":"2023-05-02T00:27:09.000Z","uid":"*********@google.com","GOOGLE-CONFERENCE":"*********","recurrenceid":"2023-06-29T20:00:00.000Z","class":"PRIVATE","created":"2023-03-09T05:29:34.000Z","description":"*********","lastmodified":"2023-05-02T00:27:09.000Z","location":"","sequence":"1","status":"CONFIRMED","summary":"*********","transparency":"OPAQUE"},"2023-07-27":{"type":"VEVENT","params":[],"start":"2023-07-27T20:00:00.000Z","datetype":"date-time","end":"2023-07-27T21:00:00.000Z","dtstamp":"2023-05-02T00:27:09.000Z","uid":"*********@google.com","GOOGLE-CONFERENCE":"*********","recurrenceid":"2023-07-27T20:00:00.000Z","class":"PRIVATE","created":"2023-03-09T05:29:34.000Z","description":"*********","lastmodified":"2023-05-02T00:27:09.000Z","location":"","sequence":"1","status":"CONFIRMED","summary":"*********","transparency":"OPAQUE"},"2023-12-28":{"type":"VEVENT","params":[],"start":"2023-12-28T21:00:00.000Z","datetype":"date-time","end":"2023-12-28T22:00:00.000Z","dtstamp":"2023-05-02T00:27:09.000Z","uid":"*********@google.com","GOOGLE-CONFERENCE":"*********","recurrenceid":"2023-12-28T21:00:00.000Z","class":"PRIVATE","created":"2023-03-09T05:29:34.000Z","description":"*********","lastmodified":"2023-05-02T00:27:09.000Z","location":"","sequence":"1","status":"CONFIRMED","summary":"*********","transparency":"OPAQUE"}}}
[04.05.2023 09:40.52.759] [DEBUG] title: *******************
[04.05.2023 09:40.52.760] [DEBUG] Search for recurring events between: Sun Mar 05 2023 00:00:00 GMT-0700 (Mountain Standard Time) and Sun Jul 02 2023 23:59:59 GMT-0600 (Mountain Daylight Time)
[04.05.2023 09:40.52.768] [DEBUG] Title: *******************, with dates: ["2023-03-09T19:00:00.000Z","2023-03-23T19:00:00.000Z","2023-04-06T19:00:00.000Z","2023-04-20T19:00:00.000Z","2023-05-04T19:00:00.000Z","2023-05-18T19:00:00.000Z","2023-06-01T19:00:00.000Z","2023-06-15T19:00:00.000Z","2023-06-29T19:00:00.000Z"]
[04.05.2023 09:40.52.768] [DEBUG] event.recurrences: [object Object]
...
[04.05.2023 09:58.37.889] [DEBUG]  recurring date is Thu May 04 2023 13:00:00 GMT-0600 (Mountain Daylight Time) offset is 360
[04.05.2023 09:58.37.890] [DEBUG]  recurring date is Thu May 04 2023 13:00:00 GMT-0600 (Mountain Daylight Time) offset is 6 Hour is 13
[04.05.2023 09:58.37.890] [DEBUG] Corrected startDate: Thu May 04 2023 13:00:00 GMT-0600 (Mountain Daylight Time)
[04.05.2023 09:58.37.890] [DEBUG] initial tz=America/Toronto
[04.05.2023 09:58.37.891] [DEBUG] corrected tz=America/Toronto
[04.05.2023 09:58.37.891] [DEBUG] start date/time=Thu Jan 12 2023 14:00:00 GMT-0700 (Mountain Standard Time)
[04.05.2023 09:58.37.892] [DEBUG] start offset=-300
[04.05.2023 09:58.37.893] [DEBUG] start date/time w tz =Thu Jan 12 2023 14:00:00 GMT-0700 (Mountain Standard Time)
[04.05.2023 09:58.37.895] [DEBUG] event date=Thu May 04 2023 13:00:00 GMT-0600 (Mountain Daylight Time)
[04.05.2023 09:58.37.896] [DEBUG] event offset=-240 hour=15 event date=Thu May 04 2023 13:00:00 GMT-0600 (Mountain Daylight Time)
[04.05.2023 09:58.37.897] [DEBUG] offset
[04.05.2023 09:58.37.897] [DEBUG] adjust down 1 hour dst change
[... cut ...]
[04.05.2023 09:40.52.863] [DEBUG] event date=Fri May 12 2023 16:00:00 GMT-0600 (Mountain Daylight Time)
[04.05.2023 09:40.52.863] [DEBUG] event offset=-360 hour=16 event date=Fri May 12 2023 16:00:00 GMT-0600 (Mountain Daylight Time)
[04.05.2023 09:40.52.863] [DEBUG] adjustHours=0
[04.05.2023 09:40.52.863] [DEBUG] duration: 2700000
[04.05.2023 09:40.52.863] [DEBUG] saving event: ***********
```
Here 16:00 is the correct start time, the GMT timestamp of 22:00, and the resulting start date/time of `date/time=Fri May 12 2023 16:00:00 GMT-0600 (Mountain Daylight Time)` are all correct.

sdetweil, 2023-05-04:
can u show the original with the rule

we have a problem with the rule library, which only deals w local time, and the calendar parser which uses tz/utc.

rrule returns the dates mangled
wrong day, right time. or right day wrong time
or right both...  (most of the time)

the edge cases are where the time and the tinezone offset cause overflow

sdetweil, 2023-05-04:
another issue, DST switching

the 1st event with trouble says 
 start date/time w tz =Thu Jan 12 2023 14:00:00 GMT-     SO, pre-DST
but now is post-DST 
sometimes the moment() gets it right, sometimes not.. hard to tell

Ryanclbryant, 2023-05-04:
Original with the rule, you mean the rrule options on this recurring event?
Cut out and formatted the details, but each of those sections are available above:
```
"rrule": {
    "_cache": {
      "all": false,
      "before": [],
      "after": [],
      "between": []
    },
    "origOptions": {
      "tzid": "America/Toronto",
      "dtstart": "2023-01-12T21:00:00.000Z",
      "freq": 2,
      "wkst": {
        "weekday": 6
      },
      "interval": 2,
      "byweekday": [
        {
          "weekday": 3
        }
      ]
    },
    "options": {
      "freq": 2,
      "dtstart": "2023-01-12T21:00:00.000Z",
      "interval": 2,
      "wkst": 6,
      "count": null,
      "until": null,
      "tzid": "America/Toronto",
      "bysetpos": null,
      "bymonth": null,
      "bymonthday": [],
      "bynmonthday": [],
      "byyearday": null,
      "byweekno": null,
      "byweekday": [
        3
      ],
      "bynweekday": null,
      "byhour": [
        21
      ],
      "byminute": [
        0
      ],
      "bysecond": [
        0
      ],
      "byeaster": null
    }
```
If I've completely misunderstood please just clarify which rule portion you're looking for and I can grab it!

The Jan 12 2023 dtstart is when this recurring meeting was originally started -- possibly in the process of going from UTC to local and back, something is accommodating for the fact that DST changed and subtracted an hour, and then calendar does it again deliberately to acknowledge the difference in offset?

For good measure, here is my config.js as it pertains to the calendar module:

```
 {
                        module: "calendar",
                        header: "Calendar",
                        position: "top_left",
                        config: {
                                maximumEntries: 2000,
                                broadcastPastEvents: true,
                                maximumNumberOfDays: 60,
                                calendars: [
                                        {
                                                url: "***********.ics",
                                                name: "family_calendar",
                                                color: "yellow"
                                        },
                                        {
                                                url: "*********.ics",
                                                name: "personal_calendar",
                                                color: "blue"
                                        },
                                        {
                                                url: "***********.ics",
                                                name: "work_calendar",
                                                broadcastPastEvents: false,
                                                color: "purple"
                                        },
                                        {
                                                url: "**********.ics",
                                                name: "new_family_calendar",
                                                color: "red"
                                        }
                                ],
                                excludedEvents: [*****],
                                showLocation: true,
                                fetchInterval: 1200000
                        }
                }
```

sdetweil, 2023-05-04:
no, the calendar event with the RRULE in it..  start on Jan 12 2023

you know you don't need to rsync unless u are trying to minimize network traffic and accept the delays in updates

Ryanclbryant, 2023-05-04:
Ah! Yes here is the original event record:
```
BEGIN:VEVENT
DTSTART;TZID=America/Toronto:20230112T160000
DTEND;TZID=America/Toronto:20230112T170000
RRULE:FREQ=WEEKLY;WKST=SU;INTERVAL=2;BYDAY=TH
DTSTAMP:20230502T002659Z
UID:***********@google.com
X-GOOGLE-CONFERENCE:***********
CLASS:PRIVATE
CREATED:20230309T052934Z
DESCRIPTION:***********
LAST-MODIFIED:20230502T002659Z
LOCATION:
SEQUENCE:1
STATUS:CONFIRMED
SUMMARY:********
TRANSP:OPAQUE
END:VEVENT
```
Once upon a time I had a good reason for copying everything locally, can't remember what it was now, and it hasn't annoyed me enough yet to go through changing it

sdetweil, 2024-09-29:
i believe if you take off the 
BYDAY=TH
it will be ok

sdetweil, 2024-10-07:
see this.. waiting on user test for submitting PR here and node-ical
all testcases in both run successfully..

here is a test version of the fixes for all kinds of date problems.

best to make a new folder and git clone there
```
git  clone https://github.com/sdetweil/MagicMirror
git checkout fixcaldates
npm run install-mm
```
copy your config.js and custom.css from the prior folder
and the non-default modules you have installed..

this ONLY changes the default calendar
but DOES ship an updated node-ical library too

if you need to fall back, just rename the folders around again so that
your original is called MagicMirror

all the testcases for node-ical and MagicMirror execute successfully.

the 'BIG' change here is to get the local NON-TZ dates for the
rrule.between()

all the checking and conversion code is commented out or not used
the node-ical fixes are for excluded dates (exdate) values being adjusted for DST/STD time.. waiting to submit that PR

one fix in calendar.js for checking if a past date was too far back,
but it never checked to see IF the event date was in the past.. (before today) so it chopped off too many

and one change in calendarfetcher.js to put out a better diagnostic message of the parsed data.. (exdate was excluded cause JSON stringify couldn't convert the complex structure)

sdetweil, 2024-10-17:
here is an updated test version of the fixes for all kinds of calendar date problems.

NOTE: the changed branch name
NOTE: this used the node-cal@0.19.0 library UNCHANGED

best to make a new folder and git clone there

git clone https://github.com/sdetweil/MagicMirror
cd MagicMirror
git checkout fixcaldates2 // <------ note this is a changed branch name
npm run install-mm
copy your config.js and custom.css from the prior folder
and the non-default modules you have installed…

this ONLY changes the default calendar
but DOES ship an updated node-ical library too

if you need to fall back, just rename the folders around again so that
your original is called MagicMirror

all the testcases for node-ical and MagicMirror execute successfully.

the ‘BIG’ change here is to get the local NON-TZ dates for the
rrule.between()

all the checking and conversion code is commented out or not used
the node-ical fixes are for excluded dates (exdate) values being adjusted for DST/STD time… waiting to submit that PR

one fix in calendar.js for checking if a past date was too far back,
but it never checked to see IF the event date was in the past… (before today) so it chopped off too many

and one change in calendarfetcher.js to put out a better diagnostic message of the parsed data… (exdate was excluded cause JSON stringify couldn’t convert the complex structure)

I added the tests you all have documented

please re-pull and checkout the new branch (I deleted the old branch)
and npm run install-mm again

plebcity, 2025-05-31:
This problem still isn't fixed for me:
```
[2025-05-31 11:45:54.434] [DEBUG] start: Tue Mar 11 2025 09:00:00 GMT+0100 (Central European Standard Time) 
[2025-05-31 11:45:54.434] [DEBUG] end:: Tue Mar 11 2025 09:15:00 GMT+0100 (Central European Standard Time) 
[2025-05-31 11:45:54.434] [DEBUG] duration: 900000 
[2025-05-31 11:45:54.434] [DEBUG] title: xxxxxxxxxxxxxxxxxxx
[2025-05-31 11:45:54.435] [DEBUG] Search for recurring events between: Sat May 31 2025 11:45:53 GMT+0200 (Central European Summer Time) and Mon Jun 02 2025 23:59:59 GMT+0200 (Central European Summer Time) 
[2025-05-31 11:45:54.435] [DEBUG] RRule: DTSTART;TZID=Europe/Brussels:20250311T090000
RRULE:FREQ=WEEKLY;BYDAY=FR,MO,TH,TU,WE 
[2025-05-31 11:45:54.435] [DEBUG] Title: xxxxxxxxxxxxxxxx, with dates: ["2025-06-02T09:00:00.000Z","2025-06-03T09:00:00.000Z"] 
[2025-05-31 11:45:54.435] [DEBUG] Rule has byweekday, checking for correction 
[2025-05-31 11:45:54.435] [DEBUG] initial tz=Europe/Brussels 
[2025-05-31 11:45:54.435] [DEBUG] corrected tz=Europe/Brussels 
[2025-05-31 11:45:54.435] [DEBUG] start date/time=Tue Mar 11 2025 09:00:00 GMT+0100 (Central European Standard Time) 
[2025-05-31 11:45:54.435] [DEBUG] start offset=60 
[2025-05-31 11:45:54.435] [DEBUG] start date/time w tz =Tue Mar 11 2025 09:00:00 GMT+0100 (Central European Standard Time) 
[2025-05-31 11:45:54.435] [DEBUG] event date=Mon Jun 02 2025 11:00:00 GMT+0200 (Central European Summer Time) 
[2025-05-31 11:45:54.435] [DEBUG] event offset=120 hour=11 event date=Mon Jun 02 2025 11:00:00 GMT+0200 (Central European Summer Time) 
[2025-05-31 11:45:54.435] [DEBUG] offset 
[2025-05-31 11:45:54.435] [DEBUG] adjust down 1 hour dst change 
[2025-05-31 11:45:54.435] [DEBUG] adjustHours=-1 
[2025-05-31 11:45:54.435] [DEBUG] Applying timezone adjustment hours=-1 to Mon Jun 02 2025 11:00:00 GMT+0200 (Central European Summer Time) 
[2025-05-31 11:45:54.435] [DEBUG] initial tz=Europe/Brussels 
[2025-05-31 11:45:54.435] [DEBUG] corrected tz=Europe/Brussels 
[2025-05-31 11:45:54.435] [DEBUG] start date/time=Tue Mar 11 2025 09:00:00 GMT+0100 (Central European Standard Time) 
[2025-05-31 11:45:54.435] [DEBUG] start offset=60 
[2025-05-31 11:45:54.435] [DEBUG] start date/time w tz =Tue Mar 11 2025 09:00:00 GMT+0100 (Central European Standard Time) 
[2025-05-31 11:45:54.436] [DEBUG] event date=Tue Jun 03 2025 11:00:00 GMT+0200 (Central European Summer Time) 
[2025-05-31 11:45:54.436] [DEBUG] event offset=120 hour=11 event date=Tue Jun 03 2025 11:00:00 GMT+0200 (Central European Summer Time) 
[2025-05-31 11:45:54.436] [DEBUG] offset 
[2025-05-31 11:45:54.436] [DEBUG] adjust down 1 hour dst change 
[2025-05-31 11:45:54.436] [DEBUG] adjustHours=-1 
[2025-05-31 11:45:54.436] [DEBUG] Applying timezone adjustment hours=-1 to Tue Jun 03 2025 11:00:00 GMT+0200 (Central European Summer Time) 
[2025-05-31 11:45:54.436] [DEBUG] event.recurrences: [object Object] 
[2025-05-31 11:45:54.436] [DEBUG] event.start=Tue Mar 11 2025 09:00:00 GMT+0100 (Central European Standard Time) was not included in results from rrule; adding 
[2025-05-31 11:45:54.436] [DEBUG] duration: 900000 
[2025-05-31 11:45:54.436] [DEBUG] duration: 900000 
```

What I think is the problem is that RRule outputs 2 new dates which are correct but the Z needs to be removed from them. If we remove the Z and then convert those dates to the correct timezone it would work and we also don't need to adjust the hours or anyhting like that.

sdetweil, 2025-05-31:
I also need to see the MagicMirror version info

Top of the output of npm start

closed 2025-05-31 by sdetweil