MagicMirrorOrg/MagicMirror #3100: Clock in analog mode doesn't show date

jorgenwahlberg, 2023-05-07:
When using the clock module and setting displayType to "analog" and showDate to true, date is no longer shown. In earlier releases, it did.

This bug can be reproduced by setting up a the module like this:

```
                {
                        module: "clock",
                        position: "top_left",
                        config: {
                                displayType: "analog",
                                analogSize: "360px",
                                analogPlacement: "bottom",
                                analogFace: "face-003",
                                analogShowDate: "top",
                                showDate: true,
                                displaySeconds: true,
                        }
                }
```

When setting displayType to "both" or "digital", the date is shown.

rejas, 2023-05-08:
Thanks for the report (and the PR)! Will discuss it in the PR.

khassel, 2023-07-01:
fixed with new release `v2.24.0`

commit babd22b0 (2023-05-19): Bug fix: adding date to the analog clock if showDate is true (#3101)

Adding date to the clock module when displayType is "analog" and
"showDate" is true. The setting in analogShowDate is respected.

Fixes #3100

---------
closed 2023-07-01 by khassel