MagicMirrorOrg/MagicMirror #3107: Update CodeCov integration

rejas, 2023-05-21:
CodeCov complains:

"❗ Your organization is not using the GitHub App Integration. As a result you may experience degraded service beginning May 15th. Please [install the Github App Integration](https://github.com/apps/codecov) for your organization. [Read more](https://about.codecov.io/blog/codecov-is-updating-its-github-integration/?utm_medium=referral&utm_source=github&utm_content=comment&utm_campaign=pr+comments&utm_term=None)." 

We need to start using https://github.com/apps/codecov instead, which I think can only be configured by @MichMich

khassel, 2023-09-27:
The Codecov integration is useless at the moment because the reports are not correct, see e.g. [discussion in this PR](https://github.com/MichMich/MagicMirror/pull/3206#issuecomment-1735267834).

So as long there is no progress here in updating the integration we should disable the workflow for Codecov.

@MichMich @rejas

khassel, 2023-10-12:
```bash
[2023-10-12T08:31:38.919Z] ['error'] There was an error running the uploader: Error uploading to [https://codecov.io:](https://codecov.io/) Error: There was an error fetching the storage URL during POST: 404 - {'detail': ErrorDetail(string='Unable to locate build via Github Actions API. Please upload with the Codecov repository upload token to resolve issue.', code='not_found')}
Error: Codecov: Failed to properly upload: The process '/home/runner/work/_actions/codecov/codecov-action/v3/dist/codecov' failed with exit code 255
```

as CodeCov now fails with every run I will remove this integration.

rejas, 2023-10-12:
With integration you mean the codecov github action? Yes to that. Would like to keep the npm script for creating that locally for now.

rejas, 2024-01-18:
Closing this for now

closed 2024-01-18 by rejas