MagicMirrorOrg/MagicMirror #3110: Suggestion - calendar module (not sure how to do it)

BKeyport, 2023-05-23:
I'd like to make a small change to the next release, but I have no clue on how to do it. 

Currently, calendar logs by default "Broadcasting: <X> events." 
I'd like to change it to be "Broadcasting: <X> events from <Y>" 
X: Number of events
Y: Name of calendar. 

This would make debugging easier without enabling the debug flag, which *massively* overdisplays what I'm looking for. 

Thanks!

rejas, 2023-05-23:
With "Name of calendar" do you mean the url? That would be fairly easy to implement.

BKeyport, 2023-05-23:
Not exactly. 

```
{
 url: "https://calendar.google.com/calendar/ical/en.usa%23holiday%40group.v.calendar.google.com/public/basic.ics",
 name: "Holdays",
},
```

The "name" here if used, if not, then the URL.

sdetweil, 2023-05-23:
we don't have a name attribute. that is added into config for calendarExt3 and agenda. 

you can see the whole config, so given one known value, you can find the rest.

sdetweil, 2023-05-23:
but.  we already have that function

from the doc
[image]

BKeyport, 2023-05-23:
sam: This is making **logging** better - not **broadcast** purposes. I'm aware It's already there - I already use the name as you can see from my code snip above.

sdetweil, 2023-05-23:
sorry, logging...

BKeyport, 2023-05-24:
```
	this.broadcastEvents = function () {
		Log.info(`Calendar-Fetcher: Broadcasting ${events.length} events.`);
		eventsReceivedCallback(this);
	};
```

this statement.

rejas, 2023-05-24:
We dont have much meta-information there about the calendar so passing name down there involves a lot of code just for logging purpose. Wouldnt the url be sufficient?

BKeyport, 2023-05-24:
The URL would be a fine backup. The name I'm looking for is in the config.js and passed through to the CALENDAR_EVENTS broadcast array in every event as a named item "calendarName", so I know it's stored _somewhere_ in the code, I just can't find it.

sdetweil, 2023-05-24:
the problem is things are all over the place.

calendar.js does and add calendar() which doesn't pass the name
this goes down to the node helper, which doesn't have the name, which starts an async fetcher, which doesn't get the name..

it gets the events back from the util lib, and sends them up to the socketNotificationReceived in calendar.js, but there is no name in the data.
so all we know is there are a bunch of events to broadcast.  

I was about to post the 'simple' code yesterday, but it turned out to need changes in 10 places to pass the name around, just to put it in that message IF it was specified (95% of the time not)

BKeyport, 2023-05-25:
Sam, compared to the raw data put out in debug mode, it's nothing, and it's being logged - if you have access to the log, you also have access to the config.js, no? 

Silly questions, for my own benefit and clarification - 

the function that creates this log item is doing the actual broadcast, right? Why can't we pull the name out of the broadcast string (calendarName), or is the spaghetti code too much for that, even? are the logging points in the wrong spot?

sdetweil, 2023-05-25:
>  the function that creates this log item is doing the actual broadcast, right?

no.. it puts out the message(in calendarfetcher.js, which doesn't know the config.js and doesn't have the name) , 
```
	this.broadcastEvents = function () {
		Log.info(`Calendar-Fetcher: Broadcasting ${events.length} events.`);
		eventsReceivedCallback(this);
	};
```

and then another function (in node_helper.js, which doesn't know the name) sends the events UP
```
broadcastEvents: function (fetcher, identifier) {
		this.sendSocketNotification("CALENDAR_EVENTS", {
			id: identifier,
			url: fetcher.url(),
			events: fetcher.events()
		});
	}
```
 to the calendar.js (only thing that can actually broadcast) 
```
if (notification === "CALENDAR_EVENTS") {
			if (this.hasCalendarURL(payload.url)) {
				this.calendarData[payload.url] = payload.events;
				this.error = null;
				this.loaded = true;

				if (this.config.broadcastEvents) {
					this.broadcastEvents();
				}
			}
```
and IT creates the message that is broadcast 
but the message is log since printed
```
broadcastEvents: function () {
		const eventList = this.createEventList(false);
		for (const event of eventList) {
			event.symbol = this.symbolsForEvent(event);
			event.calendarName = this.calendarNameForUrl(event.url);
			event.color = this.colorForUrl(event.url, false);
			delete event.url;
		}

		this.sendNotification("CALENDAR_EVENTS", eventList);
	}
```	

MAYBE the 'fix' is to move the message to here, at the actual broadcast 
then we could use the function to get the name using the url 
```
	calendarNameForUrl: function (url) {
		return this.getCalendarProperty(url, "name", "");
	},
```

BKeyport, 2023-05-29:
Thanks for the explaination.. I guess the short term solution then would be the URL, with some code streamlining in the future?

sdetweil, 2023-06-01:
so, I tried to implement moving the logging message as proposed above... oops.. calendar.js runs in the browser, SO the message comes out in the developers window, NOT in the terminal window like currently...

rejas, 2023-06-06:
> Thanks for the explaination.. I guess the short term solution then would be the URL, with some code streamlining in the future?

Yes indeed. Do you want to tackle it?

closed 2023-10-01 by khassel