MagicMirrorOrg/MagicMirror #3114: Potential Bug: built-in calendar module customEvents color not working on latest build

ryandrake182, 2023-05-29:
Hello,

Since the latest update, the calendar "`customEvents`" color attribute is no longer working. 
I can see that the customEvents keyword matching is working, as the custom icons are being used for the events. But all events are the default color.

My calendar config within `config.js` is as follows:

```
		module: "calendar",
			header: "Calendar",
			position: "top_left",
			config: {
				maxTitleLength: 46,
				maximumEntries: 14,
				customEvents: [
						{keyword: 'birthday', symbol: 'birthday-cake', color: 'DarkMagenta'}, 
						{keyword: 'dog', symbol: 'paw', color: 'orange'}, 
						{keyword: 'gym', symbol: 'fa-solid fa-dumbbell', color: 'lightcoral'}, 
						{keyword: 'holiday', symbol: 'fas fa-umbrella-beach', color: 'green'}, 
						{keyword: 'day off', symbol: 'fas fa-umbrella-beach', color: 'green'}, 
						{keyword: 'annual leave', symbol: 'fas fa-umbrella-beach', color: 'green'}, 
						{keyword: 'haircut', symbol: 'fas fa-cut', color: 'lightcoral'},
						{keyword: 'flight', symbol: 'fas fa-plane-departure', color: 'green'},
				],
				calendars: [
					{
						url: "redacted"
					}
				]
			}
		},
```

Appreciate any advice, thank you.

sdetweil, 2023-05-29:
you also need 
```
coloredText: true, 
```

ryandrake182, 2023-05-29:
Excellent, that fixed it. Thank you! This wasn't a required attribute in an older build I'm assuming? Either way I'll close this off as resolved.

closed 2023-05-29 by ryandrake182