MagicMirrorOrg/MagicMirror #3115: module translations getTranslations() extension

sdetweil, 2023-06-01:
for module translations, is there any mechanism to select all files in the translations folder (without enumerating them in the getTranslations() call? 

like "translations/*"
means load them all. 

that would make it easy for someone to add a new one, and NOT have to change the code...

bugsounet, 2023-06-10:
yes it would be easier
Reading of the `translations` directory of the module

But... it will be more complicated to determine the default language file to use (actually the first one of `getTranslations`)

rejas, 2023-06-10:
> But... it will be more complicated to determine the default language file to use (actually the first one of `getTranslations`)

So, in case of https://github.com/MichMich/MagicMirror/blob/abe5c08a52ffee6b37c30c1f3160e130f0e966a2/modules/default/alert/alert.js#L28 it would be "bulgaria" which is the default language?

sdetweil, 2023-06-10:
but we have the language setting in config.js

bugsounet, 2023-06-10:
sure, we know that ;)

sdetweil Idea is:

```
getTranslations() { 
  return {
    files: "translations/*"
  }
}
```

For read all files automaticaly files of translations folder of the module
Maybe another function is needed

bugsounet, 2023-06-10:
> but we have the language setting in config.js

but if json file of this translatated part don't exist ?

sdetweil, 2023-06-10:
read all translations files available.. 

this would help a novice user create a new file , put it in folder, and it works.. 
doesn't have to edit  source (which breaks updates)

sdetweil, 2023-06-10:
what happens now?

bugsounet, 2023-06-10:
> read all translations files available..
> 
> this would help a novice user create a new file , put it in folder, and it works.. doesn't have to edit source (which breaks updates)

it's ok, we are talking about the same feature ;)

bugsounet, 2023-06-10:
I had thought this:
  * define a directory with translations
  * create a json file with a new specificity / syntax
```
{
    lang: [language tag]
    default: [boolean]
    translations: [object Translations]
}
```
   * Read automatically this directory and make translations job
   * use `default` lang to apply for the module (fallback)
   * if getTranslations() used, ignore this rule

sdetweil, 2023-06-10:
BUT... that file would belong to the module and locally changed would break updates (even to include the new file)

bugsounet, 2023-06-10:
yes and no, because there is une files by language too and can be added automaticaly

but yes, if you modify one of lang file, it's break update too but a new language file is added automaticaly without modify main core

sdetweil, 2023-06-10:
> but yes, if you modify one of lang file, it's break update too 

trying to add function that helps and does not hurt the user..

bugsounet, 2023-06-11:
I don't know if we can request a reloading of the translation files

otherwise my idea would be to go through the node helper with node fs
 * read the contents of the directory (only json files) 
 * return the result.
 * reload the transcription files with the result option

sdetweil, 2023-06-11:
why reloading?
on startup. read new list of files

closed 2024-10-13 by github-actions[bot]