MagicMirrorOrg/MagicMirror #3116: Broken layout after HDMI off

AleksanderG, 2023-06-01:
If I turn off the screen with xrandr and then restore it after a while, the system hangs with a broken layout. I use MMM-PIR-Sensor and have replaced vcgencmd with xrandr there. My version is 2.21.0. Is there a solution for this?

sdetweil, 2023-06-01:
is the system hung. and u have to reboot?

we don't have any control of the system functions?

what version of pi os are using

lsb_release -a
and uname -a
would help

AleksanderG, 2023-06-01:
After a while the system hangs but at the beginning the whole layout is broken after the hdmi is restored and all elements are located in the upper left corner.

`lsb_release -a`
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
`uname -a`
Linux halmirror 5.15.74-v7+ #1595 SMP Wed Oct 26 11:03:05 BST 2022 armv7l GNU/Linux

Thanks

sdetweil, 2023-06-01:
thanks. MagicMirror is a browser based app. we don't do any visual rendering ourself. I don't think the browser knows that the HDMI port was turned off or on. 

 this sounds like a system or hardware problem

sdetweil, 2023-06-01:
if u ssh in from another system, stop MagicMirror,  so you see the pi desktop

and issue the xrandr comnands on the ssh terminal, does it hang or restore oddly?

AleksanderG, 2023-06-01:
Possible. But also possible that if hdmi is off the screen resizes and the rendering is odd.
Using xrandr without MM works fine.

AleksanderG, 2023-06-01:
I will use `vc4-fkms-v3d` and `vcgencmd`. This works properly. Thanks.

sdetweil, 2023-06-01:
no idea what happens to visible apps when HDMI is turned off.  my assumption over all these years is they don't know, and writes to the 'screen' are ignored.

if you install the latest MagicMirror there is an updated electron browser.

just to test

git clone MMM_url MagicMirror2

that will put the code into the MagicMirror2 folder
cd MagicMirror2
npm install

stop your existing mm
and npm start this one
and try the test

AleksanderG, 2023-06-01:
Thanks. I will try.

sdetweil, 2023-06-01:
note that vcgencmd is removed in the next pi os release

AleksanderG, 2023-06-01:
I know but maybe till I have to update we will have a fix. :)

bugsounet, 2023-06-10:
> note that vcgencmd is removed in the next pi os release

where do you see this ?
I search but I'm not able to find

sdetweil, 2023-06-10:
not shipped as part of ios image

https://forums.raspberrypi.com/viewtopic.php?t=342865

maybe not removed, but not functional.. so effectively removed

bugsounet, 2023-06-10:
yeah... I'm not more convinced than that for the moment because it has always been a hassle to make it work with kms driver (even with buster). on the other hand it has always worked well with fkms
so the future will tell!

khassel, 2023-07-02:
author found a workaround and os discussions are out of scope of mm, so closing this for now

closed 2023-07-02 by khassel