MagicMirrorOrg/MagicMirror #3117: updatenotification module fails to load if any other module throws an error in creating their dom

Ryanclbryant, 2023-06-05:
I've had a broken third party module for several months and didn't realize it; the update was available but because the issue was with creating the dom on initial load we had issues with receiving the required notifications.

I noticed that `MODULE_DOM_CREATED` is being sent, but never `DOM_OBJECTS_CREATED`, which updatenotification.js relies on
[here.](https://github.com/MichMich/MagicMirror/blob/abe5c08a52ffee6b37c30c1f3160e130f0e966a2/modules/default/updatenotification/updatenotification.js#LL36C1-L36C48)

Would it be a feasible replacement for updatenotification to look for the `MODULE_DOM_CREATED` notification instead?

Sample of a successful load:
```
[Info] notificationReceived(): MODULE_DOM_CREATED (MMM-CalendarExt3.js, line 142)
[Info] notificationReceived(): DOM_OBJECTS_CREATED (MMM-CalendarExt3.js, line 142)
```

Sample of where one of any modules experiences an error during initial dom creation:
```
[Info] All modules started! (main.js, line 501)
[Info] notificationReceived(): ALL_MODULES_STARTED (MMM-CalendarExt3.js, line 142)
[Warning] [CX3] Module is not prepared yet, wait a while. (MMM-CalendarExt3.js, line 229)
[Error] TypeError: undefined is not an object (evaluating 'this.tasks.items') — MMM-Todoist.js:579
	error
[Info] notificationReceived(): MODULE_DOM_CREATED (MMM-CalendarExt3.js, line 142)
[Error] Unhandled Promise Rejection: TypeError: undefined is not an object (evaluating 'this.tasks.items')
	(anonymous function) (MMM-Todoist.js:579)
[Info] notificationReceived(): NEWS_FEED_UPDATE (MMM-CalendarExt3.js, line 142)
```

Making this change would support the module to raise the flag that updates are available, which actually resolved the error that MMM-Todoist was hitting.

sdetweil, 2023-06-05:
there are all kinds of places where there is no error checking

khassel, 2023-07-02:
@rejas @sdetweil any objections to replace `DOM_OBJECTS_CREATED` with `MODULE_DOM_CREATED`?

rejas, 2023-07-08:
DOM_OBJECTS_CREATED is send after all modules are done, MODULE_DOM_CREATED is send for each module, so how would you implement the logic in the updatenotification module then?

bugsounet, 2023-07-14:
@rejas but in this case if we use `MODULE_DOM_CREATED`, i'm not sure that `Module.definitions` is set correctly
we have to check around this

closed 2024-10-13 by github-actions[bot]