MagicMirrorOrg/MagicMirror #3122: showModule and force, lockString options

bugsounet, 2023-06-10:
Hi,

I just saw a strange behavior of this function

it will therefore not be possible to do

`show([speed], { force: true, lockString: [Lock-String] })`

I was seeing more behavior like "cancel all lockString" and set the new one

currently the result will be:
  * sets the lockString next
  * reset all locks
  
Is this really the desired result? It shocks me a bit

sdetweil, 2023-06-10:
the doc 
https://docs.magicmirror.builders/development/core-module-file.html#module-instance-methods
says 
force: true  
clears ALL lock strings in the list 

>Note: The locking mechanism can be overwritten by using the force tag:
```
moduleA.show(0, { force: true });
```
>This will reset the lockstring array, and will show the module.

sdetweil, 2023-06-10:
so, lockstrings is not a stack (fifo/lifo) but a collection..  

as long as there is something in the list, the module is hidden.. 

when the list is empty the module will be shown.. 

there is no 'checking' of the lockstrings (like you mod x, only mod x can remove)
if you KNOW the string, you can remove it.. the 'value' does NOT have to be a module identifier..

bugsounet, 2023-06-10:
@sdetweil 

yes yes well I know the rules

I was just surprised that you can't `reset` and put a new lock instead

sdetweil, 2023-06-10:
ah.. yes,
show never adds a lockstring..

bugsounet, 2023-06-10:
So, I will to purpose and code this

rejas, 2023-06-10:
I never looked at the whole lockstring code to be honest (so I dont know how useful or good it is), since it wasnt used really in the default modules. There's only one obscure usage [here](https://github.com/MichMich/MagicMirror/blob/abe5c08a52ffee6b37c30c1f3160e130f0e966a2/modules/default/weather/providers/openweathermap.js#L438) which is never reached anyway. 

So, are you using it @bugsounet in your modules?

sdetweil, 2023-06-10:
and MMM-Pages, and RemoteControl use it too.. (at least)

bugsounet, 2023-06-11:
@rejas: you have right, I have to use it in one of my plugin (EXT-Pages)

@sdetweil  also found another module that this option will help

Why ? it's complex to manage lockStrings on start of MM². 

I will propose a PR in this sense

khassel, 2023-09-09:
> So, in all case, I think we can close this PR

can we close the issue too?

bugsounet, 2023-09-18:
sorry, I forget to close it :)

closed 2023-09-18 by bugsounet