MagicMirrorOrg/MagicMirror #3126: enhancement: add feature "showOutdoorTemperature"

Coernel82, 2023-06-17:
The module already has the option to display the notification of ``INDOOR_TEMPERATURE`` by other modules. People also have outdoor sensors. So adding the functioncality for ``showOutdoorTemperature`` (assuming the notification ``OUTDOOR_TEMPERATURE`` is already existant.
This could then optionally replace or complement the outdoor temperature received by the weather provider.

After this https://github.com/splattner/MMM-MQTT-Notification could be used to send the notificaton if using MQTT.

rejas, 2023-06-17:
Sounds like a useful feature indeed. Do you want to work on it? or would you want help with that?

Coernel82, 2023-06-17:
I am a total noob. I can program some minimalistic PHP and CPP for Arduino. As much as I would love to this is far too complex for me I guess.
However I will have a look at the code and report back if I might be able to do it!

Coernel82, 2023-07-02:
I had the promised look at the code and it seems as this could be established without to much hassle as the code already exists for the other broadcast variable.
For some reasons however I decided not to proceed this further

- first of all I am really a noob so even though the task appears to be easy I might make mistakes and spend a long time correcting them. This time I currently lack of
- after more research I found out there is already a MMM MQTT weather module available getting a ton of MQTT readings such as wind, moisture, lightlevel and much more. Maybe there should be some cooperation? Have a look at https://github.com/spitzlbergerj/MMM-Currentweather-MQTT

Thanks for your understanding!

PR #3178 (2023-09-08): Added Override Notification Option to Weather Module
Fixes #3126 

Added the option `allowOverrideNotification` to `weather.js`. This allows the module to receive the `CURRENT_WEATHER_OVERRIDE` notification. The expected payload for this notification is a full/partial `weatherObject` that is used to supplement/replace the existing `weatherObject` returned by whichever weather provider is in use. 

No visual changes.

First time contributing - let me know if I've missed something :slightly_smiling_face:
commit 5cbdd28d (2023-09-08): Added Override Notification Option to Weather Module (#3178)

Fixes #3126 

Added the option `allowOverrideNotification` to `weather.js`. This
allows the module to receive the `CURRENT_WEATHER_OVERRIDE`
notification. The expected payload for this notification is a
full/partial `weatherObject` that is used to supplement/replace the
existing `weatherObject` returned by whichever weather provider is in
use.

No visual changes.

First time contributing - let me know if I've missed something
:slightly_smiling_face:

---------
closed 2023-10-01 by khassel