MagicMirrorOrg/MagicMirror #3128: ALT no longer works

ankonaskiff17, 2023-06-19:
I have built several MM's with no problems but went through process to create one today and while I was able to create the MM I was unable to hit ALT to start editing, adding and removing modules. 

It is almost as if the initial install of MM is running in KIOSK mode.  
I posted this issue on the MM forum and got "use ALT+Spacebar" or CTL+m but they dididn't do anything. Used both Sam's script and did the "official" manual installation but again, unable to use ALT, ALT+Spacebar, etc to quickly edit modules etc. 

I'm stumped and having done the exact same process to create MM's as gifts, I have the process down pretty well but something has changed

sdetweil, 2023-06-19:
can u provide the output of 

lsb_release -a
and 
uname -a
node -v
npm -v
and
grep version node_modules/electron/package.json
  mine on x64 shows  "version": "22.3.4"

also, on my x64 system alt+space-bar (both at the same time)  shows this

ankonaskiff17, 2023-06-19:
Okay Here it all is, I have two Pi 4B's so for ease of writing Pi1 and Pi2
Pi1 install was done on 17th, Pi2 install was started around 12:30 PM today.
My steps are same every time I  do this so:

1. Format 32 gb micro SD card
2. Install Pi Installer on my laptop
3. Install Pi using the installer, set network ssid and pw .local name, user name, password
4. Click Write and wait for process to complete.
5. Once complete I put sd card in Pi.
6. Go to MM web site, click on alternative installs choose @sdetweil option
7. On Github I Copy the installation script and paste it into terminal hit enter and let it do its thing
8. Click yes to using PM2 and yes to kill screensaver
9. Historically at this point I hit the ALT key which shows me the hidden toolbar, 
[image]

Now ALT by itself does nothing
ALT+Spacebar also does nothing if the MM is in fullscreen mode.

I noticed after running GREP that it says file doesn't exist but if I drill down through folders package.json is there but it appears  version number is different. This is what I found when I opened /MagicMirror/node_modules/electron/package.json.
You show 22.3.4 while I show 22.3.13 but that is beyond me.

Here is the result of running those commands you list on both Pi 1 and Pi 2
Pi 1, 64 bit
Installed 6/17/2023

lsb_release -a 
~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye

uname -a
Linux PiTest 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

node -v
v16.20.0

npm -v
8.19.4

grep version node_modules/electron/package.json
grep: node_modules/electron/package.json: No such file or directory

Pi 2, 64 bit
Installed 6/19/2023 approx 13:00

lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye

uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

node -v
v16.20.0

npm -v
8.19.4

grep version node_modules/electron/package.json
grep: node_modules/electron/package.json: No such file or directory

[image]

{
  "main": "index.js",
  "types": "electron.d.ts",
  "bin": {
    "electron": "cli.js"
  },
  "scripts": {
    "postinstall": "node install.js"
  },
  "dependencies": {
    "@electron/get": "^2.0.0",
    "@types/node": "^16.11.26",
    "extract-zip": "^2.0.1"
  },
  "engines": {
    "node": ">= 12.20.55"
  },
  "name": "electron",
  "repository": "https://github.com/electron/electron",
  "description": "Build cross platform desktop apps with JavaScript, HTML, and CSS",
  "license": "MIT",
  "author": "Electron Community",
  "keywords": [
    "electron"
  ],
  "version": "22.3.13"
}

sdetweil, 2023-06-19:
ok, but you are in VNC, so that is ANOTHER app that controls the keyboard..

tell me what it looks like on the physical keyboard attached to the pi..

ankonaskiff17, 2023-06-19:
Just to get VNC out of the picture, the install I did today was keyboard direct to Pi. So VNC not used at all.

ankonaskiff17, 2023-06-19:
I did turn it on afterwards because it made it easier to copy all those commands but install was direct.

ankonaskiff17, 2023-06-19:
I can turn it off on both if you want but that weird dropdown is with the keyboard direct to the pi in both instances

ankonaskiff17, 2023-06-19:
I have to run do an errand but shouldn't take to long.
Just post any questions or instructions and I'll get right back on it when I get back.

sdetweil, 2023-06-19:
i do all my installs over ssh..  

boot pi, create user,  turn on ssh, set hostname (I have 10 boxes running)
reboot, then go to my desktop to run the rest of the commands
I don't need to touch the UI, so I never do..

ankonaskiff17, 2023-06-19:
See, you are just the opposite of me.Since I'm making these to be visually appealing to me or whoever, VNC is good for that because I just set monitor and pi on table where I can see end result. Different brands and sizes of monitors seem to all The rare occasion when I look under the hood I legit marvel at how programmers can crank that stuff out. Wayne Wells

sdetweil, 2023-06-19:
i don't have a visual bone in my body!.. I can tell you if something looks bad.. but that is about the extent of that skill.

ankonaskiff17, 2023-06-19:
Isn't a programmer rated on the "elegance" of his/her solution to a problem.
Beauty is in the eye of the beholder as they say some of the things the folks in the programming community qualify as art, depending on who's looking.

In any case, where do we go from here as far as this uncooperative MM install. I don't mind doing whatever, just tell me what to do.  I can stick a memory stick in the Pi if need be to keep VNC totally out of picture. Stick whatever code, screenshots, on memory stick than stick in my Laptop and send your way.

sdetweil, 2023-06-20:
I don't know. I don't see any breaking changes listed in electron that would have this effect, no bugs posted..

it's not MagicMirror

if you use chromium on the pi to http://localhost:8080

do you still have the same problem

sdetweil, 2023-06-20:
elegance has nothing to do with visually appealing

sdetweil, 2023-06-20:
yes Parm is short for parameter

those are all parameters to the chromium browser to make it look like electron kiosk mode.  have to use all of them on one command

ankonaskiff17, 2023-06-20:
How do I pull the most recent **PREVIOUS** rendition of MM and install it? I purposely but ignorantly did it yesterday and ALT worked but can't replicate how I got the previous version onto the Pi.

khassel, 2023-07-02:
> it's not MagicMirror

yes, we are dependent on electron and other dependencies, nothing we can do in the mm code, so closing this for now.

closed 2023-07-02 by khassel