MagicMirrorOrg/MagicMirror #3130: Update Notification repo seems to be failing

mholzma, 2023-06-19:
Running MagicMirror in Docker image (from khassel) and I am seeing this error every 10 minutes (which aligns to the updatenotification module). I've tested this both on a raspberry pi and also on another linux box. 

MM version is up to date: 2.23.0

The error I am seeing is:
[ERROR] Failed to retrieve repo info for MagicMirror: TypeError: Cannot read properties of null (reading '1')

Please let me know if there is any other information I should share.

khassel, 2023-06-25:
Thanks for reporting this, I can reproduce this and will take a look within the next days

khassel, 2023-07-01:
fixed with new release `v2.24.0`

closed 2023-07-01 by khassel

mholzma, 2023-07-02:
Thanks and I validated the fix works as well.

PR #3136 (2023-06-27): fix updatenotification where no branch is checked out ...
... but e.g. a version tag

fixes #3130 

This happens e.g. in my docker image where I use the version tag to get the mm sources.

With this PR the error message is avoided and there will be never an updatenotification when using a tag. This is o.k. because a tag should never be moved.
commit 3c35d346 (2023-06-27): fix updatenotification where no branch is checked out ... (#3136)

... but e.g. a version tag

fixes #3130 

This happens e.g. in my docker image where I use the version tag to get
the mm sources.

With this PR the error message is avoided and there will be never an
updatenotification when using a tag. This is o.k. because a tag should
never be moved.