MagicMirrorOrg/MagicMirror #3142: Idea of new default module

bugsounet, 2023-07-02:
Hi,

I'm thinking about this new default module: self updating of 3rd party module
I use it with my modules/plugins since 3 years now!
So why not create a default secure module for this ?

Actually how i do this:

* I have created an embed script with `npm run update` command on each module. It's allows to fetch last version of this module and install it
sample [there](https://github.com/bugsounet/MMM-Test/blob/master/installer/update.sh)
* I have created a module to check if an update is available (+/- like updatenotification default module)
* If an update is available, it checks some rules (only my modules)
* It install the update (run the update command)
* after install: if no error: restart MagicMirror (with or without pm2)
* if any errors: display a warn (with alert)

How I plan to do this with a new default module !?

* it's a developer choice to create an embed command or not
* If developer make choice to add a self updater: add `update` feature in `package.json` like this:
```js
{
  "name": "MMM-Test",
  "description": "Test module",
  "main": "MMM-Test.js",
  "update": "update", <--- update feature with command to launch
  "scripts": {
    "preinstall": "installer/preinstall.sh",
    "postinstall": "installer/postinstall.sh",
    "update": "installer/update.sh" <--- update command to launch (with npm run xxx)
  },
  "author": "bugsounet",
  "license": "MIT",
  "repository": {
    "type": "git",
    "url": "https://github.com/bugsounet/MMM-Test"
  }
}
```
* if this feature don't exist, don't apply self updating
* actually with my update module: my rules says run `npm run update` if it's an @bugsounet module Tag. So I can check `package.json` with `update` feature
* Naturally... I will check also if no `package.json`

what do you this about this ?

khassel, 2023-07-02:
my thoughts so far:
- yes, would be a nice (optional) feature
- why a new default module, why not add a param `autoupdate` to `updatenotification` and implement it there?
- `after install: if no error: restart MagicMirror (with or without pm2)`: will not work with docker container
- may define the update command not in package.json but in the module config and execute it as it is done with the git commands in https://github.com/MichMich/MagicMirror/blob/master/modules/default/updatenotification/git_helper.js ?

sdetweil, 2023-07-02:
I think the title needs to be changed,    

Module sample, with autoupdate 

none of the default modules would use this as they are updated by base.. 

I think this should be started from npm then its clear where it gets triggered, and where it goes from there

bugsounet, 2023-07-02:
I had implemented this at the beginning but it causes too many problems.

Afterwards, a simple `git pull && npm install` to implement in the git helper is not enough
Why ?
let's take the example of `request`... missing (sam is used to it now!)
it is depreciated, ok... but some old modules still use it
You understand that the user will add it to package.json (via npm install request)
it will add it in package.json
and disaster ... you can't update because the file has been modified

this can also be the case if the user modifies one of the source files

that's why I prefer in the package.json as I explained

For me, an update should not become a headache.
It must be tested in all directions, in order to verify the proper functioning.
This is why it must be validated by the developer

Example in my case: `git pull && npm install` will never work
Why ? because the source files are minify with my installer (for speed up!)

sdetweil, 2023-07-02:
@bugsounet for this 
>it will add it in package.json
and disaster ... you can't update because the file has been modified

the module hasn't been updated in years, unlikely this will need to be addressed.. 
but your point is taken.. files will be changed.. unexpectedly

bugsounet, 2023-07-03:
and what does @rejas think of all this?
what is your point of view?

sdetweil, 2023-07-03:
well, if we have a sample for the NPM process, then we need a sample module itself..  or two.. one with and without node_helper

bugsounet, 2023-07-03:
Tomorrow, I will recode `MMM-Test` for testing self updates

khassel, 2023-09-09:
I think this is under development here: https://github.com/MichMich/MagicMirror/pull/3150

khassel, 2024-06-24:
Looks like this is done so I will close this (if not please add a comment and I will reopen).

closed 2024-06-24 by khassel