MagicMirrorOrg/MagicMirror #3143: v2.24.0 - Clock module - (showSunTimes: true) Retruns Undefined - formatTime is not defined

fire1ce, 2023-07-02:
MagicMirror v2.24.0 

Clock module:

```json
    {
      module: 'clock',
      position: 'top_bar',
      config: {
        clockBold: true,
        // showSunTimes: true,
        displayType: 'both',
        analogSize: '400px',
        analogPlacement: 'right',
        secondsColor: '#ff5263',
        lat: xx.xxx,
        lon: xx.xxx,
      },
```

when showSunTimes: true
error from the console:

```log
clock.js:191 Uncaught (in promise) ReferenceError: formatTime is not defined
    at Class.getDom (clock.js:191:5)
    at main.js:110:35
    at new Promise (<anonymous>)
    at updateDom (main.js:108:10)
    at main.js:53:31
    at Array.forEach (<anonymous>)
    at createDomObjects (main.js:20:11)
    at Object.modulesStarted (main.js:485:4)
    at startModules (loader.js:42:6)
    at async loadNextModule (loader.js:218:6)
```

khassel, 2023-07-02:
tested your above config with 

```
lat: 50,
lon: 8,
```

without getting an error

[image]

rejas, 2023-07-02:
You got to enable showSunTimes @khassel then you can reprdocue it. Got a fix ready, but wanted to write tests for it before submitting.

khassel, 2023-07-02:
enabled it, works for me with this config

```js
    {
      module: 'clock',
      position: 'top_bar',
      config: {
        clockBold: true,
        showSunTimes: true,
        displayType: 'both',
        analogSize: '400px',
        analogPlacement: 'right',
        secondsColor: '#ff5263',
        lat: 50,
        lon: 8,
      },
    },
```

fire1ce, 2023-07-02:
> enabled it, works for me with this config
>
> ```js
>     {
>       module: 'clock',
>       position: 'top_bar',
>       config: {
>         clockBold: true,
>         showSunTimes: true,
>         displayType: [...]

Tested with this clock config, Having the same issue. ¯\_(ツ)_/¯ 

[image]

fire1ce, 2023-07-02:
Btw, it happen since the last version only.

khassel, 2023-07-02:
o.k., now I can reproduce it.

I had a weather module before the clock config in `config.js` and the weather module loads the missing script so the clock module works in this case. If I move the clock config at first place I get the error.

fire1ce, 2023-07-02:
> o.k., now I can reproduce it.
> 
> I had a weather module before the clock config in `config.js` and the weather module loads the missing script so the clock module works in this case. If I move the clock config at first place I get the error.

Nothing better then possibility to reproduce a bug. No second thought no bugs is better 😶

closed 2023-10-01 by khassel

wbleek, 2023-11-04:
This issue might not be fixed entirely because it also appears when selection the time zone by text instead of lon and lat parameter. See issue https://github.com/MichMich/MagicMirror/issues/3262

PR #3146 (2023-07-02): Fix undefined error for showSunTime / showMoonTime in clock module
Fixes #3143 and adds tests for showSunTime / showMoonTime
commit f802c85a (2023-07-02): Fix undefined error for showSunTime / showMoonTime in clock module (#3146)

Fixes #3143 and adds tests for showSunTime / showMoonTime