MagicMirrorOrg/MagicMirror #3147: 2.24.0 - Calendar Error. Could not fetch calendar

The-Exterminator, 2023-07-02:
Is work before update to 2.24.0

[02.07.2023 21:19.56.057] [ERROR] Calendar Error. Could not fetch calendar:  http://localhost:8080/modules/calendars/190C3BC4-1E8B-43F4-9D1E-8950E1EBE438.ics FetchError: request to http://localhost:8080/modules/calendars/190C3BC4-1E8B-43F4-9D1E-8950E1EBE438.ics failed, reason: connect ECONNREFUSED ::1:8080
    at ClientRequest.<anonymous> (/home/pi/MagicMirror/node_modules/node-fetch/lib/index.js:1505:11)
    at ClientRequest.emit (node:events:513:28)
    at Socket.socketErrorListener (node:_http_client:502:9)
    at Socket.emit (node:events:513:28)
    at emitErrorNT (node:internal/streams/destroy:151:8)
    at emitErrorCloseNT (node:internal/streams/destroy:116:3)
    at process.processTicksAndRejections (node:internal/process/task_queues:82:21) {
  type: 'system',
  errno: 'ECONNREFUSED',
  code: 'ECONNREFUSED'

khassel, 2023-07-02:
This seems to be a new restriction of the newer electron version.

Please use the ip-address of your pi instead of localhost, e.g. `http://192.168.0.39:8080/modules/calendars/190C3BC4-1E8B-43F4-9D1E-8950E1EBE438.ics` instead of `http://localhost:8080/modules/calendars/190C3BC4-1E8B-43F4-9D1E-8950E1EBE438.ics`.

khassel, 2023-07-02:
You can also use `0.0.0.0` instead of `localhost`

So `localhost` is not usable anymore as http request on local folders, you have to use
- `0.0.0.0` or
- the ip adress of your pi

Let me know if this works for you.

The-Exterminator, 2023-07-02:
> This seems to be a new restriction of the newer electron version.
>
>
>
> Please use the ip-address of your pi instead of localhost, e.g. [...]

I Will try that tomorrow

sdetweil, 2023-07-02:
@khassel   I don't understand that.. how does el know that the http server is hosting a local file?
or is this an http clear text issue?

khassel, 2023-07-02:
I did not dig deeper into this, I only tested the different approaches ...

The-Exterminator, 2023-07-03:
> You can also use `0.0.0.0` instead of `localhost`
> 
> So `localhost` is not usable anymore as http request on local folders, you have to use
> 
>     * `0.0.0.0` or
> 
>     * the ip adress of your pi
> 
> 
> Let me know if this works for you.

Work with 0.0.0.0 thanks alot mate, you can close this :-)

closed 2023-07-03 by khassel

sdetweil, 2024-01-22:
weird. I use localhost all the time for local files, never have a problem