MagicMirrorOrg/MagicMirror #3148: Magic Mirror problem after update

Inubashi, 2023-07-05:
Hello,
I need help, i don't have good skills in Javascript. My Magic Mirror don't startet after Update, the problem is one Modul.
The Modul is MMM-Holiday from cowboysdude.

"its a very good modul but after a update from the Magic Mirror 2 it's don't working.
My problem is the Magic Mirror don't start after the instellation of the modul.
Its not a synthax fail. The modul blocket the autostart and the manuel start."
Can you fix it ?  or give me a Tip please.
I don't want a other Holiday Modul.

sdetweil, 2023-07-05:
yes, this module suffers from a missing library 
see https://forum.magicmirror.builders/topic/15778/fix-for-black-screen-in-2-16-and-later

sdetweil, 2023-07-05:
this issue should opened on the module github page as the author needs to fix the installation

khassel, 2023-07-05:
please follow the suggestions from @sdetweil above. We don't support foreign modules here, so closing this.

closed 2023-07-05 by khassel