MagicMirrorOrg/MagicMirror #3151: v2.24.0 won´t start in clientonly-Mode

PatteWi, 2023-07-15:
Hi all,

i have running MagicMirror in a VM on a NUC in server-mode (headless) and access to it with an RPI3 behind a screen/Mirror in clientonly-mode. After i updated Server and Client to v2.24.0, the client isn´t able to connect to the server again. It doesn´t even start.
As a workaround i downgraded the client to v2.23.0 and i can access again.

BR
Patrick

sdetweil, 2023-07-15:
yep, a bug.. 

we changed startup for asynchronous modules, .. but removed a part that is needed for client only..

temp fix if u want 

in clientonly/index.js 
add
```
				const env = Object.create(process.env);
				env.clientonly=true;     // <---------add this line
				const options = { env: env };
```

in js/electron.,js add 
```
/**
 * Handle errors from self-signed certificates
 */
app.on("certificate-error", (event, webContents, url, error, certificate, callback) => {
	event.preventDefault();
	callback(true);
});

if(process.env.clientonly){                                    //  add these 6 lines...  between the sections above and below
	app.whenReady().then(() => {
		Log.log("Launching application.");
		createWindow();
	});
}

// Start the core application if server is run on localhost
// This starts all node helpers and starts the webserver.
```

khassel, 2023-07-15:
@sdetweil could reproduce this and your fix works, so can you please provide a PR for this? Thanks.

sdetweil, 2023-07-16:
done

PR #3154 (2023-07-16): fix clientOnly not starting up after async startup added in 2.23
Fixes #3151 

app.whenReady() was removed when async changes made, needed for clientonly in electron.js
commit 1bd146f5 (2023-07-16): fix clientOnly not starting up after async startup added in 2.23 (#3154)

Fixes #3151 

app.whenReady() was removed when async changes made, needed for
clientonly in electron.js
closed 2023-10-01 by khassel