MagicMirrorOrg/MagicMirror #3156: default weather module appears to be showing the 'feels like'

earnestrichards, 2023-07-19:
The default weather module appears to be showing the 'feels like' temperature in the 5 day forecast rather than the forecast temperature.

snippet from config.js attached

`{
			module: "weather",
			position: "top_right",
			config: {
				weatherProvider: "openweathermap",
				type: "current",
				roundTemp: "true",
				location: "Fresno",
				locationID: "4692400", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				apiKey: "xxxxxx"
			}
		},
		{
			module: "weather",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				weatherProvider: "openweathermap",
				type: "forecast",
				roundTemp: "true",
				fade: false,
				location: "Fresno",
				locationID: "4692400", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				apiKey: "xxx"
			}
		},
		`

rejas, 2023-07-19:
The forecast shows not one temperature but rather max and min temperature for the day. So it might appear like the feelslike temp but I think it isnt.

khassel, 2023-07-19:
related to forum thread https://forum.magicmirror.builders/topic/17916/default-weather-module-question

khassel, 2023-07-19:
you can use a web browser and get the delivered json by using this url: https://api.openweathermap.org/data/2.5/forecast?id=4692400&units=metric&lang=en&APPID=xxx 

you have to substitute xxx by your apiKey.

You can see the values in the json.

I used `jq` and tested some days for your location with

```bash
❯ cat tmp.json | jq -r '.list[] | select(.dt_txt? | match("2023-07-20.*")) | "\(.main.temp_min) \(.main.temp_max)"'
34.99 35.38
27.53 27.53
25.53 25.53
25.1 25.1
25.04 25.04
31.86 31.86
38.28 38.28
39.9 39.9
```

```bash
❯ cat tmp.json | jq -r '.list[] | select(.dt_txt? | match("2023-07-21.*")) | "\(.main.temp_min) \(.main.temp_max)"'
34.41 34.41
27.65 27.65
25.88 25.88
25.48 25.48
25.56 25.56
32.1 32.1
38.66 38.66
39.91 39.91
```

and the displayed values in the mirror (Tomorrow and Fri) are correct 

[image]

jca2112, 2023-08-21:
Is there any update on this issue? The Weather module forecast (openweathermap) is showing the "feels like" (heat index) temperatures (at least for the high/max) in the five day forecast. This has been going on for a while.

[image]

rejas, 2023-08-22:
As I wrote above, I cannot redproduce the error yet. @jca2112 you have the problem with the openweathermap provider?

jca2112, 2023-08-22:
> @jca2112 you have the problem with the openweathermap provider?

Yes, using the default openweathermap config I've had for a long time - have not tried another provider.

I know that the weather forecast config does not use "showFeelsLike: false", but I tried adding it (and also to the weather current module above it) and confirmed that setting had no effect on the 5 day forecast. I believe this is the api url:

https://api.openweathermap.org/data/2.5/forecast?id=4076607&units=imperial&lang=en&APPID=enterkeyhere

khassel, 2023-08-28:
> Is there any update on this issue?

I did already a deeper analysis in [a comment above](https://github.com/MichMich/MagicMirror/issues/3156#issuecomment-1642592921) which prooved that the weather module displays the forecast temperature data provided by openweathermap.

If someone can provide the json (as file) from openweathermap and a corresponding screenshot of mm we can discuss it deeper again.

khassel, 2023-10-15:
closing for now, if someone provides an example we can reopen.

closed 2023-10-15 by khassel