MagicMirrorOrg/MagicMirror #3157: Calendar module not displaying correct time for recurring events

tenmilez, 2023-07-22:
Platform: Mac Version 13.4.1 (c), Docker Desktop 4.21.1, 

Node Version: Node.js v18.16.1.

MagicMirror² Version: 2.24.0

Description: Using an iCal feed from Google calendars, events that are part of a recurring series will not show the correct time unless that event is individually modified. 

For example, a weekly meeting at 3pm will show 9am, but if I add a note to that meeting ("this week's topic: hamburgers") then it'll display 3pm properly. 

This code snippet is from the iCal/ics output. 

```
BEGIN:VEVENT
DTSTART;TZID=Europe/Berlin:20230725T150100
DTEND;TZID=Europe/Berlin:20230725T160100
RRULE:FREQ=WEEKLY;WKST=SU;INTERVAL=4
DTSTAMP:20230722T174520Z
UID:
CREATED:20230103T070914Z
LAST-MODIFIED:20230722T174452Z
LOCATION:REDACTED
SEQUENCE:3
STATUS:CONFIRMED
SUMMARY:Staff meeting
TRANSP:TRANSPARENT
BEGIN:VALARM
ACTION:NONE
TRIGGER;VALUE=DATE-TIME:19760401T005545Z
END:VALARM
END:VEVENT

BEGIN:VEVENT
DTSTART;TZID=Europe/Berlin:20230728T133000
DTEND;TZID=Europe/Berlin:20230728T143000
DTSTAMP:20230722T174520Z
UID:
RECURRENCE-ID;TZID=Europe/Berlin:20230728T133000
CREATED:20230420T204441Z
DESCRIPTION:ALQ Bingo
LAST-MODIFIED:20230620T220352Z
LOCATION:REDACTED
SEQUENCE:0
STATUS:CONFIRMED
SUMMARY:E Flight Meeting
TRANSP:OPAQUE
BEGIN:VALARM
ACTION:NONE
TRIGGER;VALUE=DATE-TIME:19760401T005545Z
END:VALARM
END:VEVENT
```

[image]

Steps to Reproduce: Add recurring event to Google Calendar. Get iCal link and add that to MM Calendar module configuration. Timezone might be relevant, I'm using Berlin. 

Expected Results: Time of actual event. 

Actual Results: Time of event -6 hours. 

Configuration: Pretty stock

```
$ cat config.js
/* MagicMirror² Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * see https://docs.magicmirror.builders/configuration/introduction.html
 * and https://docs.magicmirror.builders/modules/configuration.html
 *
 * You can use environment variables using a `config.js.template` file instead of `config.js`
 * which will be converted to `config.js` while starting. For more information
 * see https://docs.magicmirror.builders/configuration/introduction.html#enviromnent-variables
 */
let config = {
        address: "0.0.0.0",     // Address to listen on, can be:
                                                        // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
                                                        // - another specific IPv4/6 to listen on a specific interface
                                                        // - "0.0.0.0", "::" to listen on any interface
                                                        // Default, when address config is left out or empty, is "localhost"
        port: 8080,
        basePath: "/",                  // The URL path where MagicMirror² is hosted. If you are using a Reverse proxy
                                                        // you must set the sub path here. basePath must end with a /
        ipWhitelist: [],        // Set [] to allow all IP addresses
                                                                                                                        // or add a specific IPv4 of 192.168.1.5 :
                                                                                                                        // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
                                                                                                                        // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
                                                                                                                        // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

        useHttps: false,                // Support HTTPS or not, default "false" will use HTTP
        httpsPrivateKey: "",    // HTTPS private key path, only require when useHttps is true
        httpsCertificate: "",   // HTTPS Certificate path, only require when useHttps is true

        language: "en",
        locale: "en-US",
        logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
        timeFormat: 24,
        units: "metric",

        modules: [
                {
                        module: "alert",
                },
                {
                        module: "updatenotification",
                        position: "top_bar"
                },
                {
                        module: "clock",
                        position: "top_left"
                },
                {
                        module: "calendar",
                        header: "Calendar",
                        position: "top_left",
                        config: {
                                calendars: [
                                        {
                                                symbol: "calendar-check",
                                                url: "https://ics.calendarlabs.com/76/mm3137/US_Holidays.ics"
                                        },
                                        {
                                                symbol: "user",
                                                url: REDACTED
                                        },
                                        {
                                                symbol: "briefcase",
                                                url: REDACTED
                                        },
                                        {
                                                symbol: "building",
                                                url: REDACTED
                                        }
                                ]
                        }
                },
                {
                        module: "weather",
                        position: "top_right",
                        config: {
                                weatherProvider: "openweathermap",
                                type: "current",
                                //apiVersion: "3.0",
                                locationID: REDACTED, //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
                                apiKey: REDACTED
                        }
                },
                {
                        module: "weather",
                        position: "bottom_right",
                        header: "Weather Forecast",
                        config: {
                                weatherProvider: "openweathermap",
                                type: "forecast",
                                //apiVersion: "3.0",
                                locationID: REDACTED, //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
                                apiKey: REDACTED
                        }
                }
        ]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}
```

Additional Notes: none.

sdetweil, 2023-07-22:
sounds like the system tinezone is set incorrectly

tenmilez, 2023-07-23:
Fixing the system time and restarting the system worked. Interesting that the clock module found the correct time without any configuration, but the calendar module didn't. 

For anyone else that comes along, timedatectl command didn't exist so I followed the instructions in step 1 [here](https://www.itzgeek.com/how-tos/linux/debian/how-to-change-time-zone-in-debian-10-debian-9.html) to set the timezone via command line.

closed 2023-07-23 by tenmilez

sdetweil, 2023-07-23:
avd you can set the timezone. from the pi desktop by clicking on the date

sdetweil, 2023-07-23:
also using the preferences from the top left menu