MagicMirrorOrg/MagicMirror #3158: Unexpected Token 'if'

iTheMaverick, 2023-08-03:
Whenever I try and start magicmirror it says that it has an issue with the if at the very end of the

`/* MagicMirror² Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl/
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * see https://docs.magicmirror.builders/configuration/introduction.html
 * and https://docs.magicmirror.builders/modules/configuration.html
 */
let config = {
address: "localhost", // Address to listen on, can be:
// - "localhost", "127.0.0.1", "::1" to listen on loopback interface
// - another specific IPv4/6 to listen on a specific interface
// - "0.0.0.0", "::" to listen on any interface
// Default, when address config is left out or empty, is "localhost"
port: 8080,
basePath: "/", // The URL path where MagicMirror² is hosted. If you are using a Reverse proxy
// you must set the sub path here. basePath must end with a /
ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], // Set [] to allow all IP addresses
// or add a specific IPv4 of 192.168.1.5 :
// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
// or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:[192.168.3.0/28](http://192.168.3.0/28)"],

useHttps: false, // Support HTTPS or not, default "false" will use HTTP
httpsPrivateKey: "", // HTTPS private key path, only require when useHttps is true
httpsCertificate: "", // HTTPS Certificate path, only require when useHttps is true

language: "en",
locale: "en-US",
logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
timeFormat: 24,
units: "metric",
// serverOnly:  true/false/"local" ,
// local for armv6l processors, default
//   starts serveronly and then starts chrome browser
// false, default for all NON-armv6l devices
// true, force serveronly mode, because you want to.. no UI on this device

modules: [
{
module: "alert",
},
{
module: "updatenotification",
position: "top_bar"
},
{
module: "clock",
position: "top_left"
},

{
module: "calendar",
//header: "US Holidays",
position: "top_left",
config: {
calendars: [
{
symbol: "calendar-check",
url: "webcal://[www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics](http://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics)"

}
]
}
},

{
    module: 'MMM-MonthCalendar',
    position: "top_left",
    header: "",
    config: {
        updateDelay: 5,
        showAdjacentMonths: true
    }
},
{
  module: "MMM-MyScoreboard",
  position: "bottom_left",
  classes: "default everyone",
  header: "Scoreboard",
  config: {
    showLeagueSeparators: true,
    colored: true,
    viewStyle: "largeLogos",
    sports: [
      {
        league: "NHL",
        teams: ["CAR"]
      },
      {
        league: "MLB",
        teams: ["ATL"],
      },
    ]

  }
},
{
module: 'MMM-AirQuality',
position: 'top_right', // you may choose any location
header: 'Air Quality', //choose a header if you like
config: {
 location: 'usa/north-carolina/northampton' // the location to check the index for
}
},

{
    module: 'MMM-nominalflight',
position: 'bottom_right', // you may choose any location
header: 'Upcoming Shuttle Launches', // optional
config: {
launches : 5, // optional
}
 },

{
  module: "MMM-OpenWeatherMapForecast",
  header: "Weather",
  position: "top_right",
  classes: "default everyone",
  disabled: false,
  config: {
    apikey: "197f30157744983e05e347707529340d",
    latitude: "36.42501",
    longitude: "-77.104568",      
    iconset: "6oa",
    concise: false,
    forecastLayout: "tiled"
  }
},
/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}`

sdetweil, 2023-08-03:
you are missing the close bracket on the modules:[]

list

add a line with just
```
 ]
```
before the comment before the if on the last line

KristjanESPERANTO, 2023-08-03:
Try this:

```js
/* MagicMirror² Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl/
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * see https://docs.magicmirror.builders/configuration/introduction.html
 * and https://docs.magicmirror.builders/modules/configuration.html
 */
let config = {
  address: "localhost", // Address to listen on, can be:
  // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
  // - another specific IPv4/6 to listen on a specific interface
  // - "0.0.0.0", "::" to listen on any interface
  // Default, when address config is left out or empty, is "localhost"
  port: 8080,
  basePath: "/", // The URL path where MagicMirror² is hosted. If you are using a Reverse proxy
  // you must set the sub path here. basePath must end with a /
  ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], // Set [] to allow all IP addresses
  // or add a specific IPv4 of 192.168.1.5 :
  // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
  // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
  // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:[192.168.3.0/28](http://192.168.3.0/28)"],

  useHttps: false, // Support HTTPS or not, default "false" will use HTTP
  httpsPrivateKey: "", // HTTPS private key path, only require when useHttps is true
  httpsCertificate: "", // HTTPS Certificate path, only require when useHttps is true

  language: "en",
  locale: "en-US",
  logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
  timeFormat: 24,
  units: "metric",
  // serverOnly:  true/false/"local" ,
  // local for armv6l processors, default
  //   starts serveronly and then starts chrome browser
  // false, default for all NON-armv6l devices
  // true, force serveronly mode, because you want to.. no UI on this device

  modules: [
    {
      module: "alert",
    },
    {
      module: "updatenotification",
      position: "top_bar",
    },
    {
      module: "clock",
      position: "top_left",
    },

    {
      module: "calendar",
      //header: "US Holidays",
      position: "top_left",
      config: {
        calendars: [
          {
            symbol: "calendar-check",
            url: "webcal://[www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics](http://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics)",
          },
        ],
      },
    },

    {
      module: "MMM-MonthCalendar",
      position: "top_left",
      header: "",
      config: {
        updateDelay: 5,
        showAdjacentMonths: true,
      },
    },
    {
      module: "MMM-MyScoreboard",
      position: "bottom_left",
      classes: "default everyone",
      header: "Scoreboard",
      config: {
        showLeagueSeparators: true,
        colored: true,
        viewStyle: "largeLogos",
        sports: [
          {
            league: "NHL",
            teams: ["CAR"],
          },
          {
            league: "MLB",
            teams: ["ATL"],
          },
        ],
      },
    },
    {
      module: "MMM-AirQuality",
      position: "top_right", // you may choose any location
      header: "Air Quality", //choose a header if you like
      config: {
        location: "usa/north-carolina/northampton", // the location to check the index for
      },
    },

    {
      module: "MMM-nominalflight",
      position: "bottom_right", // you may choose any location
      header: "Upcoming Shuttle Launches", // optional
      config: {
        launches: 5, // optional
      },
    },

    {
      module: "MMM-OpenWeatherMapForecast",
      header: "Weather",
      position: "top_right",
      classes: "default everyone",
      disabled: false,
      config: {
        apikey: "##################################################",
        latitude: "36.42501",
        longitude: "-77.104568",
        iconset: "6oa",
        concise: false,
        forecastLayout: "tiled",
      },
    },
  ],
};
/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {
  module.exports = config;
}
```

I recommend you to edit your first post and remove the apikey.

iTheMaverick, 2023-08-03:
> Try this:
>
> ```js
> /* MagicMirror² Config Sample
>  *
>  * By Michael Teeuw https://michaelteeuw.nl/
>  * MIT Licensed.
>  *
>  * For more information on how you can configure this file
>  * see [...]

Probably a good idea, thanks for the help.

iTheMaverick, 2023-08-03:
> Try this:
>
> ```js
> /* MagicMirror² Config Sample
>  *
>  * By Michael Teeuw https://michaelteeuw.nl/
>  * MIT Licensed.
>  *
>  * For more information on how you can configure this file
>  * see [...]

When I put this in it says: Failed to Retrieve Device Information
&
Failed to load swrast driver

It was working fine until I added 3 new modules

KristjanESPERANTO, 2023-08-03:
After fixing the calendar URL your config works on my device.

```js
    url: "webcal://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics"
```

Only MMM-AirQuality and MMM-nominalflight aren't getting data.

KristjanESPERANTO, 2023-08-03:
What the output of `uname -a`?

iTheMaverick, 2023-08-03:
Not seeing anything named uname -a

> What the output of `uname -a`?

KristjanESPERANTO, 2023-08-03:
Sorry, you have to enter this in the terminal :slightly_smiling_face: 

_"uname is a command-line utility that prints basic information about the operating system name and system hardware."_

iTheMaverick, 2023-08-03:
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

iTheMaverick, 2023-08-03:
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

> Sorry, you have to enter this in the terminal 🙂
> 
> _"uname is a command-line utility that prints basic information about the operating system name and system hardware."_

khassel, 2023-08-28:
closing this here, if you still have problems with your config please ask in the forum or on discord.

closed 2023-08-28 by khassel