MagicMirrorOrg/MagicMirror #3159: Weather module confusing days (switching to a new day before midnight)

Fredde87, 2023-08-08:
Hey guys,

I have recently updated my MagicMirror to the latest version (was running a version several years old prior). Therefore I have now had to migrate to the new weather module.

But I found that the module confuses Today and Tomorrow. I am not sure from what exact time, but at least from 10pm onwards the days start to swap over rather than at midnight.

So for example it is current Tuesday 10:45pm (in London), and the Weather module shows, Today, Tomorrow and Friday. But a few hours earlier it showed Today, Tomorrow, Thursday. So it seems to think the next day has arrived too early.

If I check the openweather data I can see that Today is actually Wednesday (so it is actually tomorrow) and Tomorrow is actually Thursday.

So the module is just a bit too keen and switching over the days too soon (not at midnight). Any idea how I can fix that?

sdetweil, 2023-08-08:
is your system time set to the correct timezone?

Fredde87, 2023-08-08:
> is your system time set to the correct timezone?

Sorry should have mentioned that. Yes it is

Fredde87, 2023-08-15:
A little update. I set absoluteDates to true in my config to get rid of the Today and Tomorrow and see whats happening behind it. I can now see that at 10pm each evening (9pm UTC), the day changes (so Tuesday now just went Wednesday).

So I think the issue is probably that the Openweather uses blocks of 2 or 3 hours perhaps (thats just a guess but correct me if I am wrong?). So when it starts the last block of the day (in my cast at 10pm), the next forecast block received is from midnight to 2 or 3am.

So this visually it all looks correct when absoluteDates is set to true. But when it is false (default), then I guess the module blindly changes the first entry to Today and the second to Tomorrow, but this work because then it will display tomorrows weather as today during this last forecast block of the day (between 10pm and midnight for me).

Does that make sense?

closed 2024-10-13 by github-actions[bot]