MagicMirrorOrg/MagicMirror #3162: All Google Calendars show no events, when broadcastPastEvents is true

Burny101, 2023-08-13:
I have multiple Google calendars and two other ones, in my MM calendar. 
As soon as i set broadcastPastEvents to true, the calendar does not show any events from the google calendars anymore.
The Google calendars are used via the private iCal-addresses.
The issue remained after updating to the newsest release as of today.

sdetweil, 2023-08-13:
can u show the output  from where you do npm start? 

if you use pm2, then you would use the command
pm2 logs --lines=100 
to display that output

you may also have to turn on calendar debug , but change the config.js line 
```
logLevel: ["INFO", "LOG", "WARN", "ERROR","DEBUG"], // Add "DEBUG" for even more logging
```

and run mm like this 

npm start  >somefile.txt 2>&1

when the output is not displayed, press ctrl-c on the commandline  and send me somefile.txt
 my email is sdetweil   on gmail

sdetweil, 2023-08-13:
do NOT post here as private details are in the debug output

khassel, 2023-09-09:
any progress here?

Burny101, 2023-10-08:
Sorry for the delay. 
I have sent the log file to @sdetweil .

sdetweil, 2023-10-08:
received and responded via email

khassel, 2023-10-08:
so can this be closed?

closed 2023-10-15 by khassel