MagicMirrorOrg/MagicMirror #3167: Issue template isnt working anymore

rejas, 2023-08-15:
... maybe something wrong in the config/settings. @MichMich can you see if somethings wrong in the settings of the repo?

rejas, 2023-09-02:
@MichMich can you please take a look?

MichMich, 2023-09-02:
It doesn't appear to see the issue template in the repo. It allows me to create one via settings. But I think it's better if it's part of the repo. Not sure why it doesn't work.

rejas, 2023-09-02:
Maybe create a new one then via settings and we move the template over?

khassel, 2023-10-20:
this was introduced with this "cleanup" commit 2908c15ea61d39a9fdee3485407e46e6bdd8a8a0

Solution is to revert this which means 

```bash
rename from .github/ISSUE_TEMPLATE/custom.md
rename to .github/ISSUE_TEMPLATE.md
```

I tested this on my fork.

@rejas I can prepare a PR with this change only (no changelog) and we merge it on `master` with `mastermerge` label to have the fix now (I will not wait until 2024). Because I changed the updatenotification logic this will trigger no notifications. O.k. or other ideas?

closed 2024-01-01 by khassel