MagicMirrorOrg/MagicMirror #3171: coverflow player

Claudiop193w, 2023-08-16:
Salve ho creato un codice coverflow player per MagicMirror ma non capisco dove sbaglio 
questi sono i miei file...
file coverflowPlayer.js 
Module.register("coverflowPlayer", {
  defaults: {
    covers: [],
    musicPath: "/path/to/music/",
  },

  start: function () {
    this.currentAlbumIndex = 0;
    this.playing = false;
    this.createCoverflow();
    this.addTouchEvents();
  },

  createCoverflow: function () {
    const coverflowContainer = document.createElement("div");
    coverflowContainer.className = "coverflow-container";
    // Creare gli elementi per le copertine degli album e aggiungerli a coverflowContainer
    // Utilizza this.config.covers per popolare le copertine degli album

    this.wrapper.appendChild(coverflowContainer);
  },

  addTouchEvents: function () {
    const self = this;
    const coverflowContainer = document.querySelector(".coverflow-container");

    coverflowContainer.addEventListener("touchstart", function (event) {
      const touchedAlbumIndex = /* Calcola l'indice dell'album toccato */;
      self.playAlbum(touchedAlbumIndex);
    });
  },

  playAlbum: function (albumIndex) {
    const albumData = this.config.covers[albumIndex];
    const audio = new Howl({
      src: [`${this.config.musicPath}${albumData.audioFile}`],
    });

    audio.play();
    this.playing = true;
    this.currentAlbumIndex = albumIndex;
  },

  notificationReceived: function (notification, payload, sender) {
    if (notification === "PAUSE") {
      if (this.playing) {
        // Pausa la riproduzione
        // Esegui altre azioni necessarie per fermare la riproduzione
        this.playing = false;
      }
    }
  },

  // ... Altre funzioni di gestione del modulo ...
});
mentre l'altro file e coverflowPlayer.css
/* Stile per il container della coverflow */
.coverflow-container {
  display: flex;
  overflow-x: scroll;
  align-items: center;
  justify-content: flex-start;
  height: 200px; /* Altezza desiderata della coverflow */
  -webkit-overflow-scrolling: touch; /* Scroll fluido su dispositivi iOS */
}

/* Stile per le copertine degli album */
.album-cover {
  width: 150px; /* Larghezza delle copertine */
  height: auto;
  margin: 0 10px; /* Margine tra le copertine */
  cursor: pointer;
  /* Transizione per effetti di feedback */
  transition: opacity 0.3s ease, transform 0.3s ease;
}

/* Effetto di feedback quando si tocca una copertina */
.album-cover.active {
  opacity: 0.7; /* Riduci leggermente l'opacità */
  transform: scale(1.05); /* Leggero effetto di ingrandimento */
}

rejas, 2023-08-16:
Hi, you are probably better of asking youur question the offical forum or the discord server

khassel, 2023-08-28:
Please ask your questions in english. AFAIS the question is about a foreign module which is out of scope of this repository.

As already mentioned ask in the forums or create an issue in the corresponding github repository. Thanks.

closed 2023-08-28 by khassel