MagicMirrorOrg/MagicMirror #3174: MagicMirror does not respect fullscreen option specificed in config.js

alejo4373, 2023-08-27:
**Platform**:  
* Raspberry Pi 4. Model B
* Raspberry Pi OS. Debian 11.7
* Electron 25.2.0
* Node : v18.17.1

**MagicMirror² Version**: 2.24.0

**Description**: Setting `electronOptions: { fullscreen: false }` in my config files didn't seem to be working. After digging a bit in `js/electron.js` I found the following snippet

```js
mainWindow.once("ready-to-show", () => {
	mainWindow.setFullScreen(true);
	mainWindow.show();
});
```

This snippet sets the window to full-screen regardless of what's specified on `electronOptions`. Once i removed or commented out the `mainWindow.setFullScreen(true);` statement I noticed that the value from `electronOptions` began applying 

**Steps to Reproduce**: 
1. Set `electronOptions: { fullscreen: false }` on your `config.js` file
2. Run MagicMirror with `npm run start`
3. Notice that the mirror app opens in full-screen despite what we set in `config.js`
4. Locate and open to edit `js/electron.js`
5. Locate the line with `mainWindow.setFullScreen(true)` and remove it or commet it out
6. Toggle the `fullscreen` setting in `electronOptions` in your config file and see how it actually has an effect now after restarting the MagicMirror app.

**Expected Results**:  Setting `electronOptions: { fullscreen: false }` opens the MM app in a window which is not full-screen. The setting is respected and applies

**Actual Results**:  Setting `electronOptions: { fullscreen: false }` in `config.js` is not respected and it does not apply, as such the MM app opens in full-screen

khassel, 2023-08-28:
You are correct with your analysis.

AFAIK magicmirror is designed to always run in fullscreen so this is probably not a bug but a feature.

What is your use case? What should be displayed on the rest of the screen?

alejo4373, 2023-08-28:
Well, I wanted it to respect `electronOptions: { fullscreen: false }` while I develop some modules on the PI itself in that way I could have MM running on one half of the screen and my code editor on the other half.

khassel, 2023-08-28:
AFAIR we did the above hack with `mainWindow.setFullScreen(true)` together with other changes to avoid a white flashing when starting electron. As this doesn't work anymore (white flashing is back) we could remove this line.

But next problem: On `develop` we are now using `electron.screen.getPrimaryDisplay().workAreaSize` for `width` and `height`, which means "another sort" of fullscreen (see [this PR](https://github.com/MichMich/MagicMirror/pull/3161/files)).

So we would need to implement another parameter e.g. `noFullscreen` in `config.js` for your use case.

Opinions? @rejas @MichMich

bugsounet, 2023-09-12:
@khassel

> So we would need to implement another parameter e.g. noFullscreen in config.js for your use case.

for me it's the better way otherwise white flashing will be back

khassel, 2023-09-12:
in my tests the white flashing is back - unrelated to my changes ...

closed 2023-10-01 by khassel