MagicMirrorOrg/MagicMirror #3177: Full Day Events showing the day before the event

r1de, 2023-09-01:
I am using the default calendar module with Google Calendar,  and all of my full-day events always show as starting 1 day before the actual date on my magic mirrors.  

For instance, I just setup an event for "Labor Day" (in the USA), it is on Sept 4th this year.  
The MagicMirror shows this:
[image]

Yet the DEBUG info in the log file shows this for that particular event:
```
[01.09.2023 20:06.28.212] [DEBUG] Event:
{"type":"VEVENT","params":[],"start":"2023-09-04T00:00:00.000Z","datetype":"date","end":"2023-09-05T00:00:00.000Z","dtstamp":"2023-09-01T20:06:27.000Z","uid":"1fucsveu5nqmoumf1qf8r783am@google.com","created":"2023-09-01T17:52:51.000Z","lastmodified":"2023-09-01T17:52:51.000Z","sequence":"0","status":"CONFIRMED","summary":"No School - Labor Day","transparency":"TRANSPARENT","method":"PUBLISH"}
[01.09.2023 20:06.28.212] [DEBUG] start: Mon Sep 04 2023 00:00:00 GMT+0000 (Coordinated Universal Time)
[01.09.2023 20:06.28.212] [DEBUG] end:: Tue Sep 05 2023 00:00:00 GMT+0000 (Coordinated Universal Time)
[01.09.2023 20:06.28.212] [DEBUG] duration: 86400000
[01.09.2023 20:06.28.212] [DEBUG] title: No School - Labor Day
[01.09.2023 20:06.28.212] [DEBUG]  if no tz, guess based on now
[01.09.2023 20:06.28.212] [DEBUG] initial tz=Africa/Abidjan
[01.09.2023 20:06.28.213] [DEBUG] corrected tz=Africa/Abidjan
[01.09.2023 20:06.28.213] [DEBUG] start date/time=Mon Sep 04 2023 00:00:00 GMT+0000 (Coordinated Universal Time)
[01.09.2023 20:06.28.213] [DEBUG] start offset=0
[01.09.2023 20:06.28.213] [DEBUG] start date/time w tz =Mon Sep 04 2023 00:00:00 GMT+0000 (Coordinated Universal Time)
[01.09.2023 20:06.28.213] [DEBUG] event date=Mon Sep 04 2023 00:00:00 GMT+0000 (Coordinated Universal Time)
[01.09.2023 20:06.28.213] [DEBUG] event offset=0 hour=0 event date=Mon Sep 04 2023 00:00:00 GMT+0000 (Coordinated Universal Time)
[01.09.2023 20:06.28.213] [DEBUG] adjustHours=0
[01.09.2023 20:06.28.213] [DEBUG] Processing entry...
```

I'm not sure why the code is guessing the TZ is in Africa, and I am not sure that's even the issue.
The debug info doesn't even have Sept 3rd in it _anywhere_! Where is it even getting the starting date from then?
  
Things I have tried:

- tried forcing `calendarfetcherutils.js` to use my local TZ (America/Los_Angeles) _and_ to use UTC and the problem persists.
- tried switching from the `master` branch to the `develop` branch and there is no change.
- tried changing `timeFormat` in config.js

What am I doing wrong? Is this a known bug?

sdetweil, 2023-09-01:
what is your system timezone set to ?  looks like UTC (Coordinated Universal Time)...

so, Labor day is US holiday, starts at 00:00 on the 4th **IN US timezones..** the cal says DTSTART;VALUE=DATE:20230904 LOCAL TIME where the calendar is running.. so it says 4th at 00:00 UTC..   take away 5 hours, is 19:00 on the 3rd in NY
but is a full day event so no hour.

r1de, 2023-09-01:
The TZ on my local system was correct _but_ the TZ on the server hosting the MM2 pages was still set to UTC. :man_facepalming:
Changing the server to use `America/Los_Angeles` solved it.  
  
Thank you for making me think of that!!   :smile:

closed 2023-09-01 by r1de

sdetweil, 2023-09-01:
calendar processing is done on the server, 100%  only display is done on client, but it is already formatted