MagicMirrorOrg/MagicMirror #3189: yr provider not working for weather - yr api deprecation

matt-thurling, 2023-09-12:
Using yr as the weather provider no longer works - it looks like it might be due to the sunrise 2.0 API being end of life.
Hitting the 2.0 URL that the yr provider uses returns an error: 
```
404 Not Found
The specified version number is end-of-lifed for this product
```
Their documentation confirms this occured on the 1st September 2023 - see the changelog at https://api.met.no/weatherapi/sunrise/3.0/documentation

rejas, 2023-09-12:
Thx for the report, can indeed confirm that. 

Not sure if we can get that updated until the next release, maybe the original author @MagMar94  can take a look? Or are you capable / willing to help here @matt-thurling ?

MagMar94, 2023-09-12:
Thanks for reporting this issue and tracking down the root cause! I have noticed it on my own mirror, but haven't had the time to troubleshoot it yet. I will get around to fixing it, but just go ahead if anyone gets a chance before me 😊 I have a few busy days ahead.

PR #3197 (2023-09-20): Update yr provider to new api 
Some changes after yr api was deprecated and replaced with a new one.

Fixes #3189
commit 8b1c279c (2023-09-20): Update yr provider to new api  (#3197)

Some changes after yr api was deprecated and replaced with a new one.

Fixes #3189
closed 2023-10-01 by khassel