MagicMirrorOrg/MagicMirror #3196: client mode instructions unclear

briodan, 2023-09-14:
are there any client mode instruction that actually describe how to use it?

the instructions [here](https://docs.magicmirror.builders/getting-started/installation.html#manual-installation) say to run with
 
`node clientonly --address 192.168.1.5 --port 8080`

but unfortunately that does not tell me where that goes?

sdetweil, 2023-09-14:
all commands are entered in a linux terminal window, bash/sh
so you would open a terminal window and paste all that, or use a tool like pm2 (node process manager) to launch a terminal window in the background and it would execute that command

briodan, 2023-09-15:
except that does not work

```
magicmirror@magicmirror:~/MagicMirror $ node clientonly --address 192.168.1.5 --port 8080
Client: [14.09.2023 19:48.12.576] [LOG]   Starting MagicMirror: v2.24.0
Client: MESA-LOADER: failed to retrieve device information
Client: MESA-LOADER: failed to retrieve device information
Client: MESA-LOADER: failed to retrieve device information
```

and assuming that it does work how does one getting working on a brand new install of magic mirror which starts in full mode by default after a reboot?

sdetweil, 2023-09-15:
this is the client mode startup bug we found after release. use the develop branch or wait til the october 1 release

khassel, 2023-09-15:
duplicate of #3151

briodan, 2023-09-15:
are you talking about the one where fixes are outline here https://github.com/MichMich/MagicMirror/pull/3154/files

however that still does not answer the question of how does one run in client mode?

after a brand new install both OS and app, magic mirror starts in full mode after a reboot.
- I can't stop it as "npm run stop" does not  not exist in the scripts.
- ps aux | grep node | grep -v "color"  | awk '{print $2}' | xargs sudo kill -9 gets it to quit but it starts back up on its own in full mode

So what changes need to be made and where? does package.json need to be edited to change the run command? if so what does it need to be change to? its not the node command because that causes it to not run at all

briodan, 2023-09-15:
> duplicate of #3151

this is not a duplicate of 3151, 

3151 deals with an issue of client mode, this deal with how do you even run client mode

khassel, 2023-09-15:
you did not say how you installed. For running in client mode you need only to clone the repo and run `node clientonly --address a.c.c.d --port n` from the mm directory. It seems you have configured pm2 to run full mode on startup.

For running client mode you need another mm instance running, the ip-address and the port of the full instance are the parameters needed in above statement for client-mode.

If you want to contribute to the docs we are happy for every PR [here](https://github.com/MichMich/MagicMirror-Documentation).

sdetweil, 2023-09-15:
if you installed using my script you would change the start script in MagicMirror/installers/mm.sh

to do node... vs npm start

briodan, 2023-09-15:
> you did not say how you installed. For running in client mode you need only to clone the repo and run `node clientonly --address a.c.c.d --port n` from the mm directory. It seems you have configured [...]

happy to contribute an documentation PR as soon as I figure it out

just pulling the repo and running `node clientonly --address a.c.c.d --port n` in the mm directory errors out with `Unable to connect to server: (Error: Cannot find module 'electron'`

based on the docs these are the manual install steps

```
Clone the repository and check out the master branch: git clone https://github.com/MichMich/MagicMirror
Enter the repository: cd MagicMirror/
Install the application: npm run install-mm
Make a copy of the config sample file: cp config/config.js.sample config/config.js
Start the application: npm run start
For Server Only use: npm run server
```

if I don't run `npm run install-mm` it errors out as above
if I run `npm run install-mm` it starts in full mode after a reboot

what am I missing?

briodan, 2023-09-15:
ok i think it figured it out

go through the first 3 steps of the install instructions

Clone the repository and check out the master branch: git clone https://github.com/MichMich/MagicMirror
Enter the repository: cd MagicMirror/
Install the application: npm run install-mm

then edit package.json

and change
"start": "DISPLAY=\"${DISPLAY:=:0}\" ./node_modules/.bin/electron js/electron.js",
to
"start": "node ./clientonly --address x.x.x.x --port 8080",

then you can run npm run start or reboot and it will run in client only mode

sdetweil, 2023-09-15:
NO. NEVER EDIT ANY FILE WE SHIP. it will break updates. 

edit the file I told you

briodan, 2023-09-15:
> NO. NEVER EDIT ANY FILE WE SHIP. it will break updates.
> 
> edit the file I told you

what does that mean? whats the difference between editing package.json vs editing mm.sh are both files not in the repo and get updated?

sdetweil, 2024-08-20:
oh btw mm.sh is removed from the distro in the next release (Oct 2024)

there is ONE install, 
from the one install you can run

full (npm start)
server (node serveronly)
or client (node clientonly --address x.x.x.x --port yyyy)

but you MUST run install regardless

you can avoid client (well, and can avoid installing MM too) , by using a system browser chrome, edge, safari, firefox, midori,surf...

rejas, 2024-11-16:
I will take on https://github.com/MagicMirrorOrg/MagicMirror-Documentation/pull/186 and get it merged (sooner or later)

closed 2025-08-29 by KristjanESPERANTO