MagicMirrorOrg/MagicMirror #3199: Google Calendar not loading for one calendar

csdorman, 2023-09-16:
I'm running into a strange issue with my MM configuration.

I am pulling data from 4 different Google calendars on my configuration - two of them are calendars from the same Google account. One of them loads fine, and the other does not load at all. 

Both of them are using the private iCal link - one of them is a "group" private link and the other is just a simple private iCal link.

Has anyone run into this issue or found a solution?

Thanks!

sdetweil, 2023-09-16:
can you show the configuration for the calendar module you can XXX out that URLs Don't change anything else

csdorman, 2023-09-17:
Here it is:

```

{
			module: "calendar",
			header: "Family Calendars",
			position: "bottom_center",
			defaultSymbol: "calendar",
			showLocation: true,
			showEnd: true,
			classes: "calendar_list",
			//maximumNumberOfDays: 14,
			config: {
				calendars: [
					{
						symbol: "music",
						url: "https://googlecalendarlink",
						color: "#4986ff",
						name: "csd_work",
						//maximumNumberOfDays: "7",
						broadcastPastEvents: false,
						maximumEntries: 100,
						auth: {
							user: 'email1@gmail',
							pass: 'password',
							method: 'basic'
							}
					},
					{
						symbol: "person",
						url: "https://googlecalendarlink",
						color: "blue",
						name: "csd_personal",
						//maximumNumberOfDays: "7",
						broadcastPastEvents: false,
						maximumEntries: 100,
						auth: {
							user: 'email1@gmail',
							pass: 'password',
							method: 'basic'
							}
					},
					{
						symbol: "music",
						url: "https://googlecalendarlink",
						color: "green",
						name: 'wife_work',
						//maximumNumberOfDays: "7",
						broadcastPastEvents: false,
						maximumEntries: 100,
						auth: {
							user: "email2@gmail",
							pass: "password",
							method: "basic"
							}
					},
					{
						symbol: "house",
						url: "https:googlecalendarlink",
						color: "orange",
						name: "house",
						//maximumNumberOfDays: "14",
						broadcastPastEvents: true,
						pastDayCount: '14',
						maximumEntries: 100,
						auth: {
							user: "email3@gmail",
							pass: "password",
							method: "basic"
							}
					},
				]
			}
		},

```

sdetweil, 2023-09-17:
with the public or secret URL you shouldn't need any of the authentication information in your blocks So what I would do is test just make sure that you can download each ICS

at the command line do 
```
curl -sL ical_url  >xxx.ics
```
for each Ical url.    make sure it's the Ical url , not embedded . 

after each command  check the contents of the xxx.ics to make sure that the download work properly

csdorman, 2023-09-17:
Thanks - I've removed the authentication and tested it from the terminal, and I can download the .ICS files, and they appear to be fine. 

However, the calendar still isn't appearing in the MagicMirror calendar module.

sdetweil, 2023-09-17:
would u be willing to share the calendar config and the urls so i can test?    

don't post here. 

send to my email , the same userid as here, at gmail

Themaster312, 2023-10-02:
Same Problem here

khassel, 2023-10-15:
I'm closing this as there is no response

faspina, 2024-01-05:
Did you ever fix this . I have 3 google calendar I am sharing and one of them freezes MM. Curl download works find it has to be the way its processing the data. I can email you a URL

sdetweil, 2024-01-05:
@faspina yes please sdetweil at gmail

faspina, 2024-01-05:
> @faspina yes please sdetweil at gmail

Sent

sdetweil, 2024-01-05:
SO... the calendar loads fine on 2.25, and 2.26 on multiple machines..

so, its got to be environmental.. 

how did you upgrade to the release you are on? 

my script, or manually? 

please run this 

```
bash -c  "$(curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/upgrade-script.sh)" force
```

then restart mm

can u show me the config.js for this cal, xxx out the url.

faspina, 2024-01-05:
This problem was happening on 2.25 , I just never figured it out it was a
single calendar.
I did a manual upgrade
git pull then npm run install-mm
I had a few error so I had to reinstall few package, but it wasn't that
big.
I am using MMM-CalendarExt3 successfully to display the calendar in a
monthly view on one page except the one I sent you. The other two google
calendars for the same account work one is public. the other is the private
url.
The one I sent you is private as well.

I ran your upgrad script , uncommented that calendar no luck. THe logs tell
me it  broad cast the events from the other two calendar then it just stops
debug shows all the event broad casting then nothing

My mm no longer responds to anything , I have some hardwire buttons, and
MQTT commands I can send it , but they don't work.

When I comment out that one calendar. it all works with the other two in
place. It is not the MMM-CalendarExt modules , when I comment that out the
same thing happens. So I have narrowed it down to the calendar section that
loads this calendar , it is the last one in the list. I also tried having
only that calendar no luck.

It weird

Here is my config.js section

{
module: "calendar",
// position: "middle_center", // This can be any of the regions. Best
results in left or right regions.
   config: {
coloredText: true,
coloredBorder: true,
coloredSymbol: true,
maximumNumberOfDays: 30,
tableClass: "small",
coloredBackground: true,
broadcastEvents: true,
broadcastPastEvents: true,
calendars: [

   {
url: 'xxxx',
fetchInterval:  60 * 60 * 1000,
name: 'work',
color: "green",
},

   {
url: 'xxxx',
name: 'auburn',
color: 'orange',
fetchInterval:  60* 60 * 1000,
},

{
color: "blue",
name: "family",
fetchInterval:  60 * 60 * 1000,
url: "
https://calendar.google.com/calendar/ical/brigidspina%40gmail.com/private-065de47caae525540e8b162d9af31b3f/basic.ics
",
bgColor: 'white',
},

],
},
},

faspina, 2024-01-05:
No luck on the upgrade script. That calendar is shared from my wifes account to mine but I can read/write . I pulled the secret Url right from her account and its not working.

sdetweil, 2024-01-05:
what do you mean on no luck w upgrade?  it did the upgrade, but no difference?

but you can curl download the file using hte secret url?

faspina, 2024-01-05:
Yes to both question 

upgrade went through , no difference 

the url works fine when I down lot or import it to another calendar

sdetweil, 2024-01-05:
and it works fine on my machines, intel desktop, raspi 4, raspi3, odroid.

michaelwschultz, 2024-03-01:
I had what seems to be a similar issue. Turns out, my calendar was just to big. I've used the same calendar for 15 years and searching through all that slowed everything down to a crawl. My swap blows up and then everything just haults.

sdetweil, 2024-10-07:
@michaelwschultz   so, can we close this?  my cal is also old, 2016

michaelwschultz, 2024-10-09:
@sdetweil I'd say so. At least that was my issue. Glad you don't see the same thing. Maybe I just have too many events? lol who knows.

sdetweil, 2024-10-09:
can you do something 

download your cal to a file
from a terminal window
``` 
curl  url-to-cal >somefile.txt
``` 
then count the events
``` 
grep -i  "begin:vevent" |wc -l
``` 
wc is word count -l is lines

closed 2025-01-11 by github-actions[bot]