MagicMirrorOrg/MagicMirror #3208: Feature Request: Generic Notification for Compliments Module

NolanKingdon, 2023-09-26:
### Context

I recently [got a request](https://github.com/NolanKingdon/MMM-MoonPhase/issues/35) for my Moonphase module to integrate with the compliments module so that users can define compliments to appear during certain moonphases. 

### Requested Feature

I noticed that there's [already some code to enable this for the current weather](https://github.com/MichMich/MagicMirror/blob/master/modules/default/compliments/compliments.js#L176) in the `compliments.js` module and was wondering if there was any interest in expanding this to be more flexible/generic with module provided compliments. 

This would allow any module to send a notification that would be picked up by the compliments module without needing to explicitly define a new unique notification in `compliments.js` whenever someone wants to integrate.

If that doesn't sound like a good idea, I'd could also take a stab at a `CURRENT_MOONPHASE` notification handler that behaves similarly to the `CURRENTWEATHER_TYPE` would also enable the requested feature from my module :slightly_smiling_face:.

If this seems like a good idea, let me know and I can take a stab at an implementation.

### Idea/Details

I think a generic notification like `COMPLIMENT_CONTEXT` who's payload (in combination with the notification's `sender`) can be used to look in the compliments config file for module specifics would allow for this flexibility. 

The payload for the notification would be fairly simple and have one or more key/value pairs which could be used to map to the config file using a pattern like:

```
{
    compliments: {
        <notification sender>: {
            <payload key>: {
                <payload value>: [ ...compliments... ]
            }
        }
    }
}
```

So for example - a `COMPLIMENT_CONTEXT` notification for the current moonphase would simply look like:

```
{
    phase: "full"
}
```

Using the `sender` argument from `notificationReceived`, we would look for a config that looks something like this:

```
{
    compliments: {
        MMM-MoonPhase: {
            phase: {
                full: [ "The moon tonight is as bright as your future!" ],
                first-quarter: [ "The moon tonight is the same shape as your smile" ]
            }
        }
    }
}
```

These context payloads could be stored/updated in a similar way that the `currentWeatherType` is in the existing code, just using an object instead of a string.

bugsounet, 2023-09-27:
Maybe I'm wrong but I don't think the team will want to make this kind of change.
Simply because the third party module is not officially part of MagicMirror (made by a user and not by the MM² team)

I think more, in your case, use the existing (currentWeatherType) rather than making an object and check the sender because in all cases it will be added automatically

This is just my point of view but what does @rejas think?

sdetweil, 2023-09-27:
i don't think a more generic notification is a bad thing really.  BUT, I don't like the proposal. we shouldn't need unique by source module restrictions/ 

just the key and list of potential strings associated with that key

NolanKingdon, 2023-09-27:
> Maybe I'm wrong but I don't think the team will want to make this kind of change. Simply because the third party module is not officially part of MagicMirror (made by a user and not by the MM² team)
> 
> I think more, in your case, use the existing (currentWeatherType) rather than making an object and check the sender because in all cases it will be added automatically
> 
> This is just my point of view but what does @rejas think?
> 
> Note: However, you can code it and propose a PR ;)

Hijacking `currentWeatherType` would definitely work, but would cause a "competition" of sorts between the two modules should the user want both weather and moonphase compliments - Moonphase would overwrite current weather and vice versa. If one module updates at a much slower interval than the other, there's a good chance you would only ever see the compliments from the most frequently updating module.

> i don't think a more generic notification is a bad thing really. BUT, I don't like the proposal. we shouldn't need unique by source module restrictions/
> 
> just the key and list of potential strings associated with that key

Hmm that's fair. The sender shouldn't "own" the property since that kind of defeats the point of it being generic. I agree with this - the json body should probably look more like:

```
{
    compliments: {
        moonphase-full: [ "The moon tonight is as bright as your future!" ],
        moonphase-first-quarter: [ "The moon tonight is the same shape as your smile" ]
    }
}
```

sdetweil, 2023-09-27:
and the notification is just whatever the generic notification identifier is, and the key in your case moonphase - full.

NolanKingdon, 2023-09-27:
> and the notification is just whatever the generic notification identifier is, and the key in your case moonphase - full.

Yes that's right.

Thinking a bit more about this, I think I would implement it as something along the lines of:

**Notification**: `COMPLIMENT_CONTEXT`
**Payload**:
```
{
  moonphase: "full"
}
```
**Compliments config**:
```
{
    compliments: {
        moonphase: {
            full: [ "The moon tonight is as bright as your future!" ],
            first-quarter: [ "The moon tonight is the same shape as your smile" ]
        }
    }
}
```

**compliments.js**:
An object that contains any of the payload keys the module receives from `COMPLIMENT_CONTEXT`:
```js
this.contextState = {
  moonphase: "first-quarter",
  // ... And other provided values
}
```
Which is used to inform the [compliments available](https://github.com/MichMich/MagicMirror/blob/master/modules/default/compliments/compliments.js#L97):

```js
foreach(const [key, value] in Object.entries(this.contextState)) {
	if (this.config.compliments[key] && this.config.compliments[key][value] ) {
		Array.prototype.push.apply(compliments, this.config.compliments[key][value]);
	}
}
```

*Obligatory I have not run this code disclaimer* :slightly_smiling_face:

sdetweil, 2023-09-27:
I think that unnecessarily complicates it and makes it back towards module specific notifications rather than strings So I'm against this modification this way again

NolanKingdon, 2023-09-27:
Maybe I'm misunderstanding what you meant with your comment. My understanding (using moonphases as an example) was that we would do something like:

**Notification**: `CURRENT_CONTEXT`
**Payload**: `"moonphase-full"`

Which could work, but when the moonphase changes we wouldn't be able to easily remove old moonphase out of wherever we store it. We would just be getting a new phase like `moonphase-first-quarter`, which wouldn't indicate that `moonphase-full` no longer applies.

Having the payload be an object with a `moonphase` key allows any module to send a payload object that has that key, but also lets the code go to our `this.contextState` object and just update the `moonphase` key with the new value.

Regardless - I agree that this adds some complexity, which it sounds like the team wants to avoid. If that's the case, I'm more than happy to close the request :slightly_smiling_face:

sdetweil, 2023-09-27:
the sending module decides when that 'moonphase-full' notice gets alerted to the user.(note that is can be an array of strings too) the compliments module combines then selects random.    

there is one full phase every month.. what day is not determined in the config...  so you can post all 4 of them waxing/waning, full and nothing..   and post a notification whenever...  the 'message' is hard coded in the compliments config.. so no other module will interfere, cause they don't know what your module is sending..

NolanKingdon, 2023-09-29:
Yes this is my understanding as well, but my comment is concerned solely with how we *structure* the config and payload, not how the compliments module works.

What I'm saying is that if the payload was a string, we have to store it in `compliments.js` so the module knows to incorporate the related `config.compliments` key into the available pool of compliments. But compliments - specifically weather based ones - don't stay available forever; they go away when the weather changes. So we have to remove it as well at some point.

Say we got a `moonphase-full` followed by a `moonphase-waning_gib` at some point after the first notification.

To know we have to replace `moonphase-full` with `moonphase-waning_gib`, we would need to:
1) split the string on *something* we trust all modules send in their payloads (a `-` in our case),
2) find something common that links the two payloads (in this case, probably "moonphase", but maybe also "MoonPhase" and "moon_phase"?), 
3) look to see if any previous payloads that have that key exist in `compliments.js`'s payload history, 
4) see if the trailing string values have changed, and 
5) if they have replace them with our new value so we don't get full moon compliments when it's a waning gibbous.

Which is a bunch of steps and code to basically:
1) Identify a key
2) Replace it's value

Which sounds like the perfect fit for an object.

NolanKingdon, 2023-09-29:
But this is all debate over implementation. I guess to refocus things - Is there a non-implementation related reason that this feature should or should not be added? Totally fine if that reason is "I don't like it", I just want to get some answers to bring back to my module's issue author.

sdetweil, 2023-09-29:
well the notification takes two things

1. The keyword to identify what kind of notification it is. Note that this is broadcast to every module in the system so it's the ID of what to do with the data
2. the data.  as you want to have selectable types of notifications moon full moon waning whatever and you want to message that goes along with it then in the config you would have to have that key and the data hard coded or you would need to pass it in as part of the message but that's not the way compliments works

So in the config you would have the keyword Moon face full and then the array of choices of messages to display when that keyword is detected that array could be only one element but 

so then you're module sends the notification with custom alert or whatever the string is on the left and then the ID of the message that you configured moonphase full. The definition never goes away it's there forever and tell you somebody edits and restarts magic mirror

but then there could be the other kind that says the notification data is an object which has a key with data along with it so the key could be moon phase full and then some custom message that you want to send and maybe it's a replacement of a variable in the configured string. but it's not sending module specific. 

I don't think we want to get into validating senders. and I don't think we want a heavy rewrite of the functionality that picks the messages Right now it adds a bunch of messages to the array and then decides whether it's going to walk it next next next next next or whether it's going to randomize select one out of the list that were built concatenated collected

NolanKingdon, 2023-09-30:
I think we're having two different conversations.

I've made a copy (fork kind of?) of the compliments component [here](https://github.com/NolanKingdon/MMM-compliments/tree/master) with the changes I've talked about so I can have something for my issue reporter if they want it. 

Feel free to check it out/add it here if you like it. Hopefully it adds some clarity.

closed 2024-10-13 by github-actions[bot]