MagicMirrorOrg/MagicMirror #3212: Feature - Custom Positions

Entropy159, 2023-09-29:
It is really annoying being stuck to the limiting default positions built into Magic Mirror. I would like to be able to set custom X and Y values for positioning modules.

sdetweil, 2023-09-29:
You can use position absolute in the CSS and move your module anywhere you like, regardless of the position setting

RahulSoni2532, 2023-10-10:
can you tell me where you want to move this 

[image]

RahulSoni2532, 2023-10-10:
i can work on this

Entropy159, 2023-10-10:
Something interesting I've noticed is when I set an absolute position in the CSS, it's still relative to the area and gets messed up when other modules are toggled.

khassel, 2023-10-10:
[this thread in the forum](https://forum.magicmirror.builders/topic/18069/multiple-modules-in-a-region) is an example how to override module positions.

I don't think we will change something concerning the module positions which will break existing mm instances.

sdetweil, 2023-10-10:
@Entropy159   
> when I set an absolute position in the CSS, it's still relative to the area and gets messed up when other modules are toggled.

that is not correct... absolute is absolute. 

see here on how to use the developers console.
https://forum.magicmirror.builders/topic/14862/help-with-a-couple-css-issues?_=1696866322265

you can see and change the classes & styles live..

Entropy159, 2023-10-10:
I have the CSS for the custom positioning, and it looks like this:
#module_14_MMM-TouchButton {
 top: 150px;
 left: 200px;
 position: absolute !important;
}
When I toggle the modules above it, the button moves to the right. I used the developer console, and it's applying the position correctly, but it still seems like it's relative.

sdetweil, 2023-10-10:
yes, you are using the relative dimension tags..  top and left

use x and y

Entropy159, 2023-10-11:
X and Y don't seem to do anything...

closed 2024-10-13 by github-actions[bot]