MagicMirrorOrg/MagicMirror #3217: Calendar

Themaster312, 2023-10-02:
Since the last update from the mirror, my calendar module (default) no longer shows any appointments

KristjanESPERANTO, 2023-10-02:
Are errors shown in the console? Which node version are you using?

Themaster312, 2023-10-02:
No errors node 18.18.0

Themaster312, 2023-10-02:
node: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by node)
pi@raspberrypi:~ $ 
Using Node -v yet

sdetweil, 2023-10-02:
can you do 
lsb_release -a 

I think you are running on buster..  which takes some special work for V20 I think... testing now..

the development testing is not on raspian

Themaster312, 2023-10-02:
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
pi@raspberrypi:~ $

Themaster312, 2023-10-02:
how can I update to Bullseye

sdetweil, 2023-10-02:
ok, hang on...

sdetweil, 2023-10-02:
the node distro provides a different process for install.. supposed to work on buster...   still working on getting my buster system up... 

please try the new temp script listed at the top of the README (at the warning) 
https://github.com/sdetweil/MagicMirror_scripts

jsavko, 2023-10-02:
Also having a problem with calendar module. This is reported in the log 
```
[02.10.2023 14:52.14.250] [INFO]  Calendar-Fetcher: Broadcasting 0 events from ######.
[02.10.2023 14:52.14.252] [ERROR] Whoops! There was an uncaught exception...
[02.10.2023 14:52.14.254] [ERROR] ReferenceError: fetch is not defined
    at fetchCalendar (/magicmirror2/modules/default/calendar/calendarfetcher.js:59:3)
    at CalendarFetcher.startFetch (/magicmirror2/modules/default/calendar/calendarfetcher.js:104:3)
    at Class.createFetcher (/magicmirror2/modules/default/calendar/node_helper.js:80:11)
    at Class.socketNotificationReceived (/magicmirror2/modules/default/calendar/node_helper.js:21:9)
    at Socket.<anonymous> (/magicmirror2/js/node_helper.js:104:11)
    at Socket.emit (node:events:513:28)
    at Socket.emitUntyped (/magicmirror2/node_modules/socket.io/dist/typed-events.js:69:22)
    at /magicmirror2/node_modules/socket.io/dist/socket.js:704:39
    at processTicksAndRejections (node:internal/process/task_queues:78:11)
```

khassel, 2023-10-02:
> ReferenceError: fetch is not defined

check your node version, you must have a version >= 18 for the new release

gregimagines, 2023-10-03:
Finally got around to updated yesterday and no longer loads. Getting the same error for calendar fetching.
```
[03.10.2023 08:43.02.919] [LOG]   Create new calendarfetcher for url: https://www.calendarlabs.com/ical-calendar/ics/39/Canada_Holidays.ics - Interval: 3600000
[03.10.2023 08:43.02.921] [ERROR] Whoops! There was an uncaught exception...
[03.10.2023 08:43.02.927] [ERROR] ReferenceError: fetch is not defined
    at fetchCalendar (/home/gillisrg/MagicMirror/modules/default/calendar/calendarfetcher.js:58:3)
    at CalendarFetcher.startFetch (/home/gillisrg/MagicMirror/modules/default/calendar/calendarfetcher.js:103:3)
    at Class.createFetcher (/home/gillisrg/MagicMirror/modules/default/calendar/node_helper.js:80:11)
    at Class.socketNotificationReceived (/home/gillisrg/MagicMirror/modules/default/calendar/node_helper.js:21:9)
    at Socket.<anonymous> (/home/gillisrg/MagicMirror/js/node_helper.js:104:11)
    at Socket.emit (node:events:513:28)
    at Socket.emitUntyped (/home/gillisrg/MagicMirror/node_modules/socket.io/dist/typed-events.js:69:22)
    at /home/gillisrg/MagicMirror/node_modules/socket.io/dist/socket.js:703:39
    at process.processTicksAndRejections (node:internal/process/task_queues:78:11)
```
I went through and updated node, ended up on v19.9.0 (jumped from 16.2). Tried restarting things but getting the error every time.
lsb_release -a provides
```
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```

sdetweil, 2023-10-03:
after you update node, you have to redo the npm install and in each module that requires it..

best to use my upgrade script.. see
https://github.com/sdetweil/MagicMirror_scripts

gregimagines, 2023-10-03:
Ran the script you provided, and updated. Did not get any errors. Restarted magicmirror and same issue.
[upgrade.log](https://github.com/MichMich/MagicMirror/files/12793113/upgrade.log)

sdetweil, 2023-10-03:
no.. you ran in test mode

doing test run = true, NO updates will be applied!

the second link is apply (to actually DO the changes) 
and because u are partially installed, you need to use force instead of apply
(as the script will not update a same version system be default)

gregimagines, 2023-10-03:
that's on me. I just reran, and applied. still no luck.
[upgrade.log](https://github.com/MichMich/MagicMirror/files/12793181/upgrade.log)

sdetweil, 2023-10-03:
@gregimagines 

you did the apply path, not force 

>user requested to apply changes

the upgrade command is 
```
bash -c  "$(curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/upgrade-script.sh)" apply
```
change the word **apply** **to force**

my script will NOT upgrade a system already at the current level by default

gregimagines, 2023-10-03:
That solved it. Thank you so much! Hopefully others will benefit from this too.

sdetweil, 2023-10-03:
awesome.. thanks for the feedback

gukuyarek, 2023-10-06:
> after you update node, you have to redo the npm install and in each module that requires it..
> 
> best to use my upgrade script.. see https://github.com/sdetweil/MagicMirror_scripts

im trying to run your script to upgrade to v25 but my node is vers 12.12.12 and when i run update it tells me im on the latest version. running Bulseye.

Themaster312, 2023-10-06:
My issue is Solved, After a new Installation

Themaster312, 2023-10-06:
Can be closed

sdetweil, 2023-10-06:
@gukuyarek  change the apply to force on the end of the update command

gukuyarek, 2023-10-06:
> @gukuyarek change the apply to force on the end of the update command

still node version 12

Node.js installation Done! version=v12.22.12
Check current NPM installation ...
npm is not installed.
Installing npm ...
Extracting templates from packages: 100%
npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: 'npm@1.4.29',
npm WARN EBADENGINE   required: { node: '>=0.8', npm: '1' },
npm WARN EBADENGINE   current: { node: 'v12.22.12', npm: '7.5.2' }
npm WARN EBADENGINE }
evalmachine.<anonymous>:36

and when trying to run MM get this error

/MagicMirror $ npm start
npm ERR! code EJSONPARSE
npm ERR! path /home/pi/MagicMirror/package.json
npm ERR! JSON.parse Unexpected token "<" (0x3C) in JSON at position 3350 while parsing near "...\"colors\": \"^1.4.0\",\n<<<<<<< Updated upst..."
npm ERR! JSON.parse Failed to parse JSON data.
npm ERR! JSON.parse Note: package.json must be actual JSON, not just JavaScript.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2023-10-06T16_50_31_538Z-debug.log

sdetweil, 2023-10-06:
@gukuyarek  yes  sadly the platform is is messed up, w no real way to fix it.  

backup your mm system, see my scripts
https://github.com/sdetweil/MagicMirror-backup-restore

and rebuild the SD card and install mm again, then restore the config and modules.