MagicMirrorOrg/MagicMirror #3219: Calendar module: Digest auth missing

nkl-kst, 2023-10-02:
Hi,

I updated my MM to v2.25.0 and as stated in the changelog, the digest authentication method was removed as described in https://github.com/MichMich/MagicMirror/pull/3184.

https://github.com/MichMich/MagicMirror/pull/3184#issuecomment-1712130668 says:
> So I think we can remove digest auth because it has never worked since we replaced request

I'm using digest auth for a local setup since a few years:

```js
// config/config.js - calendar module

url: "http://local.nas/path/to/calendar",
auth: {
    user: "username",
    pass: "password",
    method: "digest",
},                                                },
```

And now I'm getting this error:

```
0|mm       | [02.10.2023 20:41.25.095] [ERROR] Calendar Error. Could not fetch calendar:  http://local.nas/path/to/calendar Error: Unauthorized
0|mm       |     at NodeHelper.checkFetchStatus (/home/pi/MagicMirror/js/node_helper.js:116:9)
0|mm       |     at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
```

It previously worked with v2.24.0. Is there a workaround to use digest auth with v2.25.0?

khassel, 2023-10-02:
I'm surprised that this worked in `v2.24.0` because there was a coding error (missing `await`).

I don't think this will come back, you can use basic auth instead.

nkl-kst, 2023-10-02:
> I'm surprised that this worked in `v2.24.0` because there was a coding error (missing `await`).
> 
> I don't think this will come back, you can use basic auth instead.

Thanks for your fast reply.

I'm not sure why the [originally linked code](https://github.com/MichMich/MagicMirror/blob/e87f50e64aeaf3a8587f4855f4679c3223959fd0/modules/default/calendar/calendarfetcher.js#L57) should not work. There are multiple `fetcher.then(...)` calls to handle the fulfilled promise, no need to `await` (also `await` doesn't work in non-`async` functions). But maybe I'm missing something.

khassel, 2023-10-02:
> I'm not sure why the [originally linked code](https://github.com/MichMich/MagicMirror/blob/e87f50e64aeaf3a8587f4855f4679c3223959fd0/modules/default/calendar/calendarfetcher.js#L57) should not work. [...]

I did dig into this weeks ago but if it works for you I made a mistake while testing.

We removed `node-fetch` in the `v2.25.0` release and are using native fetch. Could not get `digest-fetch` working with native fetch. So if you have solution to get digest working with native fetch let us know.

closed 2024-10-13 by github-actions[bot]