MagicMirrorOrg/MagicMirror #3225: Calendar not work

Tombobln23, 2023-10-08:
Hello,
I installed Batch 2.25.0 yesterday, and since then I've had problems with the calendar. Unfortunately it no longer shows me appointments with 1x Google Calendar and 2x Owncloud Calendar.

Unfortunately I can't get any further with the fix.

sdetweil, 2023-10-08:
can you show the config for the two calendars in config.js? just XXX out part of the URL for each

Tombobln23, 2023-10-08:
Google Calendar
{
			module: "calendar",
			header: "Termine Allgemein",
			position: "top_left",
			config: {
				showEnd: true,
                timeFormat: 'absolute',
				showLocation: true,
                nextDaysRelative: true,
                dateFormat: "Do MMMM, HH:mm",
                fullDayEventDateFormat: "Do MMMM",
				urgency: 0,
				getRelative: 0,
				calendars: [
					{
						symbol: "calendar-check",
						url: "https://calendar.google.com/calendar/ical/xxx%40group.calendar.google.com/public/basic.ics",
						maximumEntries: '5',
                        maximumNumberOfDays: '7',
					}
				]
			}
		},`

OWN Cloud Calendar
`{
            module: "calendar",
            header: "Müllabfuhr Parzelle 108",
            position: "top_left",
            config: {
                showEnd: false,
                timeFormat: 'absolute',
                nextDaysRelative: true,
                dateFormat: "Do MMMM, h:mm",
                fullDayEventDateFormat: "Do MMMM",
                calendars: [
                    {
                        symbol: "trash",
                        maximumEntries: '5',
                        maximumNumberOfDays: '14',
                        url: "http://xxx/remote.php/caldav/calendars/Tombo/bsr2?export",
                        auth: {
                            user: 'xxx',
                            pass: 'xxx',
                            method: 'basic'
                        }
                   }
                ]
            }
},
{
            module: "calendar",
            header: "Geburtstage",
            position: "top_left",
            config: {
                showEnd: false,
                timeFormat: 'absolute',
                nextDaysRelative: true,
                dateFormat: "Do MMMM, h:mm",
                fullDayEventDateFormat: "Do MMMM",
                calendars: [
                    {
                        symbol: "birthday-cake",
                        maximumEntries: '5',
                        maximumNumberOfDays: '14',
                        url: "http://xxx/remote.php/caldav/calendars/Tombo/bday?export",
                        auth: {
                            user: 'xxx',
                            pass: 'xxx',
                            method: 'basic'
                        }
                   }
                ]
            }
},

sdetweil, 2023-10-08:
digest auth was removed but basic should still work

can you show the messages at mm startup?
```
pm2 stop all
cd~/MagicMirror
npm  start >somefile.txt > 2&1
```
wait 20 seconds
ctrl-c to stop mm
examine the somefile.txt

make sure to xxx out any urls.

sdetweil, 2023-10-08:
also  numbers do not use quotes
```
maximumEntries: '5',
maximumNumberOfDays: '14',
```

both of those are incorrect

khassel, 2023-10-08:
probably running a node version < 18

Tombobln23, 2023-10-09:
> wahrscheinlich wird eine Knotenversion < 18 ausgeführt

Node Version 20.8.0

Tombobln23, 2023-10-09:
I found the solution. Simply install node-fetch@2. Then everything works again.

Solution found here: https://forum.magicmirror.builders/topic/18055/cannot-find-module-node-fetch-error-with-multiple-mm-modules/2

closed 2023-10-09 by khassel