MagicMirrorOrg/MagicMirror #3226: Error on startup question...

BKeyport, 2023-10-09:
Does anyone know what this is about, and how to fix it? 

[28358:1008/184025.188520:ERROR:object_proxy.cc(590)] Failed to call method: org.freedesktop.portal.Settings.Read: object_path= /org/freedesktop/portal/desktop: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.portal.Desktop was not provided by any .service files

-- while we're at it, can we modify base code to start up with the mesa-loader error disabled? 

Thanks!

sdetweil, 2023-10-09:
for the mesa loader thing add
```
export ELECTRON_DISABLE_GPU=1
```
in from of the npm start 
if you use my install script, that's
```
~/MagicMirror/installers/mm.sh
```

the other error is an assumption on electrons part that there is a desktop control file there  but it doesn't.exist on most of our systems anymore. nothing we can do about it

BKeyport, 2023-10-09:
I'm aware how to fix it after using your installer - I'm suggesting we come up with some way to proactively disable it.

As for the other one, good to know it's something not able to be fixed. Darn assumptions...

khassel, 2023-10-09:
this is coded [here](https://github.com/MichMich/MagicMirror/blob/343e7de7bd295a69e68e4ee520552e0785a99e1c/js/electron.js#L13)

I introduced this with the defensive approach, I'm open to doing it the other way around (e.g. with a new env var `ELECTRON_ENABLE_GPU`).

Opinions @sdetweil @rejas

sdetweil, 2023-12-13:
@khassel I'm ok with enable_gpu
most wouldn't need it anyhow.

jalibu, 2024-01-24:
Modules with CSS transition animations (like stock tickers) still need it.
Some MMM-Jast users report choppy rendering since 2.26.0

I would welcome a more user-friendly way to activate this option. Not all users know how to customize the startup commands and how to set ENV variables.

khassel, 2024-01-24:
> What about introducing .env files for configuration?

AFAIS this is built-in with nodejs version v20.6.0, electron is still on v18 so we would need an extra npm package for this (e.g. `dotenv`).

Beside this we have only 3 env vars: `ELECTRON_DISABLE_GPU`, `MM_CONFIG_FILE`, `MM_PORT` and the 2 last one are for running 2 mm instances from the same mm dir so it makes no sense to put them in an `.env` file.

So as long as we are talking about 1 variable I would prefer to add another line in `package.json`

```json
        "scripts": {
                "start": "DISPLAY=\"${DISPLAY:=:0}\" ./node_modules/.bin/electron js/electron.js",
                "start:gpu": "ELECTRON_ENABLE_GPU=1 DISPLAY=\"${DISPLAY:=:0}\" ./node_modules/.bin/electron js/electron.js",
                "start:dev": "DISPLAY=\"${DISPLAY:=:0}\" ./node_modules/.bin/electron js/electron.js dev",
```

so users could start mm with `npm run start:gpu` if they need the gpu stuff. But maybe there are better ideas ...

KristjanESPERANTO, 2024-01-25:
> So as long as we are talking about 1 variable I would prefer to add another line in `package.json`

I like this idea :slightly_smiling_face:  I just have a small change suggestion to avoid redundancy:

```js
    "start:gpu": "ELECTRON_ENABLE_GPU=1 && npm run start",
```

sdetweil, 2024-01-25:
this && won't work on windows.  it's not a supported platform but we give instructions for it

KristjanESPERANTO, 2024-01-25:
> this && won't work on windows. it's not a supported platform but we give instructions for it

Since `ELECTRON_ENABLE_GPU=1` (or `export ELECTRON_ENABLE_GPU=1`) also does not work on Windows I see no problem in using `&&` here. 

In [the Windows part of the documentation](https://docs.magicmirror.builders/getting-started/installation.html#other-operating-systems) we could add how to set the variable. It think it should be like this:
- PowerShell: `$env:ELECTRON_ENABLE_GPU=1`
- CMD: `set ELECTRON_ENABLE_GPU=1`

khassel, 2024-01-25:
this works: 
```json
"start:gpu": "ELECTRON_ENABLE_GPU=1 npm start",
```

khassel, 2024-01-27:
@jalibu is the above solution o.k. for you?

jalibu, 2024-01-28:
@khassel to be honest, I did not really understand this issue with the GPU.
Does it occur with everyone who uses a Raspi with Bullseye? Is it an annoying warning or is it an error?

I'm asking because unfortunately, I don't have an MM myself at the moment and have never seen the problem.

Regardless of whether we opt-in or opt-out, I think the solution is ok. It would be great if we had a selection option in the installer scripts @sdetweil

sdetweil, 2024-01-28:
@jalibu  what would be the prompt for the user. 90% are installing mm for the first time, and don't know what a GPU is, or if they might use it.   I am 7 years into using mm and not sure myself.

sdetweil, 2024-06-28:
we changed the GPU  setting to OFF  by default.  so the env string is 
ELECTRON_ENABLE_GPU =1

codac, 2024-06-28:
I have removed the "ELECTRON_DISABLE_GPU = 1" parameter (as you said its default now, but the protocol messages remain the same.

sdetweil, 2024-06-28:
correct, you have to add 

export ELECTRON_ENABLE_GPU=1

notice the output says enable

sdetweil, 2024-06-28:
the docker deployment  doesn't use electron
so it doesn't matter

closed 2024-09-26 by khassel